ZoneMinder/zoneminder #2565: Live view not working - Unable to authenticate user (zms.cpp 212)

Giulio2333, 2019-04-01:
**Describe Your Environment**
- Version of ZoneMinder: **1.32.3-bionic**
- How you installed ZoneMinder: **https://wiki.zoneminder.com/Ubuntu_Server_18.04_64-bit_with_Zoneminder_1.32.x_the_easy_way**
- Full name and version of OS: **Ubuntu 18.04.1 LTS - Bionic**
- Browser name and version: **Google Chrome 73.0.3683.86, Firefox Quantum 64.0**

**Describe the bug**
Zm is recording feed from the 4 camers is connected to, but when I try to access live feed can't see nothing and geneates errors in log:
`- Unable to authenticate user	(zms.cpp	212)`
`- Socket /var/run/zm/zms-999953s.sock does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early. Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly. Make sure that ZM is actually recording. If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.	/usr/share/zoneminder/www/includes/functions.php	2048`

The system worked properly for several months and seems to have stop working yesterday.
Don't know if it's correlated, but between yesterday and the day before here in Italy time changed (DST).

**To Reproduce**
Steps to reproduce the behavior:
Just open a live feed tab (Cycle, Montage, Montage Review...)

**Expected behavior**
Should be able to see live feed

**Relevant part of the config**
OPT_USE_AUTH - yes
AUTH_TYPE - builtin
AUTH_RELAY - hashed
AUTH_HASH_IPS - yes
AUTH_HASH_LOGINS - yes
Apache HTTPS server

**Debug Logs**

`2019-04-01 08:00:14	zmc_m4		2677	INF	East: images:300 - Capturing at 5.00 fps, capturing bandwidth 15145bytes/sec	zm_monitor.cpp	2477
2019-04-01 08:00:14	zms_m1		2949	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:13	zms_m4		2944	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:13	zms_m4		2945	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:13	web_php		1706	ERR	Socket /var/run/zm/zms-999953s.sock does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early. Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly. Make sure that ZM is actually recording. If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.	/usr/share/zoneminder/www/includes/functions.php	2048
2019-04-01 08:00:13	zms_m1		2946	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:13	zmfilter_1		2687	INF	Checking filter PurgeWhenFull delete returned 0 events	zmfilter.pl	
2019-04-01 08:00:13	zms_m1		2948	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:13	zma_m2		2664	INF	West: 200 - Analysing at 4.76 fps	zm_monitor.cpp	1247
2019-04-01 08:00:12	web_php		1288	ERR	Socket /var/run/zm/zms-307188s.sock does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early. Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly. Make sure that ZM is actually recording. If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.	/usr/share/zoneminder/www/includes/functions.php	2048
2019-04-01 08:00:12	zms_m4		2942	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:12	zms_m4		2943	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:11	zms_m3		2936	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:11	web_php		1289	ERR	Socket /var/run/zm/zms-938276s.sock does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early. Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly. Make sure that ZM is actually recording. If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.	/usr/share/zoneminder/www/includes/functions.php	2048
2019-04-01 08:00:11	zms_m3		2937	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:11	zms_m2		2938	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:11	zms_m3		2939	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:11	zms_m2		2940	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:11	zms_m3		2941	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:10	web_php		1288	ERR	Socket /var/run/zm/zms-726399s.sock does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early. Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly. Make sure that ZM is actually recording. If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.	/usr/share/zoneminder/www/includes/functions.php	2048
2019-04-01 08:00:10	zms_m3		2933	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:10	zms_m3		2934	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:10	zms_m3		2935	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:09	zms_m2		2932	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:09	zma_m1		2647	INF	Magazzino: 300 - Analysing at 5.00 fps	zm_monitor.cpp	1247
2019-04-01 08:00:09	zmc_m3		2668	INF	South: images:200 - Capturing at 5.00 fps, capturing bandwidth 19985bytes/sec	zm_monitor.cpp	2477
2019-04-01 08:00:09	zms_m2		2931	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:09	zms_m1		2930	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:09	zms_m2		2929	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:09	zms_m1		2928	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:08	web_php		1706	ERR	Socket /var/run/zm/zms-938276s.sock does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early. Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly. Make sure that ZM is actually recording. If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.	/usr/share/zoneminder/www/includes/functions.php	2048
2019-04-01 08:00:08	zmc_m2		2651	INF	West: images:200 - Capturing at 5.00 fps, capturing bandwidth 22357bytes/sec	zm_monitor.cpp	2477
2019-04-01 08:00:08	web_php		1290	ERR	Socket /var/run/zm/zms-999953s.sock does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early. Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly. Make sure that ZM is actually recording. If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.	/usr/share/zoneminder/www/includes/functions.php	2048
2019-04-01 08:00:08	zmc_m1		2642	INF	Magazzino: images:300 - Capturing at 5.00 fps, capturing bandwidth 176070bytes/sec	zm_monitor.cpp	2477
2019-04-01 08:00:08	zms_m4		2927	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:08	zms_m2		2926	ERR	Unable to authenticate user	zms.cpp	212
2019-04-01 08:00:08	zms_m2		2925	ERR	Unable to authenticate user	zms.cpp	212
[... cut ...] 2048
2019-04-01 08:00:04	web_php		2047	ERR	Socket /var/run/zm/zms-938276s.sock does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early. Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly. Make sure that ZM is actually recording. If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to `http://zoneminder.readthedo`

Giulio2333, 2019-04-01:
Update: 
With OPT_USE_AUTH = OFF, live feed start working again

wormszer, 2019-04-24:
I am seeing the same issue on ubuntu 18.04.2 LTS and zone minder 1.32
It was a brand new install and i followed the directions listed at https://zoneminder.readthedocs.io/en/latest/installationguide/ubuntu.html#easy-way-ubuntu-18-04

Everything kept pointing to the PATH_ZMS issue but it wasn't until i found some other post about 1.31 and trying AUTH_RELAY = none then it started working.

The camera was RTMP using ffmpeg and h264.

ctgreybeard, 2019-05-21:
I have just installed 1.32.3 on Ubuntu 18.04 and am also see the "Socket /var/run/zm/zms-956384s.sock does not exist." I tried turning OPT_USE_AUTH off with no change. zmc appears to be running but there is no socket for it in /var/run/zm. There is a zmdc.sock= there.

academo, 2019-06-09:
I can report the same issue on a fresh install on ubuntu 18.04.

I switched the auth relay option to "none" and solved the issue but this is of course reducing the security of the system.

MAF-398, 2019-06-09:
Same as academo. Solved by auth relay to none.

tahphallus, 2019-06-24:
same issue here on Ubuntu **18.04.2 LTS**
No images vissible on the montage tab or live view.
After setting ### OPT_USE_AUTH = off, everything works again.

However, when setting   ### AUTH_RELAY  to ### @NONE it does work.
So only when selecting hashed   or  plain   it doesn't work.

Zoneminder version:
You are running the most recent version of ZoneMinder, v1.32.3.
No update is necessary.

Contents of log:

When not working:
2019-06-24 09:38:41 | zms_m2 |   | 3057 | ERR | Unable to authenticate user | zms.cpp | 212

When working:
2019-06-24 09:51:08 | zms_m3 |   | 3401 | INF | Authenticated user 'username' | zm_user.cpp | 126
2019-06-24 09:51:08 | zms_m3 |   | 3399 | INF | Got signal 15 (Terminated), exiting | zm_signal.cpp | 40

pliablepixels, 2019-07-18:
setting auth_relay to none will most likely break newer version of zmNinja. Is this issue prevalent in 1.33?

Siggemada, 2019-07-23:
Getting this as well on multi-server install. Trying to view a USB camera on a oragePi (armhf, Debian Stretch with Armbian Linux 4.19.57-sunxi) from other server (intel, Ubuntu 18.04.2).

Also used easy way:
https://zoneminder.readthedocs.io/en/latest/installationguide/ubuntu.html#easy-way-ubuntu-18-04

gr1nd, 2019-08-25:
Have the same issue on Ubuntu 18.04 and Zoneminder 1.33.14~20190821101210-bionic as well. Setting auth_relay to none doesn't help here. Then in get: ```
fcgiwrap[3595]: terminate called after throwing an instance of 'std::logic_error'
fcgiwrap[3595]:   what():  basic_string::_M_construct null not valid```
Disabling Auth helps, but I don't want having an open installation.

mludvig, 2019-09-17:
Same experience as everyone above here. Brand new install of Ubuntu 18.04 + ZM 1.32.3 + Apache.  Had to set AUTH_RELAY to NONE to make it work.

What implications does this have? Who can do what without authenticating??

mludvig, 2019-09-17:
BTW I can see there were quite a few changes in the `zms` code related to auth in the recent months. Perhaps some of it broke it? 

https://github.com/ZoneMinder/zoneminder/commits/master/src/zms.cpp

connortechnology, 2019-09-17:
We believe this is not relevant to 1.33.14

closed 2019-09-17 by connortechnology

mludvig, 2019-09-17:
Apparently it affects a lot of users. Any chance to backport the fix to **stable** (!) 1.32 branch? I’m guessing it won’t be a complicated fix.
Thanks!

connortechnology, 2019-09-17:
No. We will be releasing 1.34 soon.

Siggemada, 2019-09-18:
I would also check the system timezone with tzdata.

That helped me.

mludvig, 2019-09-18:
> I would also check the system timezone with tzdata. That helped me.

Brilliant, that was the problem! Mismatch between system timezone and PHP `date.timezone`. Seems to work now with `AUTH_RELAY=hashed` :)

mozilla2012, 2019-12-10:
I updated my timezone to match my system timezone, but I still can't see any camera's current view without `OPT_USE_AUTH = off`, but that's not a safe way to run the system.

It appears to still capture events in the background though with the timezone fixed. So I guess a temporary solution is to keep `OPT_USE_AUTH = on` all the time unless I _need_ to see a live view temporarily.

Setting `AUTH_RELAY=plain` doesn't fix it for me, so I'm going to leave it as `hashed` for now since this workaround works until a more permanent fix is found.

connortechnology, 2019-12-10:
If RELAY=plain doesn't fix it, then the issue is not with your timezone.

Marvo2011, 2020-02-17:
Same issue here but just after an upgrade from 1.33 to 1.34.2
PHP and system timezone are the same.

`AUTH_RELAY=none` fixed it but makes me not happy.

Jemyy-AR, 2020-06-07:
the problem still exists. version 1.34.15
 Has any body solved the problem?

mludvig, 2020-06-07:
@Jemyy-AR I don't remember exactly how I fixed it but I think it had something to do with the server timezone? Somehow I managed to do it without having to resort to disabling the authentication.

(I wish I had reported here so I can find it when I need it next :)

fred9176, 2020-06-09:
Same problem here with 1.34.15.

TZ is the same between system (Ubuntu 18.04) and php

Checked with 
```
timedatectl | grep "Time zone"
cat /etc/timezone
grep timezone /etc/php/7.2/apache2/php.ini
grep timezone /etc/php/7.2/cli/php.ini
grep timezone /etc/php/7.2/phpdbg/php.ini
```

paul-h, 2020-09-27:
Making sure the PHP and system timezone are the same seems to have fixed this for me. So in Ubuntu 20.04.1 LTS I did this

`timedatectl set-timezone Europe/London`

and also make sure the php date.timezone settings match and are uncommented in both
/etc/php/7.4/cli/php.ini and /etc/php/7.4/apache2/php.ini

`date.timezone = Europe/London`

eddriesen, 2020-11-05:
> Everything kept pointing to the PATH_ZMS issue but it wasn't until i found some other post about 1.31 and trying AUTH_RELAY = none then it started working.

THANK YOU !!! auth relay did fix it.

It stopped working after I migrated the existing install from an overcrowded ub 18.04 install to a fresh 20.04 install. Went trough the install instructions nevertheless (for correct folder permissions, ....)

DentonGentry, 2021-01-17:
Setting the system timezone solved it for me as well:
`timedatectl set-timezone America/Los_Angeles`

I had set the timezone in the Zoneminder options but not in the system. The Zoneminder logs contained errors about a socket being missing, and authentication failure. Setting the system timezone resolved it.

thillRobot, 2021-07-18:
This also fixed the problem for me.

jfernandz, 2022-01-26:
@thillRobot 
> Setting the timezone with `timedatectl` also fixed the problem for me.

Same for me, the problem was actually the mismatching between system, ZM and php timezone. You have to set these three properly. I think this info should be included in here

https://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc

ohaka860, 2022-02-01:
Hi all, i had the issue when i turn on OPT_USE-AUTH and different timezone between Options in zoneminder 1.36 and the Ubuntu LTS20.04 timezone. After change server timezone (Europe/Paris same as zoneminder option), i had to refresh my browser page and the issue has gone. This due cause the browser session was also stay with a generic access, and was not reconnected with the user loging in.