ZoneMinder/zoneminder #2566: API Change State of Monitor 1 can't change

dragonsongohan, 2019-04-01:
**Describe Your Environment**
- Version of ZoneMinder: v1.33.4.
- How you installed ZoneMinder : PPA
- Full name and version of OS: Ubuntu 18.04

Steps to reproduce the behavior:
1. I use API:
curl -XOST http://192.168.1.207/zm/api/monitors/2.json -d "Monitor[Function]=Modect&Monitor[Enabled]=1"

2. result:
{
    "success": false,
    "data": {
        "name": "Class &#039;Logger&#039; not found",
        "message": "Class &#039;Logger&#039; not found",
        "url": "\/zm\/api\/monitors\/2.json",
        "exception": {
            "class": "Error",
            "code": 0,
            "message": "Class 'Logger' not found",
            "trace": [
                "#0 \/usr\/share\/zoneminder\/www\/api\/app\/Controller\/MonitorsController.php(174): MonitorsController->daemonControl('2', 'stop')",
                "#1 [internal function]: MonitorsController->edit('2')",
                "#2 \/usr\/share\/zoneminder\/www\/api\/lib\/Cake\/Controller\/Controller.php(491): ReflectionMethod->invokeArgs(Object(MonitorsController), Array)",
                "#3 \/usr\/share\/zoneminder\/www\/api\/app\/Plugin\/Crud\/Lib\/CrudControllerTrait.php(31): Controller->invokeAction(Object(CakeRequest))",
                "#4 \/usr\/share\/zoneminder\/www\/api\/lib\/Cake\/Routing\/Dispatcher.php(193): AppController->invokeAction(Object(CakeRequest))",
                "#5 \/usr\/share\/zoneminder\/www\/api\/lib\/Cake\/Routing\/Dispatcher.php(167): Dispatcher->_invoke(Object(MonitorsController), Object(CakeRequest))",
                "#6 \/usr\/share\/zoneminder\/www\/api\/app\/webroot\/index.php(107): Dispatcher->dispatch(Object(CakeRequest), Object(CakeResponse))",
                "#7 {main}"
            ]
        },
        "queryLog": {
            "default": {
                "log": [
                    {
                        "query": "SELECT COUNT(*) AS `count` FROM `zm`.`Monitors` AS `Monitor`   WHERE `Monitor`.`Id` = 2",
                        "params": [],
                        "affected": 1,
                        "numRows": 1,
                        "took": 0
                    },
                    {
                        "query": "BEGIN",
                        "params": [],
                        "affected": false,
                        "numRows": false,
                        "took": false
                    },
                    {
                        "query": "SELECT COUNT(*) AS `count` FROM `zm`.`Monitors` AS `Monitor`   WHERE `Monitor`.`Id` = 2",
                        "params": [],
                        "affected": 1,
                        "numRows": 1,
                        "took": 0
                    },
                    {
                        "query": "SELECT COUNT(*) AS `count` FROM `zm`.`Monitors` AS `Monitor`   WHERE `Monitor`.`Id` = 2",
                        "params": [],
                        "affected": 1,
                        "numRows": 1,
                        "took": 0
                    },
                    {
                        "query": "SELECT COUNT(*) AS `count` FROM `zm`.`Monitors` AS `Monitor`   WHERE `Monitor`.`Id` = 2",
                        "params": [],
                        "affected": 1,
                        "numRows": 1,
                        "took": 0
                    },
                    {
                        "query": "UPDATE `zm`.`Monitors` SET `Function` = 'Modect', `Enabled` = 1  WHERE `zm`.`Monitors`.`Id` = '2'",
                        "params": [],
                        "affected": 0,
                        "numRows": 0,
                        "took": 0
                    },
                    {
                        "query": "COMMIT",
                        "params": [],
                        "affected": false,
                        "numRows": false,
                        "took": false
                    },
                    {
                        "query": "SELECT `Monitor`.`Function`, `Monitor`.`ServerId` FROM `zm`.`Monitors` AS `Monitor`   WHERE `Id` = 2    LIMIT 1",
                        "params": [],
                        "affected": 1,
                        "numRows": 1,
                        "took": 0
                    },
                    {
                        "query": "SELECT `Monitor`.`Type`, `Monitor`.`Function`, `Monitor`.`Device` FROM `zm`.`Monitors` AS `Monitor`   WHERE `Id` = 2    LIMIT 1",
                        "params": [],
                        "affected": 1,
                        "numRows": 1,
                        "took": 0
                    }
                ],
                "count": 9,
                "time": 0
            }
        }
    }
}

```

connortechnology, 2019-04-02:
I believe I have committed a fix for this.  Please update and test.

dragonsongohan, 2019-04-05:
Thank you, i will check it out

closed 2019-04-21 by no-response[bot]