ZoneMinder/zoneminder #2569: Limit the number of cameras for user

dragonsongohan, 2019-04-06:
**Describe Your Environment**
- Version of ZoneMinder : release version v1.32.3
- How you installed ZoneMinder : PPA 
- Full name and version of OS : Ubuntu 18.04 LTS

**Describe the bug**
I have 300 cameras and I can't grant the camera full view to users.
I check in zoneminder database, in "User" table,  attribute "MonitorIds" has the data type "tinytext", and "Tinytext" data type limits 256 bytes ~~ 256 characters.

Please fix that.

connortechnology, 2019-04-07:
I could have sworn that I fixed this a long time ago but looks like I screwed it up instead. 

You can fix your db using the following:
ALTER TABLE Users MODIFY MonitorIds text;

closed 2019-04-07 by connortechnology

dragonsongohan, 2019-04-08:
Thank you very much!

commit d776f63b (2019-04-07): Change MonitorIds from TinyText to TEXT. Fixes #2569