ZoneMinder/zoneminder #2572: Audio is not properly transcoded to AAC when using MP2 audio with H264 camera passthrough

neillbell, 2019-04-11:
**Describe Your Environment**
- Version of ZoneMinder: 1.33.6 (1.33.6-bionic)
- How you installed ZoneMinder: ppa:iconnor/zoneminder-master
- Ubuntu 18.04.2 LTS

**If the issue concerns a camera**
- Hikvision DS-2CD2442FWD-IW
- 10 FPS
- 1920x1080
- Audio selected as: MP2L2, Sampling rate 16kHz, Stream bitrate 64kbps
- ZoneMinder Source Type: FFMPEG, H264 Camera Passthrough, audio storage enabled

**Describe the bug**
When using MP2 audio, the audio is not properly transcoded to AAC and is therefore not saved. The audio for this camera works properly when using G.726 audio.

**Debug Logs**
The following debug log shows the zmc_m?.log file at debug level 1 from the time of Zoneminder startup until the end of the first event capture.
[zmc_m9.log](https://github.com/ZoneMinder/zoneminder/files/3070348/zmc_m9.log)

connortechnology, 2019-04-25:
Can you update to latest master and give me a log at debug level 2?

neillbell, 2019-04-25:
Here you go.
[zmc_m9-2019042502.log](https://github.com/ZoneMinder/zoneminder/files/3118320/zmc_m9-2019042502.log)

neillbell, 2019-04-25:
Well don't I feel dumb. It looks like this line:
https://github.com/ZoneMinder/zoneminder/blob/ea7c38ceff7b1231c416cd45f96657c979020c98/src/zm_videostore.cpp#L1195
is looking at the input frame instead of the output frame. I made the switch and it works fine.

neillbell, 2019-04-25:
I spoke too soon. The output file has an audio duration that is longer than the video duration.
```
General
Complete name                            : 202965-video.mp4
Format                                   : MPEG-4
Format profile                           : Base Media
Codec ID                                 : isom (isom/iso2/avc1/mp41)
File size                                : 4.56 MiB
Duration                                 : 1 min 9 s
Overall bit rate mode                    : Variable
Overall bit rate                         : 552 kb/s
Movie name                               : Zoneminder Security Recording
Writing application                      : Lavf57.83.100

Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : High@L4.1
Format settings                          : CABAC / 1 Ref Frames
Format settings, CABAC                   : Yes
Format settings, ReFrames                : 1 frame
Format settings, GOP                     : M=1, N=20
Codec ID                                 : avc1
Codec ID/Info                            : Advanced Video Coding
Duration                                 : 12 s 500 ms
Bit rate mode                            : Variable
Bit rate                                 : 3 029 kb/s
Maximum bit rate                         : 4 910 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate mode                          : Constant
Frame rate                               : 10.000 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 0.146
Stream size                              : 4.51 MiB (99%)
Color range                              : Full
Color primaries                          : BT.709
Transfer characteristics                 : BT.709
Matrix coefficients                      : BT.709

Audio
ID                                       : 2
Format                                   : AAC
Format/Info                              : Advanced Audio Codec
Format profile                           : LC
Codec ID                                 : mp4a-40-2
Duration                                 : 1 min 9 s
Bit rate mode                            : Variable
Bit rate                                 : 5 264 b/s
Maximum bit rate                         : 32.8 kb/s
Channel(s)                               : 2 channels
Channel(s)_Original                      : 1 channel
Channel positions                        : Front: C
Sampling rate                            : 16.0 kHz
Frame rate                               : 15.625 FPS (1024 SPF)
Compression mode                         : Lossy
Stream size                              : 44.6 KiB (1%)
Default                                  : Yes
Alternate group                          : 1
```

connortechnology, 2019-04-25:
that was a good catch, I'm sure the duration is something stupid like that too.

connortechnology, 2019-05-10:
So what is happening is that our sample buffer appears to not be accepting our samples... I can't imagine why.  I have added a bit more debugging.  Can you update to latest master and send me another debug log?

neillbell, 2019-05-10:
I assume you mean the latest from the PPA. If so, here you go.
[zmc_m9.log](https://github.com/ZoneMinder/zoneminder/files/3168089/zmc_m9.log)

If you want the latest from github, I can do that too.

Also, here is the mediainfo output for the file generated.
```
General
Complete name                            : /var/cache/zoneminder/events/9/2019-05-10/228734/228734-video.mp4
Format                                   : MPEG-4
Format profile                           : Base Media
Codec ID                                 : isom (isom/iso2/avc1/mp41)
File size                                : 4.78 MiB
Duration                                 : 1 min 15 s
Overall bit rate mode                    : Variable
Overall bit rate                         : 535 kb/s
Movie name                               : Zoneminder Security Recording
Writing application                      : Lavf57.83.100

Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : High@L4.1
Format settings                          : CABAC / 1 Ref Frames
Format settings, CABAC                   : Yes
Format settings, ReFrames                : 1 frame
Format settings, GOP                     : M=1, N=20
Codec ID                                 : avc1
Codec ID/Info                            : Advanced Video Coding
Duration                                 : 13 s 400 ms
Bit rate mode                            : Variable
Bit rate                                 : 2 960 kb/s
Maximum bit rate                         : 4 910 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate mode                          : Constant
Frame rate                               : 10.000 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 0.143
Stream size                              : 4.73 MiB (99%)
Color range                              : Full
Color primaries                          : BT.709
Transfer characteristics                 : BT.709
Matrix coefficients                      : BT.709

Audio
ID                                       : 2
Format                                   : AAC
Format/Info                              : Advanced Audio Codec
Format profile                           : LC
Codec ID                                 : mp4a-40-2
Duration                                 : 1 min 15 s
Bit rate mode                            : Variable
Bit rate                                 : 5 259 b/s
Maximum bit rate                         : 32.8 kb/s
Channel(s)                               : 2 channels
Channel(s)_Original                      : 1 channel
Channel positions                        : Front: C
Sampling rate                            : 16.0 kHz
Frame rate                               : 15.625 FPS (1024 SPF)
Compression mode                         : Lossy
Stream size                              : 48.2 KiB (1%)
Default                                  : Yes
Alternate group                          : 1

```

connortechnology, 2019-05-11:
First glance at that loss makes me think that it is working. Can you confirm that it isn't?

neillbell, 2019-05-11:
I think I misunderstood what you were looking for. Fixing the problem of looking at the input buffer instead of the output buffer definitely allows the encoding to proceed as intended. The only problem remaining is the incorrect duration recorded for the audio stream.

connortechnology, 2019-05-12:
Ok, please update and give me another debug log.  The issue I think is due to rescaling the duration and pts values.  Can you crank up the debug level to 3 please?

neillbell, 2019-05-12:
Here you are.
[zmc_m9.log](https://github.com/ZoneMinder/zoneminder/files/3170441/zmc_m9.log)

neillbell, 2019-05-12:
It looks like it is fixed! Here is the mediainfo output:
```
General
Complete name                            : /var/cache/zoneminder/events/9/2019-05-12/230812/230812-video.mp4
Format                                   : MPEG-4
Format profile                           : Base Media
Codec ID                                 : isom (isom/iso2/avc1/mp41)
File size                                : 4.38 MiB
Duration                                 : 12 s 200 ms
Overall bit rate mode                    : Variable
Overall bit rate                         : 3 014 kb/s
Movie name                               : Zoneminder Security Recording
Writing application                      : Lavf57.83.100

Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : High@L4.1
Format settings                          : CABAC / 1 Ref Frames
Format settings, CABAC                   : Yes
Format settings, ReFrames                : 1 frame
Format settings, GOP                     : M=1, N=20
Codec ID                                 : avc1
Codec ID/Info                            : Advanced Video Coding
Duration                                 : 12 s 200 ms
Bit rate mode                            : Variable
Bit rate                                 : 2 982 kb/s
Maximum bit rate                         : 4 910 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate mode                          : Constant
Frame rate                               : 10.000 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 0.144
Stream size                              : 4.34 MiB (99%)
Color range                              : Full
Color primaries                          : BT.709
Transfer characteristics                 : BT.709
Matrix coefficients                      : BT.709

Audio
ID                                       : 2
Format                                   : AAC
Format/Info                              : Advanced Audio Codec
Format profile                           : LC
Codec ID                                 : mp4a-40-2
Duration                                 : 12 s 152 ms
Duration_FirstFrame                      : -8 ms
Duration_LastFrame                       : -8 ms
Bit rate mode                            : Variable
Bit rate                                 : 29.3 kb/s
Maximum bit rate                         : 32.8 kb/s
Channel(s)                               : 2 channels
Channel(s)_Original                      : 1 channel
Channel positions                        : Front: C
Sampling rate                            : 16.0 kHz
Frame rate                               : 15.625 FPS (1024 SPF)
Compression mode                         : Lossy
Stream size                              : 43.5 KiB (1%)
Default                                  : Yes
Alternate group                          : 1

```

connortechnology, 2019-05-17:
If it is working then please close this issue!

neillbell, 2019-05-17:
Sorry, I didn't realize that I could close an issue myself. I'm kind of new at this.

neillbell, 2019-05-27:
The issue with the audio duration being much greater than the video duration has started happening again. My zoneminder version is now 1.33.9~20190524141652-bionic. I'm attaching a log with debug level 3.

[zmc_m9.log](https://github.com/ZoneMinder/zoneminder/files/3224329/zmc_m9.log)

The video file generated had the following mediainfo output:
```
General
Complete name                            : /var/cache/zoneminder/events/9/2019-05-27/258053/258053-video.mp4
Format                                   : MPEG-4
Format profile                           : Base Media
Codec ID                                 : isom (isom/iso2/avc1/mp41)
File size                                : 2.54 MiB
Duration                                 : 41 s 438 ms
Overall bit rate mode                    : Variable
Overall bit rate                         : 513 kb/s
Movie name                               : Zoneminder Security Recording
Writing application                      : Lavf57.83.100

Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : High@L4.1
Format settings                          : CABAC / 1 Ref Frames
Format settings, CABAC                   : Yes
Format settings, ReFrames                : 1 frame
Codec ID                                 : avc1
Codec ID/Info                            : Advanced Video Coding
Duration                                 : 7 s 400 ms
Bit rate mode                            : Variable
Bit rate                                 : 2 842 kb/s
Maximum bit rate                         : 4 910 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate mode                          : Constant
Frame rate                               : 10.000 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 0.137
Stream size                              : 2.51 MiB (99%)
Color range                              : Full
Color primaries                          : BT.709
Transfer characteristics                 : BT.709
Matrix coefficients                      : BT.709

Audio
ID                                       : 2
Format                                   : AAC
Format/Info                              : Advanced Audio Codec
Format profile                           : LC
Codec ID                                 : mp4a-40-2
Duration                                 : 41 s 438 ms
Duration_FirstFrame                      : -341 ms
Duration_LastFrame                       : -341 ms
Bit rate mode                            : Variable
Bit rate                                 : 5 310 b/s
Maximum bit rate                         : 32.8 kb/s
Channel(s)                               : 2 channels
Channel(s)_Original                      : 1 channel
Channel positions                        : Front: C
Sampling rate                            : 16.0 kHz
Frame rate                               : 15.625 FPS (1024 SPF)
Compression mode                         : Lossy
Stream size                              : 26.9 KiB (1%)
Default                                  : Yes
Alternate group                          : 1
```

closed 2019-05-27 by close-issue-app[bot]