ZoneMinder/zoneminder #2573: Cannot Control Netcat ONVIF

dragonsongohan, 2019-04-12:
**Describe Your Environment**
- Version of ZoneMinder : v.1.33.6
- How you installed ZoneMinder : PPA
- Full name and version of OS: Ubuntu 18.04 LTS
- Browser name and version (if this is an issue with the web interface) : All

**If the issue concerns a camera**
- ONVIF Camera (VMS contains 100 camera) and i can't ptz camera 

**Describe the bug**
1) I read file Netcat.pm, and the file content, the method's body POST is as follows:

my $msg ='<s:Envelope xmlns:s="http://www.w3.org/2003/05/soap-envelope"><s:Body xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema"><ContinuousMove xmlns="http://www.onvif.org/ver20/ptz/wsdl"><ProfileToken>000</ProfileToken><Velocity><PanTilt x="0" y="0.5" xmlns="http://www.onvif.org/ver10/schema"/></Velocity></ContinuousMove></s:Body></s:Envelope>';
  my $content_type = 'application/soap+xml; charset=utf-8; action="http://www.onvif.org/ver20/ptz/wsdl/ContinuousMove"';
  $self->sendCmd($cmd, $msg, $content_type);
  $self->autoStop($self->{Monitor}->{AutoStopTimeout});

=> tag <ProfileToken>000</ProfileToken> alway 000 and i can't change ProfileToken

2) In on my ONVIF system, the method's body POST is as follows:
<s:Body xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema"><ContinuousMove xmlns="http://www.onvif.org/ver20/ptz/wsdl"><ProfileToken>00ef0bd7-fe13-483d-9275-8842b910b353</ProfileToken><Velocity><PanTilt x="0.5" y="0" xmlns="http://www.onvif.org/ver10/schema"/></Velocity></ContinuousMove></s:Body>

=> tag <ProfileToken>00ef0bd7-fe13-483d-9275-8842b910b353</ProfileToken> 

*** How can I identify ProfileToken in the Zoneminder. Profile Token works to identify a device in my ONVIF system???

[image]

knight-of-ni, 2019-04-13:
You are trying to use the Netcat ptz control script for cameras that are not Netcat. Yes, both camera types are onvif, and there is the potential that the script could work with other caermas, but there is no guarantee if it doesn't. 

What you need to do is write a control script for your cameras. However this is not the place to have that discussion. Feel free to reach out for support in our slack channel or user forum.

closed 2019-04-13 by knight-of-ni

connortechnology, 2019-04-17:
Why don't we use the Control device field to fill in the ProfileToken and just have it default to 000?