ZoneMinder/zoneminder #2577: Missing Path setting for event video

firebladed, 2019-04-15:
Zoneminder 1.32 (deb http://ppa.launchpad.net/iconnor/zoneminder-1.32/ubuntu bionic main)
Firefox  66.02
Ubuntu 18.04 (4.15.0-47-generic)

**Describe the bug**
There appears to be no setting to change path for event videos e.g /zm/ to /

**To Reproduce**
Attempting to setup zoneminder as a single virtual host on new install (with cameras setup and working)
have disabled zoneminder.conf (a2disconf zoneminder)
have disabled default site  (a2dissite 000-default.conf)
created and new site config zoneminder.conf (see attachment 
[zoneminder.conf.txt](https://github.com/ZoneMinder/zoneminder/files/3078317/zoneminder.conf.txt)
enabled new site (a2ensite zoneminder)

edit /etc/zm/conf.d/01-system-paths.conf
change ZM_PATH_ZMS=/zm/cgi-bin/nph-zms ZM_PATH_ZMS=/cgi-bin/nph-zms

Click an event link e.g (http://[fqdn]/index.php?view=event&eid=52&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Battr%5D=MonitorId&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bop%5D=%3D&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bval%5D=6&sort_field=StartTime&sort_asc=1&page=1)

get "The video could not be loaded, either because the server or network failed or because the format is not supported." error 

if inspect page can see video source is 
src="[fqdn]/zm/index.php?mode=mpeg&format=h264&eid=52&view=view_video&auth=[deleted]"

when should be 

src="[fqdn]/index.php?mode=mpeg&format=h264&eid=52&view=view_video&auth=[deleted]"

if edit page html in browser video is displayed

**Expected behavior**
a Path setting allow correct display of video

**Debug Logs**
```

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

connortechnology, 2019-04-15:
This has been resolved in master and will be released in 1.34. Until then it is a simple matter of apache config to get around.  Add an Alias for /zm.

closed 2019-04-15 by connortechnology

cassiel74, 2019-09-04:
Alias for /zm is there in my apache config but error persist

```
ScriptAlias /zm/cgi-bin "/usr/lib/zoneminder/cgi-bin"
<Directory "/usr/lib/zoneminder/cgi-bin">
    Options +ExecCGI -MultiViews +SymLinksIfOwnerMatch
    AllowOverride All
    Require all granted
</Directory>

    # Order matters. This alias must come first.
    Alias /zm/cache "/var/cache/zoneminder"
    <Directory "/var/cache/zoneminder">
        Options -Indexes +FollowSymLinks
        AllowOverride None
        <IfModule mod_authz_core.c>
           # Apache 2.4
           Require all granted
        </IfModule>
    </Directory>

Alias /zm /usr/share/zoneminder/www
<Directory /usr/share/zoneminder/www>
  Options -Indexes +FollowSymLinks
  <IfModule mod_dir.c>
    DirectoryIndex index.php
  </IfModule>
</Directory>

<Directory /usr/share/zoneminder/www/api>
    AllowOverride All
</Directory>
```