ZoneMinder/zoneminder #2578: NODECT linked camera record time does not match MODECT camera

bahbarnett, 2019-04-15:
**THIS FORUM IS FOR BUG REPORTS ONLY**

Do not post feature or enhancement requests, general discussions, or support questions here.

Feature and enhancement requests, general discussions, and support questions should occur in one of the following areas:

- The [ZoneMinder-Chat Slack channel](https://zoneminder-chat.herokuapp.com/)
- The [ZoneMinder Forum](https://forums.zoneminder.com/)

Docker related issues should be posted here: https://github.com/ZoneMinder/zmdockerfiles

In order to submit a bug report, please populate the fields below this line. This is required.

----------------------------------------------------------------------------------------------------

**Describe Your Environment**

Debian Unstable with package rev 1.32.3-2, Zoneminder v1.32.3.

**Describe the bug**

I am working with linked monitors.  Events do trigger correctly, the linked monitor triggers, and the secondary camera monitor is kicked off.

The issue is that event lengths aren't the same.  In the primary camera, setup with modect, I have the following on the buffers tab:

Image Buffer Size (frames) - 100
Warmup Frames - 10
Pre Event Image Count - 30	
Post Event Image Count	- 60
Stream Replay Image Buffer - 0
Alarm Frame Count - 4

As I used 'clone' to make the secondary NODECT camera feed, the above is identical for it.

Every event on MODECT camera, does indeed cause NODECT to record.  But the recording times are *different*.

I would expect one of the following to happen:

1) The recording time in seconds for NODECT, is identical to the length of recording time on MODECT, and that therefore buffers on the NODECT camera are ignored.

or

2) The trigger time for NODECT, is identical to trigger time on MODECT, plus buffers set for the NODECT camera.

Yet neither of the above is happening.

[image]
[image]

The attached images show the two cameras.  It seems to me that two things might be happening:

- it takes about 2 seconds to start recording on the NODECT camera?  And some frames to stop?
- ZM records the same number of FRAMES on each camera, and ignores all buffers/etc on NODECT

As NODECT and MODECT cameras might have different FPS, syncing frames won't really work for time...

Regardless, there doesn't seem to be any way to ensure that NODECT records very closely to the same time that MODECT does.

sgofferj, 2019-10-01:
I probably have the same issue, just that in my case, not only the length differs but some events aren't recorded at all...
[image]

Running 1.32.3 on docker

xztraz, 2020-01-17:
same here. nodect/modect linked cameras record different numbers of events. v1.32.3 on rh7
events on high resolution feed seem to miss out one second events for example

jangrewe, 2020-02-07:
Do you have the same frame rate for both the main- and the sub-stream? Total noob, but that would be my first guess.

xztraz, 2020-02-11:
v1.34.1

It's even weirder in newer version.

In Console it says 7(low rez) and 2(high rez) (day)

and if cklicking the day number it reports 10 events for low rez.!?

[image]

High Rez
[image]

Low Rez
[image]

connortechnology, 2020-02-11:
Your issue is unrelated.  You havn't set your timezone correctly.

xztraz, 2020-02-12:
the timezone for php was set an hour wrong. I fixed that.  doesn't explain why more than half of events missing thou?