ZoneMinder/zoneminder #2579: Deleting event deletes last event and not one clicked

kdhosting, 2019-04-17:
**Describe Your Environment**
- Version of ZoneMinder: 1.32.3
- How you installed ZoneMinder: PPA
- Full name and version of OS: Ubuntu 18.04
- Browser name and version (if this is an issue with the web interface):
Chrome

**Describe the bug**
When deleting events in the Feed page, only the top (last) event gets deleted regardless of which event is clicked. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Console'
2. Click on any camera
3. Scroll down to list of events
4. Click 'X'
5. Observer error

**Expected behavior**
That only the event that is intended to be delete, gets deleted.

connortechnology, 2019-05-10:
This has already been fixed in master.

gaaf, 2019-07-10:
I could not find the fix in the master branch, so I created a fix myself for use in 1.32.3. It applies to master too. See PR #2660

commit a6a2239e (2019-07-24): Fixes #2579 by putting the event Id into an attribute of the delete button.  Also uses a button instead of a link.  Also only does parent refresh if parent exists (#2662)
closed 2019-07-24 by connortechnology