ZoneMinder/zoneminder #2583: zma crashing when dereferecing null event pointer in Monitor::Analyse()

neillbell, 2019-04-18:
**Describe Your Environment**
- Version of ZoneMinder: 1.33.6 (1.33.6~20190416113530-bionic)
- How you installed ZoneMinder: ppa:iconnor/zoneminder-master
- Full name and version of OS: Ubuntu 18.04.02

**Describe the bug**
The zma process is often crashing in the Monitor::Analyse() function when a null event pointer is dereferenced. This happens with several different models of camera every 5-10 minutes during normal event capture . The cameras immediately recover.

I wasn't sure how to track this down using the existing back trace, so I hooked up [libbacktrace](https://github.com/ianlancetaylor/libbacktrace) to generate [zma_l1.log](https://github.com/ZoneMinder/zoneminder/files/3096291/zma_l1.log). This pointed to the following line:

https://github.com/ZoneMinder/zoneminder/blob/eb76cd87bbe959e6e8f334bd749f55a17fd06c67/src/zm_monitor.cpp#L1723

This call to Warning() is occasionally deferencing a null event pointer.

I modified the warning call to be:
```
              Warning("%s: %03d - event %" PRIu64 ", has exceeded desired section length. %d - %d = %d >= %d",
                      name, image_count, event ? event->Id() : 0,
                  timestamp->tv_sec, video_store_data->recording.tv_sec,
                  timestamp->tv_sec - video_store_data->recording.tv_sec,
                  section_length
                  );
```
This new code now produces the following: [zma_l2.log](https://github.com/ZoneMinder/zoneminder/files/3096293/zma_l2.log)

connortechnology, 2019-04-24:
I believe this was fixed with commit [https://github.com/ZoneMinder/zoneminder/commit/22adb243ec1d679fb85d55cbcf7cb1a48a61b1ed](url)

closed 2019-04-24 by connortechnology

neillbell, 2019-04-24:
When I click your link I get "No results matched your search".

connortechnology, 2019-04-25:
Ok let's try this: https://github.com/ZoneMinder/zoneminder/commit/22adb243ec1d679fb85d55cbcf7cb1a48a61b1ed