ZoneMinder/zoneminder #2588: Netcat ONVIF: Added support for "profile token"

redaco, 2019-04-29:
Sorry for template, my case does not really apply to it.

I was offered a camera (Wifi / FullHD / PTZ / SDCard / IR) that looks a lot like a Sawake, but there is no name on the camera or on the box. Maybe a clone.

ZoneMinder detected the image via OnVif without any problem, but none of the control scripts worked.

Android TinyCam could connect to it and also control it (pan and tilt). I sniffed the network and was able to change Netcat.pm so that ZoneMinder could control this camera.

The first problem was that Netcat.pm uses the "profile token" '000', while my camera needs 'IPCProfilesToken1'.

So I have a patch to offer that adds support for this "token profile" by the Control Device field. If the Control Device field is empty, then we use '000' (so the script still works for Netcat.pm users), but if the field is not empty we use it as a "token profile".

I'll do a pull request soon, you'll tell me what you think.

connortechnology, 2019-04-29:
Sounds good.  I suspect that we should be able to detect the correct value for this field in talking to the camera. Havn't had the time to read the onvif spec yet though.

redaco, 2019-04-29:
It's done: https://github.com/ZoneMinder/zoneminder/pull/2589

Trivial but functional patch.

Indeed it would be more elegant to detect the good value (Android TinyCam does). But without XML parser, not easy to do it cleanly.

What do you say ?

knight-of-ni, 2019-04-29:
My Netcat camera was picked up by the trash tuck, literally this morning, so I am no longer able to maintain the script. Feel free to modify it to suit your needs.

knight-of-ni, 2019-05-01:
closed via #2589

closed 2019-05-01 by knight-of-ni