ZoneMinder/zoneminder #2590: Netcat ONVIF: Adding authentication support

redaco, 2019-04-29:
Still for my camera I'm talking about in #2588, I had to add support for ONVIF authentication to Netcat.pm.

I chose to configure the login / password using the Control Address field. Previously, we could store something of the form address:port, now we can use login:password@address:port (classic format). It's questionable, it would probably be cleaner to add 2 fields in the interface. But at this point it was faster for me to implement it as well.

The old address format:port still works, of course.

As for #2588, I sniffed the network using Android TinyCam; I did not read the ONVIF specifications so I can not guarantee that it works with other cameras.

There are also some "essential" dependencies: MIME::Base64, Digest::SHA, DateTime.

Sounds ok? I make a pull request?

connortechnology, 2019-04-29:
Sure, make the PR and we will discuss it.

I like having the username/password in the address field.  It's how I went with other Control scripts.

redaco, 2019-04-30:
It's done, I'm waiting for your remarks.

knight-of-ni, 2019-05-01:
Closed via #2598 ...I think. If I am mistaken, please let me know and I will reopen this.

closed 2019-05-01 by knight-of-ni