ZoneMinder/zoneminder #2595: SQL syntax error when assigning an X10 Output Alarm String to monitor.

bartonrr, 2019-05-03:
- Version of ZoneMinder [1.30.0]
- How you installed ZoneMinder [aptitude]
- Full name and version of OS: Raspbian GNU/Linux 9
- Browser name and version (if this is an issue with the web interface)

A clear and concise description of what the bug is.
When adding an X10 Output Alarm String to a monitor, I get the following message:

019-05-02 20:34:37.420997 | web_php |   | 8537 | FAT | SQL-ERR  'SQLSTATE[42000]: Syntax error or access violation: 1064 You have an  error in your SQL syntax; check the manual that corresponds to your  MariaDB server version for the right syntax to use near 'Activation =  '', AlarmInput = '', AlarmOutput = '3'' at line 1', statement was  'insert into TriggersX10 set MonitorId = ?Activation = '', AlarmInput =  '', AlarmOutput = '3'' | includes/database.php | 115
-- | -- | -- | -- | -- | -- | -- | --

I fixed the problem by changing line 618 in actions.php to read as follows:

dbQuery( "insert into TriggersX10 set MonitorId = ?, ".implode( ", ", $x10Changes ), array( $mid ) );

by adding the ', ' after the 'MonitorId = ?'

closed 2019-05-03 by close-issue-app[bot]

connortechnology, 2019-05-03:
K, don';t worry about the stupid closing issue bits.  I'll get to bottom of this. Thanks for your bug report!

bartonrr, 2019-05-03:
I do apologize for not following the template.  I got lazy since I had diagnosed the problem and provided a fix.

connortechnology, 2019-05-06:
This is already fixed in 1.32 and master.