ZoneMinder/zoneminder #2599: zmtrigger showtext missing from h264 video caption during event playback

j-marz, 2019-05-05:
**Describe Your Environment**
- v1.32.3
- iconnor PPA http://ppa.launchpad.net/iconnor/zoneminder-1.32/ubuntu
- Ubuntu 16.04.6 LTS 4.4.0-146-generic
- Firefox 66.0.3 (64-bit)

**Describe the bug**
The `%Q` option which is supposed to display the `showtext` from a zmtrigger event isn't displayed in the video caption during H264 video playback. Instead, `%2` is always displayed.

`showtext` is displayed in the live monitor view during the event, but not shown in the historical event during browser video playback.

Example caption:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Configure monitor to use FFMPEG with H264 camera passthrough for video writer
2. Configure monitor to display the `showtext` from zmtrigger via the timestamp label format option `%N - %d/%m/%y %H:%M:%S - %Q`
3. Trigger an alarm via zmtrigger with `showtext` populated
4. Playback event video in browser and observe the video caption

**Expected behavior**
Display the actual `showtext` in the video caption instead of `%2`.

connortechnology, 2019-05-06:
The showtext code is not well developed.  It is not saved with the event.  It should be.  It appears to only be used when timestamping the image, which we don't do in passthrough mode. I think we should append it to the event cause.

j-marz, 2019-07-06:
Comment to avoid auto closure due to no activity.

SteveGilvarry, 2026-06-13:
Revisiting this. A status update and a question.

**Current state:** `showtext` (`%Q`) is only ever consumed by the timestamp **overlay** that gets drawn onto a decoded image. Unlike the trigger's `cause` (→ event Cause) and `text` (→ event Notes), `showtext` is **not stored with the event** anywhere — once the trigger clears, it's gone.

**Why the original report can't be addressed as-is:** in H264 passthrough recording ZoneMinder copies the raw camera stream without decoding/re-encoding, so it never draws the timestamp overlay — there's no way to burn `%Q` (or any caption) into a passthrough video without re-encoding it, which defeats the purpose of passthrough. As passthrough becomes the common setup, the overlay path is shrinking, so "show it in the playback video" isn't a realistic target.

**A possibly better direction — Tags.** Tags attach to an event and display on the event/playback page, and they work regardless of passthrough because they're metadata, not pixels. The catch is that triggers fire *before* the event exists, so there's currently no path from a trigger to a tag the way there is for `cause`/`text`. Extending the trigger→event flow to set a tag (in the event-creation path, where the event id exists) could satisfy the underlying need as searchable/filterable metadata rather than an overlay.

**Before anyone builds that — can you confirm the actual use case?** What information are you trying to surface per event via `showtext`, and how do you want to *consume* it: glance at it during playback, search/filter events by it, include it in a notification/export, something else? That determines whether tags, the event Notes, or the Cause is the right home — and whether this is worth extending triggers for at all.