ZoneMinder/zoneminder #2606: "Blocked loading mixed active content" behind reverse proxy

jtickle, 2019-05-14:
**Describe Your Environment**
I just upgraded to ZM 1.32 release version. Built my own Docker image using the zmdockerfiles repo and used the Ubuntu option. Issue experienced in Firefox 66.0.5

**If the issue concerns a camera**
Does not concern a camera

**Describe the bug**
ZoneMinder appears to be functioning properly but I am not sure if I am missing something important. This error shows up in my web console:

Blocked loading mixed active content “http://redacted/zm/index.php?view=request&request=stream&connkey=948391&command=99”

The Docker container is serving on 80 HTTP, but my RP is serving 443 HTTPS. I have tracked the bug down to this line:

web/includes/Server.php:65:
return (!empty($_SERVER['HTTPS']) && $_SERVER['HTTPS'] !== 'off' || $_SERVER['SERVER_PORT'] == 443) ? 'https' : 'http';

This logic is different from the logic in index.php which seems to properly detect the HTTP_X_FORWARDED_PROTO header which I am setting in nginx. It appears to be duplicated code and I think the line in Server.php should be trying to read from the global ZM_BASE_PROTOCOL.

I would consider submitting a patch but don't have a dev environment set up and won't have time this week or next.

Just for fun I also grepped the source for the string ['HTTPS'] and there appears to be another duplication of logic in web/api/lib/Cake/basics.php around line 323, so you may want to check that out while looking into this.

**To Reproduce**
Steps to reproduce the behavior:
1. Run Zoneminder Docker container built from Ubuntu Dockerfile in zmdockerfiles
2. View a monitor in the web browser
3. Open your dev tools
4. See error repeated approximately 1/second

**Expected behavior**
All areas where a URL is generated should be based on calculated ZM_BASE_PROTOCOL which is correctly set

**Debug Logs**
```

Blocked loading mixed active content “http://redacted/zm/index.php?view=request&request=stream&connkey=948391&command=99”[Learn More] mootools-core.js:1:85965
XMLHttpRequest
​
mozAnon: false
​
mozSystem: false
​
onabort: null
​
onerror: null
​
onload: null
​
onloadend: null
​
onloadstart: function d()​
onprogress: function d()​
onreadystatechange: function d()
​
ontimeout: null
​
readyState: 4
​
response: ""
​
responseText: ""
​
responseType: ""
​
responseURL: ""
​
responseXML: null
​
status: 0
​
statusText: ""
​
timeout: 0
​
upload: XMLHttpRequestUpload { onloadstart: null, onprogress: null, onabort: null, … }
​
withCredentials: false
​
<prototype>: XMLHttpRequestPrototype { open: open(), setRequestHeader: setRequestHeader(), send: send(), … }
skins_classic_views_js_watch-base-1544285133.js:132:3
Detected streamWatch of type: stream stopped, restarting

```

james-choncholas, 2019-07-18:
Same issue running behind traefik reverse proxy

redlegoman, 2019-09-17:
I have this exact same problem. If I change the line from 
` ( isset($_SERVER['HTTP_X_FORWARDED_PROTO']) and ( $_SERVER['HTTP_X_FORWARDED_PROTO'] == 'https' ) )
            ) ? 'https' : 'http';`
to
` ( isset($_SERVER['HTTP_X_FORWARDED_PROTO']) and ( $_SERVER['HTTP_X_FORWARDED_PROTO'] == 'https' ) )
            ) ? 'https' : 'https';`
It works via reverse proxy on https, but breaks my local connection which I make on http. Anybody suggest the correct code to make it work?

connortechnology, 2019-11-04:
Please test latest master.  A recent commit may have fixed this.

PrplHaz4, 2020-01-10:
@connortechnology can you point me to the commit?  if possible a monkey patch is quicker to test than standing up a dev environment...

PrplHaz4, 2020-03-17:
This looks to be fixed somewhere in the 1.34.0-6 releases.

eddy-dot, 2020-05-20:
I have the same issue behind aws elb and traefik

closed 2021-03-10 by stale[bot]

MacGyver27, 2021-06-17:
Same issue.. anybody here was able to address it? My version is: v1.34.26.
Thanks