ZoneMinder/zoneminder #2610: Web GUI run state always showing as 'Stopped', regardless of actual state

mjnewton, 2019-05-15:
**Describe Your Environment**
- Version 1.32.3
- PPA (Issac Connor's)
- Ubuntu Bionic 18.04.2 LTS
- Chrome 74.0.3729.131 (Official Build) (64-bit)

**If the issue concerns a camera**
- N/A

**Describe the bug**
The web GUI run state is always showing 'STOPPED' regardless whether Zoneminder is running or not. A consequence of this, aside from the misleading run state information, is that the default 'Stop' and 'Restart' run states are not available in the state-change drop-down menu and thus unable to be enacted.

**To Reproduce**
The issue appears continuously and requires no trigger to reproduce.

**Expected behavior**
The run state indicator should reflect the true state of the system and thus, in turn, present the appropriate run state option in the state-change drop down menu (i.e. 'Start' when 'STOPPED', and 'Stop'/'Restart' when 'RUNNING').

**Debug Logs**
Nothing of not include. Happy to provide a full log if there's a liklihood I may have missed something relevant.

mjnewton, 2019-05-16:
For info I've started a discussion [here](https://forums.zoneminder.com/viewtopic.php?f=38&t=28288&p=110488)

ijarraki, 2019-05-20:
Hey, this was fixed in 1.33, time to move on

ijarraki, 2019-05-20:
1.32.3 has serious flaws all over. Not whats is being worked on now. Try master

mjnewton, 2019-05-20:
I thought 1.32.3 was the latest production release? (1.33 being development only?)

Do you have a link to the fix reference for 1.33?

closed 2019-06-03 by connortechnology

connortechnology, 2020-03-06:
So the code that determines the run state calls 
/usr/bin/zmdc.pl check
as the web user.  You can try it from a command line as root, and then maybe try it as the web user as follows:
sudo su -s/bin/bash -c"/usr/bin/zmdc.pl check" www-data

and see what it says.  

if it says stopped both times then it is unable to talk to the running zmdc.pl process.  Is there a running zmdc.pl process ?  ps ax | grep zmdc

we talk to the running zmdc process through a socket located at /run/zm/zmdc.sock .  If this is a multi-server situation then it will be zmdc1.sock or zmdc2.sock or something like that.  So take a look in /run/zm.

mjnewton, 2020-03-06:
Thanks for the comments and suggestions Isaac.

Running zmdc.pl show the state as running:

```
# /usr/bin/zmdc.pl check
running
# sudo su -s/bin/bash -c "/usr/bin/zmdc.pl check" www-data
running
```

and the other command outputs as follows:

```
# ps ax |grep zmdc
  869 pts/0    S+     0:00 grep --color=auto zmdc
23321 ?        S      0:04 /usr/bin/perl -wT /usr/bin/zmdc.pl startup

# ls -l /run/zm
total 8
-rw-r--r-- 1 www-data www-data 5 Mar  6 00:23 zmaudit.pid
-rw-r--r-- 1 www-data www-data 5 Mar  6 00:23 zm.pid
```

(I can confirm the GUI is still showing Stopped at this point in time as per the original screenshot)

connortechnology, 2020-03-07:
This doesn't make sense.  There should be a zmdc.sock in /run/zm/

Is there an 03custom.conf in /etc/zm/conf.d?  What are it's contents? Is there a value for ZM_PATH_SOCKS?  can you do a 
grep ZM_PATH_SOCKS /etc/zm.conf /etc/conf.d/*.conf

mjnewton, 2020-03-07:
Thanks so much for your perseverance with this... and I am pleased to say you led me to the problem:

Here is my /etc/zm/conf.d/ :

```
# ls -l /etc/zm/conf.d/
total 24
-rw-r--r-- 1 root root     1726 Feb 26 20:42 01-system-paths.conf
-rw-r--r-- 1 root root      395 Oct 14  2018 02-multiserver.conf
-rw-r--r-- 1 root root      506 Dec  5  2018 mysettings.conf
-rw-r--r-- 1 root root      514 Dec  5  2018 mysettings.conf~
-rw-r--r-- 1 root root      740 Oct 14  2018 README
-rw-r----- 1 root www-data  139 Dec  5  2018 zmcustom.conf
```

Whilst there is no ZM_PATH_SOCKS in /etc/zm/zm.conf there is in /etc/zm/conf.d/01-system-paths.conf:

```
# grep ZM_PATH_SOCKS /etc/zm/conf.d/01-system-paths.conf 
ZM_PATH_SOCKS=/var/run/zm
```

However, it is **also** present in /etc/zm/conf.d/zmcustom.conf:

```
# more /etc/zm/conf.d/zmcustom.conf
# These values were autogenerated by zmupdate.pl
# You may edit these values. ZoneMinder will not overwrite them.
#

ZM_PATH_SOCKS=/tmp/zm
```

Removing zmcustom.conf and restarting has restored the GUI to working properly:

[image]

I am pretty sure I hadn't manually edited the zmcustom.conf file as I tend to annotate any changes I make, and there was no zmcustom.conf~ file that emacs would've created if I had, and so I guess it must've come from a previous install. The file date does coincide with when I upgraded from v1.30.4 to v1.32.3.

Thanks again for your help in solving this one for me.

connortechnology, 2020-03-08:
In the upgrade to 1.32 we auto generated that file to contain the settings that differed from the defaults.  So that's how it got there.

mjnewton, 2020-03-08:
Ah, okay. My original installation (many, many years ago) was from another repository (not your PPA) so I guess it may have originated from there.