ZoneMinder/zoneminder #2613: bulk frames playback is broken

bahbarnett, 2019-05-22:
----------------------------------------------------------------------------------------------------

**Describe Your Environment**
Zoneminder v1.32.3 from Debian Unstable 1.32.3-2.
Recent upgrade from v1.25

**If the issue concerns a camera**
Not camera related

**Describe the bug**
Bulk frames playback does not seem to correctly see events frames on disk.  Instead, it plays events from other, random events... or nothing at all.

I'm using MOCORD, and as designed zoneminder records all frames, even if bulk = 100, and no motion is detected.  EG:
```
MariaDB [zm]> select * from Frames where EventId=99054;
+---------+---------+---------+--------+---------------------+--------+-------+
| Id      | EventId | FrameId | Type   | TimeStamp           | Delta  | Score |
+---------+---------+---------+--------+---------------------+--------+-------+
| 7766191 |   99054 |     100 | Bulk   | 2019-05-22 03:50:09 |   9.89 |     0 |
| 7766192 |   99054 |     200 | Bulk   | 2019-05-22 03:50:19 |  19.89 |     0 |
| 7766194 |   99054 |     300 | Bulk   | 2019-05-22 03:50:29 |  29.88 |     0 |
....

root@zoneminder:/var/cache/zoneminder/events/11/2019-05-22/99054# ls -l -t -r | more
total 794524
-rw-r--r-- 1 www-data www-data 134244 May 22 03:50 001-capture.jpg
-rw-r--r-- 1 www-data www-data 134379 May 22 03:50 002-capture.jpg
-rw-r--r-- 1 www-data www-data 134531 May 22 03:50 003-capture.jpg
-rw-r--r-- 1 www-data www-data 134545 May 22 03:50 004-capture.jpg
-rw-r--r-- 1 www-data www-data 134652 May 22 03:50 005-capture.jpg
-rw-r--r-- 1 www-data www-data 134806 May 22 03:50 006-capture.jpg
-rw-r--r-- 1 www-data www-data 134835 May 22 03:50 007-capture.jpg
-rw-r--r-- 1 www-data www-data 134944 May 22 03:50 008-capture.jpg
-rw-r--r-- 1 www-data www-data 134810 May 22 03:50 009-capture.jpg
-rw-r--r-- 1 www-data www-data 134989 May 22 03:50 010-capture.jpg
-rw-r--r-- 1 www-data www-data 135052 May 22 03:50 011-capture.jpg
```

So all is well in terms of recording data.  Note that this is precisely how 1.25 worked as well.  When examining older records with the same config under 1.25.. they are recorded the same.

DB table with 'bulk', yet frames written to disk when 'MOCORD' is used.

However, during playback 1.32.3 zoneminder seems to ignore on-disk data, and play back "random stuff" instead.  I've noticed this trying to work with pre-upgrade events, and post-upgrade events.

For example with the above event... should I click on 'frames' and then on any frame listed, I get to see that frame.  Yet if I am on a 'bulk' frame, and click 'next', I am shown an event from an entirely different camera.  One a decade ago.

And if I try to play back events with bulk frames, it won't play past the first frame.

I've checked, and it isn't a DB table issue.  EG, there don't seem to be overlapping events.

I've also set logging to 'debug' and 'info', and in each case no error was presented.

To verify this does seem to be a bulk frames issue, I set bulk frames to 0, noted that all frames were recorded in the DB, and playback worked fine at that point.

**To Reproduce**
Steps to reproduce the behavior:
1. Set bulk frames to 100
2. Set camera to 'MOCORD'
3. Note that all frames are being stored in the filesystem, database is storing BULK as frame type every 100 frames (expected)
4. Try to play back, note that ZM will play back random frames, or no frames, or can't play back at all

**Expected behavior**
ZM should play back all frames correctly.

Note that pre-upgrade, all was fine. 

**Debug Logs**
No error logs were found, see above for data
I checked into what's happening via the view... so some info:

```
Path of 'good image' shown: /zm/index.php?view=image&fid=15&eid=98691&show=capture
Path of 'bad image' shown:   /zm/index.php?view=image&fid=16&eid=&show=capture
```
So for some reason, 'eid' is dropped before being passed to the image view, when trying to view a bulk frame (eg, image on disk, bulk frames = 100, DB shows no actual frame 101).

Yet the URL/etc does seem correct, with:

```
/zm/index.php?view=frame&eid=98691&fid=16&scale=100&show=capt
```

For quick debugging, I edited /usr/share/zoneminder/www/skins/classic/views/frame.php and slapped @ line 77:
```
$imagePath = $imageData['thumbPath'];
echo "<BR>2: $imagePath<BR>";
```
And the correct image is shown for "imagePath", but not displayed.

So, I dumpted $this in zoneminder/www/includes/Frame.php
```
  public function getImageSrc( $show='capture' ) {
    error_log(print_r($this,1));
```
And got (for a good and bad image):

```
[22-May-2019 05:32:19 US/Eastern] Frame Object
(
    [Id] => 7724756
    [EventId] => 98691
    [FrameId] => 15
    [Type] => Normal
    [TimeStamp] => 2019-05-20 00:30:04
    [Delta] => 4.50
    [Score] => 0
)

[22-May-2019 05:32:21 US/Eastern] Frame Object
(
    [FrameId] => 16
    [Type] => Normal
    [Score] => 0
)

Back in the view, I noticed this @ lined 36:

if ( !empty($fid) ) {
  $sql = 'SELECT * FROM Frames WHERE EventId = ? AND FrameId = ?';
  if ( !($frame = dbFetchOne( $sql, NULL, array($eid, $fid) )) )
    $frame = array( 'FrameId'=>$fid, 'Type'=>'Normal', 'Score'=>0 );
} else {
  $frame = dbFetchOne( 'SELECT * FROM Frames WHERE EventId = ? AND Score = ?', NULL, array( $eid, $Event->MaxScore() ) );
}
$Frame = new Frame($frame);
```

I'm not arm deep into this code, but from where I sit the first SQL will always fail when trying to display a bulk frame.  It looks like the second is simply trying to get a MaxScore by a column sort (my grep shows that in functions.php?), so it makes sense sort of that this borks.

 I haven't been able to track anything down further .. I'm a bit strapped for time here.  Any suggestions?

bahbarnett, 2019-05-22:
OK.. to just "make it work" for now, I did:

```
views/frame.php @ 105 changed from
<img id="frameImg" src="<?php echo $Frame->getImageSrc($show=='anal'?'analyse':'capture') ?>" width="<?php echo reScale( $Event->Width(), $Event->DefaultScale(), $scale ) ?>" height="<?php echo reScale( $Event->Height(), $Event->DefaultScale(), $scale ) ?>" alt="<?php echo $Frame->EventId()."-".$Frame->FrameId() ?>" class="<?php echo $imageData['imageClass'] ?>"/>
to:
<img id="frameImg" src="<?php echo $Frame->getImageSrc($show=='anal'?'analyse':'capture',$eid,$fid) ?>" width="<?php echo reScale( $Event->Width(), $Event->DefaultScale(), $scale ) ?>" height="<?php echo reScale( $Event->Height(), $Event->DefaultScale(), $scale ) ?>" alt="<?php echo $Frame->EventId()."-".$Frame->FrameId() ?>" class="<?php echo $imageData['imageClass'] ?>"/>
```

I then changed include/Frame.php @ 52:

```
from:
  public function getImageSrc( $show='capture' ) {

    return $_SERVER['PHP_SELF'].'?view=image&fid='.$this->{'FrameId'}.'&eid='.$this->{'EventId'}.'&show='.$show;

to:
  public function getImageSrc( $show='capture',$eid=0,$fid=0 ) {
    if ($eid == 0 || $fid == 0) {$eid=$this->{'EventId'};$fid=$this->{'FrameId'};}
    error_log(print_r($this,1));
    return $_SERVER['PHP_SELF'].'?view=image&fid='.$fid.'&eid='.$eid.'&show='.$show;
```

This replicates the current behaviour, unless eid and fid are both set to something when calling getImageSrc.  "WorksForMe", definitely hacky, but OK my current usage..

bahbarnett, 2019-05-22:
Hmm, playback is still broken though.  More code to poke at.  Again, ideas here?

bahbarnett, 2019-05-22:
Note: my initial report says I have issues with playback + bulk frames -- I do not.  Only still frames as noted above had issues.

connortechnology, 2019-05-24:
I do not think that this problem affects current master.

closed 2019-07-30 by stale[bot]