ZoneMinder/zoneminder #2615: ZM allows currently recording events to be deleted

bahbarnett, 2019-05-23:
**Describe Your Environment**
Zoneminder v1.32.3 from Debian Unstable 1.32.3-2.

**If the issue concerns a camera**
Not camera related

**Describe the bug**

When deleting events, you can check and delete a currently recording event.  This should not be permitted, as the following happens:
```
Can't open /var/cache/zoneminder/events/10/2019-05-23/99510/953-capture.jpg: No such file or directory
```

**To Reproduce**
Steps to reproduce the behavior:
1. Pick a montior/camera, click on 'events'
2. Select events to delete.. including currently recording event (MOCORD)
3. Current event is deleted, now ZM can't write MOCORD frames to disk

**Expected behavior**
ZM should not remove events that are recording

**Debug Logs**
```
Can't open /var/cache/zoneminder/events/10/2019-05-23/99510/953-capture.jpg: No such file or directory
```

bahbarnett, 2019-05-23:
Just realised an additional condition -- FAST_DELETE must be off...

connortechnology, 2019-05-24:
Fixed by 2fc7b3cf4d69e25fa68d190dd6b052d7fef94077

closed 2019-05-24 by connortechnology