ZoneMinder/zoneminder #2621: Audio duration is longer than video duration when using MP2 audio with H264 camera passthrough

neillbell, 2019-05-27:
**Describe Your Environment**
- Version of ZoneMinder: 1.33.9~20190524141652-bionic
- How you installed ZoneMinder: ppa:iconnor/zoneminder-master
- Full name and version of OS: Ubuntu 18.04.02 LTS

**If the issue concerns a camera**
- Hikvision DS-2CD2442FWD-IW
- 10 FPS
- 1920x1080
- Audio selected as: MP2L2, Sampling rate 16kHz, Stream bitrate 64kbps
- ZoneMinder Source Type: FFMPEG, H264 Camera Passthrough, audio storage enabled

**Describe the bug**
When using MP2 audio, the audio is properly transcoded to AAC, but the audio duration is much larger than the video duration. This was fixed before in issue #2572, but has reappeared.

**Debug Logs**
Here is a log for the camera at debug level 3.
[zmc_m9.log](https://github.com/ZoneMinder/zoneminder/files/3224368/zmc_m9.log)

**Mediainfo output**
```
General
Complete name                            : /var/cache/zoneminder/events/9/2019-05-27/258053/258053-video.mp4
Format                                   : MPEG-4
Format profile                           : Base Media
Codec ID                                 : isom (isom/iso2/avc1/mp41)
File size                                : 2.54 MiB
Duration                                 : 41 s 438 ms
Overall bit rate mode                    : Variable
Overall bit rate                         : 513 kb/s
Movie name                               : Zoneminder Security Recording
Writing application                      : Lavf57.83.100

Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : High@L4.1
Format settings                          : CABAC / 1 Ref Frames
Format settings, CABAC                   : Yes
Format settings, ReFrames                : 1 frame
Codec ID                                 : avc1
Codec ID/Info                            : Advanced Video Coding
Duration                                 : 7 s 400 ms
Bit rate mode                            : Variable
Bit rate                                 : 2 842 kb/s
Maximum bit rate                         : 4 910 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate mode                          : Constant
Frame rate                               : 10.000 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 0.137
Stream size                              : 2.51 MiB (99%)
Color range                              : Full
Color primaries                          : BT.709
Transfer characteristics                 : BT.709
Matrix coefficients                      : BT.709

Audio
ID                                       : 2
Format                                   : AAC
Format/Info                              : Advanced Audio Codec
Format profile                           : LC
Codec ID                                 : mp4a-40-2
Duration                                 : 41 s 438 ms
Duration_FirstFrame                      : -341 ms
Duration_LastFrame                       : -341 ms
Bit rate mode                            : Variable
Bit rate                                 : 5 310 b/s
Maximum bit rate                         : 32.8 kb/s
Channel(s)                               : 2 channels
Channel(s)_Original                      : 1 channel
Channel positions                        : Front: C
Sampling rate                            : 16.0 kHz
Frame rate                               : 15.625 FPS (1024 SPF)
Compression mode                         : Lossy
Stream size                              : 26.9 KiB (1%)
Default                                  : Yes
Alternate group                          : 1

```

connortechnology, 2019-05-27:
I may have fixed this, can you test again with latest master?

neillbell, 2019-05-27:
I still see 1.33.9~20190524141652-bionic as the latest version on the PPA and that's what I'm using.

I noticed that the xenial & cosmic builds have been updated on the PPA.

neillbell, 2019-05-28:
I see that all the builds except bionic have been updated to 20190527171713. What gives?

connortechnology, 2019-05-28:
New packages are building now.

neillbell, 2019-05-28:
It seems to be working better now, but the video duration seems to be too short. Here's a log at debug 3:
[zmc_m9.log](https://github.com/ZoneMinder/zoneminder/files/3229938/zmc_m9.log)

Here is the mediainfo output. When played, the video seems to be the duration of the audio, but the video duration from mediainfo seems to be too short:
```
General
Complete name                            : /var/cache/zoneminder/events/9/2019-05-28/259735/259735-video.mp4
Format                                   : MPEG-4
Format profile                           : Base Media
Codec ID                                 : isom (isom/iso2/avc1/mp41)
File size                                : 12.7 MiB
Duration                                 : 50 s 240 ms
Overall bit rate mode                    : Variable
Overall bit rate                         : 2 120 kb/s
Movie name                               : Zoneminder Security Recording
Writing application                      : Lavf57.83.100

Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : High@L4.1
Format settings                          : CABAC / 1 Ref Frames
Format settings, CABAC                   : Yes
Format settings, ReFrames                : 1 frame
Format settings, GOP                     : M=1, N=20
Codec ID                                 : avc1
Codec ID/Info                            : Advanced Video Coding
Duration                                 : 36 s 200 ms
Bit rate mode                            : Variable
Bit rate                                 : 2 899 kb/s
Maximum bit rate                         : 4 910 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate mode                          : Constant
Frame rate                               : 10.000 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 0.140
Stream size                              : 12.5 MiB (99%)
Color range                              : Full
Color primaries                          : BT.709
Transfer characteristics                 : BT.709
Matrix coefficients                      : BT.709

Audio
ID                                       : 2
Format                                   : AAC
Format/Info                              : Advanced Audio Codec
Format profile                           : LC
Codec ID                                 : mp4a-40-2
Duration                                 : 50 s 240 ms
Duration_FirstFrame                      : -8 ms
Duration_LastFrame                       : -8 ms
Bit rate mode                            : Variable
Bit rate                                 : 29.2 kb/s
Maximum bit rate                         : 32.8 kb/s
Channel(s)                               : 2 channels
Channel(s)_Original                      : 1 channel
Channel positions                        : Front: C
Sampling rate                            : 16.0 kHz
Frame rate                               : 15.625 FPS (1024 SPF)
Compression mode                         : Lossy
Stream size                              : 179 KiB (1%)
Default                                  : Yes
Alternate group                          : 1

```

neillbell, 2019-06-03:
I thought you might like to know that this problem is still present. It occurs for both MP2 and G.726 audio.

Here is a debug log and mediainfo output for a camera with G.726 audio:
[zmc_m5.log](https://github.com/ZoneMinder/zoneminder/files/3249255/zmc_m5.log)
```
General
Complete name                            : 266275-video.mp4
Format                                   : MPEG-4
Format profile                           : Base Media
Codec ID                                 : isom (isom/iso2/avc1/mp41)
File size                                : 2.77 MiB
Duration                                 : 18 s 436 ms
Overall bit rate                         : 1 261 kb/s
Movie name                               : Zoneminder Security Recording
Writing application                      : Lavf57.83.100

Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : Main@L4.2
Format settings                          : CABAC / 1 Ref Frames
Format settings, CABAC                   : Yes
Format settings, ReFrames                : 1 frame
Format settings, GOP                     : M=1, N=10
Codec ID                                 : avc1
Codec ID/Info                            : Advanced Video Coding
Duration                                 : 9 s 400 ms
Bit rate                                 : 2 386 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate mode                          : Constant
Frame rate                               : 10.000 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 0.115
Stream size                              : 2.73 MiB (99%)
Color range                              : Full
Color primaries                          : BT.709
Transfer characteristics                 : BT.709
Matrix coefficients                      : BT.709

Audio
ID                                       : 2
Format                                   : AAC
Format/Info                              : Advanced Audio Codec
Format profile                           : LC
Codec ID                                 : mp4a-40-2
Duration                                 : 18 s 436 ms
Bit rate mode                            : Constant
Bit rate                                 : 16.0 kb/s
Channel(s)                               : 2 channels
Channel(s)_Original                      : 1 channel
Channel positions                        : Front: C
Sampling rate                            : 8 000 Hz
Frame rate                               : 7.812 FPS (1024 SPF)
Compression mode                         : Lossy
Stream size                              : 36.1 KiB (1%)
Default                                  : Yes
Alternate group                          : 1
```

Here is a debug log and mediainfo output for a camera with MP2 audio:
[zmc_m9.log](https://github.com/ZoneMinder/zoneminder/files/3249261/zmc_m9.log)
```
General
Complete name                            : 266276-video.mp4
[... 38 lines cut ...]
Transfer characteristics                 : BT.709
Matrix coefficients                      : BT.709

Audio
ID                                       : 2
Format                                   : AAC
Format/Info                              : Advanced Audio Codec
Format profile                           : LC
Codec ID                                 : mp4a-40-2
Duration                                 : 29 s 72 ms
Duration_FirstFrame                      : -8 ms
Duration_LastFrame                       : -8 ms
Bit rate mode                            : Variable
Bit rate                                 : 29.2 kb/s
Maximum bit rate                         : 32.8 kb/s
Channel(s)                               : 2 channels
Channel(s)_Original                      : 1 channel
Channel positions                        : Front: C
Sampling rate                            : 16.0 kHz
Frame rate                               : 15.625 FPS (1024 SPF)
Compression mode                         : Lossy
Stream size                              : 104 KiB (2%)
Default                                  : Yes
Alternate group                          : 1
```

neillbell, 2019-06-05:
This is odd. I just realized that what is going on isn't what I thought. It appears that the audio length is correct. When I watch the videos, the video and audio play appropriately at first, then the video stops and the audio continues normally for the length specified in the audio portion.

Previously, the duration for the audio was only **marked** longer than the video. Now the audio actually **is** longer than the video.

connortechnology, 2019-06-05:
Does the audio sound ok?  I wonder if we are slowing it down...

neillbell, 2019-06-05:
I justed tested it with my camera that does G.726 and the video is definitely slowed down. Strangely, it doesn't seem slowed down enough to account for the duration difference. I did a test where I walked in front on the camera while chanting. On the test the video was 7 s 900 ms and you could see the movement at the start and it quit after 7 seconds, but the audio duration was 8 min 10 s and you could only hear my chant, slowed down, at the very end of the audio.

connortechnology, 2019-06-05:
Ok, so we are screwing up the resampling... making it slower than it should be. I'll take a look and try to figure it out today.

neillbell, 2019-06-05:
I take it back. I tested again and what I thought was slowing down was just distortion.

neillbell, 2019-06-05:
I also noticed that the difference between the audio and video duration recorded in the files greatly decreases if I restart the camera by moving them from "modect" to "none" and back again.

connortechnology, 2019-06-05:
ok that might make more sense if I was just screwing up the timing.

connortechnology, 2019-06-05:
Hmm input audio frame has a duration of 6480... out frame has a duration of 1152... I think that is incorrect.

neillbell, 2019-06-05:
OK. Here is a test i made as follows:
1. Turn on debug level 3.
2. Set the camera to "None"
3. Shut down ZoneMinder
4. Truncate the zmc_m?.log file
5. Start up ZoneMinder
6. Get next to the camera and set it to "Modect" using zmNinja.
7. Wait 10 seconds then tap on something in view of the camera.
8. Shut down ZoneMinder and copy the log and video file

The audio was 8 seconds delayed from the video, was of normal speed, not distorted and the audio was 8 seconds longer than the video.
[zmc_m9.log](https://github.com/ZoneMinder/zoneminder/files/3258326/zmc_m9.log)
```
General
Complete name                            : 268238-video.mp4
Format                                   : MPEG-4
Format profile                           : Base Media
Codec ID                                 : isom (isom/iso2/avc1/mp41)
File size                                : 4.53 MiB
Duration                                 : 20 s 720 ms
Overall bit rate mode                    : Variable
Overall bit rate                         : 1 833 kb/s
Movie name                               : Zoneminder Security Recording
Writing application                      : Lavf57.83.100

Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : High@L4.1
Format settings                          : CABAC / 1 Ref Frames
Format settings, CABAC                   : Yes
Format settings, ReFrames                : 1 frame
Format settings, GOP                     : M=1, N=20
Codec ID                                 : avc1
Codec ID/Info                            : Advanced Video Coding
Duration                                 : 12 s 600 ms
Bit rate mode                            : Variable
Bit rate                                 : 2 963 kb/s
Maximum bit rate                         : 4 910 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate mode                          : Constant
Frame rate                               : 10.000 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 0.143
Stream size                              : 4.45 MiB (98%)
Color range                              : Full
Color primaries                          : BT.709
Transfer characteristics                 : BT.709
Matrix coefficients                      : BT.709

Audio
ID                                       : 2
Format                                   : AAC
Format/Info                              : Advanced Audio Codec
Format profile                           : LC
Codec ID                                 : mp4a-40-2
Duration                                 : 20 s 720 ms
Duration_FirstFrame                      : -8 ms
Duration_LastFrame                       : -8 ms
Bit rate mode                            : Variable
Bit rate                                 : 29.3 kb/s
Maximum bit rate                         : 32.8 kb/s
Channel(s)                               : 2 channels
Channel(s)_Original                      : 1 channel
Channel positions                        : Front: C
Sampling rate                            : 16.0 kHz
Frame rate                               : 15.625 FPS (1024 SPF)
Compression mode                         : Lossy
Stream size                              : 74.0 KiB (2%)
Default                                  : Yes
Alternate group                          : 1
```

connortechnology, 2019-06-05:
I think I'm doing something stupid... First off, I think I am dropping samples.  Which I shouldn't have to do.  Also seem to be doing something stupid with the timestamps.  Going to have to think hard on this.

connortechnology, 2019-06-05:
I don't think resampling is even needed here.

neillbell, 2019-06-05:
I thought you had to since the MP2 frame size is 1152 and the AAC frame size is 1024.

connortechnology, 2019-06-10:
I finally understand what is happening here. Fix should come soon.

connortechnology, 2019-06-11:
I believe this is fixed.  Please test & verify.

neillbell, 2019-06-12:
This appears to be fixed 100% for G.726 audio. With MP2 audio, however, something odd is going on. When I play the files back using Chrome, Chromium or zmNinja, they work fine. When I play the files back using anything else, Firefox, mpv, mplayer, vlc,  Windows Media Player, the audio is delayed by a fraction of a second. Strangely enough Chrome works correctly on both Linux and Windows.

What could be going on, and how could I help you with it?

connortechnology, 2019-06-12:
I can't imagine why that would be, but I'm seeing some other oddities and things I may have missed.  Stay tuned.

connortechnology, 2019-06-20:
Can you update and report if it's still doing it?  I believe I have improved the situation.

neillbell, 2019-06-21:
Definitely not better. This one is going to be hard to characterize because it seems to behave strangely depending on how long the camera has been up or whether its MP2 or G726 or something. Things play differently depending on the tool. I'm really not sure what is going on.

I'll try one example from my MP2 camera. I shut down ZoneMinder, cleared the zmc log for the camera, restarted ZoneMinder and produced some motion and sound in front of the camera. The results:
* On Firefox, the video showed a single still frame of the middle of the action and played a portion of the audio that started about halfway through the action. The length shown by the player looked normal, but kept incrementing even after getting to the end.
* On Chrome, the video showed the same single frame, but there was no audio and the length was listed as 26 minutes.
* When the file was downloaded, mpv, mplayer and vlc all showed active video that started at a place that looks similar to the single frame from the browsers. There is no sound.

Here is the log:
[zmc_m9.log](https://github.com/ZoneMinder/zoneminder/files/3315767/zmc_m9.log)

Here is the mediainfo output:
```
General
Complete name                            : 287694-video.mp4
Format                                   : MPEG-4
Format profile                           : Base Media
Codec ID                                 : isom (isom/iso2/avc1/mp41)
File size                                : 2.74 MiB
Duration                                 : 7 s 832 ms
Overall bit rate mode                    : Variable
Overall bit rate                         : 2 939 kb/s
Movie name                               : Zoneminder Security Recording
Writing application                      : Lavf57.83.100

Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : High@L4.1
Format settings                          : CABAC / 1 Ref Frames
Format settings, CABAC                   : Yes
Format settings, ReFrames                : 1 frame
Codec ID                                 : avc1
Codec ID/Info                            : Advanced Video Coding
Duration                                 : 7 s 600 ms
Bit rate mode                            : Variable
Bit rate                                 : 2 996 kb/s
Maximum bit rate                         : 4 910 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate mode                          : Constant
Frame rate                               : 10.000 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 0.144
Stream size                              : 2.71 MiB (99%)
Color range                              : Full
Color primaries                          : BT.709
Transfer characteristics                 : BT.709
Matrix coefficients                      : BT.709

Audio
ID                                       : 2
Format                                   : AAC
Format/Info                              : Advanced Audio Codec
Format profile                           : LC
Codec ID                                 : mp4a-40-2
Duration                                 : 7 s 832 ms
Duration_FirstFrame                      : -8 ms
Duration_LastFrame                       : -8 ms
Bit rate mode                            : Variable
Bit rate                                 : 29.3 kb/s
Maximum bit rate                         : 32.8 kb/s
Channel(s)                               : 2 channels
Channel(s)_Original                      : 1 channel
Channel positions                        : Front: C
Sampling rate                            : 16.0 kHz
Frame rate                               : 15.625 FPS (1024 SPF)
Compression mode                         : Lossy
Stream size                              : 28.0 KiB (1%)
Default                                  : Yes
Alternate group                          : 1
```

connortechnology, 2019-07-01:
Can you confirm/deny that this issue is resolved?  It is working well for me.  I basically reverted everything and went a different way.

neillbell, 2019-07-01:
This issue is definitely better, but still not perfect. As usual, Chromium works pretty well. I still see some synchronization issues when using Firefox, mpv, vlc and Windows Media Player. Depending on the player the audio is out of sync by fraction of a second. Even using Chromium, the synchronization appears to be better for MP2 than for G.711 or G.726.

I'm sure that you are sick to death of this. Do we need to address it further?

connortechnology, 2019-07-01:
Lets try one more time.  debug logs please.  I suspect it is due to delayed packets in the encoder.

neillbell, 2019-07-01:
Here's a log from my camera that is running G.711 audio:
[zmc_m5.log](https://github.com/ZoneMinder/zoneminder/files/3346270/zmc_m5.log)
```
General
Complete name                            : /var/cache/zoneminder/events/5/2019-07-01/300262/300262-video.mp4
Format                                   : MPEG-4
Format profile                           : Base Media
Codec ID                                 : isom (isom/iso2/avc1/mp41)
File size                                : 2.45 MiB
Duration                                 : 8 s 436 ms
Overall bit rate mode                    : Variable
Overall bit rate                         : 2 435 kb/s
Movie name                               : Zoneminder Security Recording
Writing application                      : Lavf57.83.100

Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : Main@L4.2
Format settings                          : CABAC / 1 Ref Frames
Format settings, CABAC                   : Yes
Format settings, ReFrames                : 1 frame
Format settings, GOP                     : M=1, N=10
Codec ID                                 : avc1
Codec ID/Info                            : Advanced Video Coding
Duration                                 : 8 s 400 ms
Bit rate                                 : 2 414 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate                               : 9.762 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 0.119
Stream size                              : 2.42 MiB (99%)
Color range                              : Full
Color primaries                          : BT.709
Transfer characteristics                 : BT.709
Matrix coefficients                      : BT.709

Audio
ID                                       : 2
Format                                   : AAC
Format/Info                              : Advanced Audio Codec
Format profile                           : LC
Codec ID                                 : mp4a-40-2
Duration                                 : 8 s 436 ms
Bit rate mode                            : Variable
Bit rate                                 : 27.9 kb/s
Maximum bit rate                         : 32.8 kb/s
Channel(s)                               : 2 channels
Channel(s)_Original                      : 1 channel
Channel positions                        : Front: C
Sampling rate                            : 8 000 Hz
Frame rate                               : 7.812 FPS (1024 SPF)
Compression mode                         : Lossy
Stream size                              : 28.7 KiB (1%)
Default                                  : Yes
Alternate group                          : 1
```

Here is a log from my camera running G.726 audio:
[zmc_m9.log](https://github.com/ZoneMinder/zoneminder/files/3346275/zmc_m9.log)
```
General
Complete name                            : /var/cache/zoneminder/events/9/2019-07-01/300258/300258-video.mp4
Format                                   : MPEG-4
[... 34 lines cut ...]
Stream size                              : 3.37 MiB (99%)
Color range                              : Full
Color primaries                          : BT.709
Transfer characteristics                 : BT.709
Matrix coefficients                      : BT.709

Audio
ID                                       : 2
Format                                   : AAC
Format/Info                              : Advanced Audio Codec
Format profile                           : LC
Codec ID                                 : mp4a-40-2
Duration                                 : 9 s 456 ms
Bit rate mode                            : Constant
Bit rate                                 : 16.3 kb/s
Channel(s)                               : 2 channels
Channel(s)_Original                      : 1 channel
Channel positions                        : Front: C
Sampling rate                            : 8 000 Hz
Frame rate                               : 7.812 FPS (1024 SPF)
Compression mode                         : Lossy
Stream size                              : 18.8 KiB (1%)
Default                                  : Yes
Alternate group                          : 1
```

connortechnology, 2019-07-02:
Okay try latest.  Should move audio back a fraction of a second.

neillbell, 2019-07-05:
G.726 and G.711 look good. They are consistent across different tools.

MP2 is not working at all.
[zmc_m9.log](https://github.com/ZoneMinder/zoneminder/files/3363707/zmc_m9.log)

connortechnology, 2019-07-05:
well that's something I guess!

connortechnology, 2019-07-07:
Did some fixes.  Audio now works for my resample cameras.  It was way off in negative time before.  

Unfortunately I am finding that the audio is now a fraction of a second early.  Which is annoying.

neillbell, 2019-07-10:
G.726 and G.711 look spot on and are consistent with both Chromium and mpv.

MP2 is sort of working. Viewing with Chromium looks normal, but the audio is delayed by something like 0.25 seconds and the length is incorrect. Playing with mpv strangely shows the video at about 2x speed with audio present and unsynchronized, of course. Vlc shows normal rate video, but an audio delay of about 45 seconds.

[zmc_m9.log](https://github.com/ZoneMinder/zoneminder/files/3377914/zmc_m9.log)

Mediainfo output shows an incorrect audio length:
```
General
Complete name                            : 312836-video.mp4
Format                                   : MPEG-4
Format profile                           : Base Media
Codec ID                                 : isom (isom/iso2/avc1/mp41)
File size                                : 4.70 MiB
Duration                                 : 3 min 51 s
Overall bit rate mode                    : Variable
Overall bit rate                         : 171 kb/s
Movie name                               : Zoneminder Security Recording
Writing application                      : Lavf57.83.100

Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : High@L4.1
Format settings                          : CABAC / 1 Ref Frames
Format settings, CABAC                   : Yes
Format settings, ReFrames                : 1 frame
Format settings, GOP                     : M=1, N=20
Codec ID                                 : avc1
Codec ID/Info                            : Advanced Video Coding
Duration                                 : 13 s 0 ms
Bit rate mode                            : Variable
Bit rate                                 : 3 002 kb/s
Maximum bit rate                         : 4 910 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate mode                          : Constant
Frame rate                               : 10.000 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 0.145
Stream size                              : 4.65 MiB (99%)
Color range                              : Full
Color primaries                          : BT.709
Transfer characteristics                 : BT.709
Matrix coefficients                      : BT.709

Audio
ID                                       : 2
Format                                   : AAC
Format/Info                              : Advanced Audio Codec
Format profile                           : LC
Codec ID                                 : mp4a-40-2
Duration                                 : 3 min 51 s
Bit rate mode                            : Variable
Bit rate                                 : 1 658 b/s
Maximum bit rate                         : 32.8 kb/s
Channel(s)                               : 2 channels
Channel(s)_Original                      : 1 channel
Channel positions                        : Front: C
Sampling rate                            : 16.0 kHz
Frame rate                               : 15.625 FPS (1024 SPF)
Compression mode                         : Lossy
Stream size                              : 46.8 KiB (1%)
Default                                  : Yes
Alternate group                          : 1
```

connortechnology, 2019-07-11:
I believe I have fixed the length being wrong problem  Wasn't applying pts adjustments to the frames flushed from the encoder at the end of recording...

neillbell, 2019-07-12:
I'm still seeing something similar. G.726 and G.711 look great, but MP2 still has an audio delay and shows an incorrect length.

[zmc_m9.log](https://github.com/ZoneMinder/zoneminder/files/3386890/zmc_m9.log)

```
General
Complete name                            : 316742-video.mp4
Format                                   : MPEG-4
Format profile                           : Base Media
Codec ID                                 : isom (isom/iso2/avc1/mp41)
File size                                : 4.62 MiB
Duration                                 : 1 min 11 s
Overall bit rate mode                    : Variable
Overall bit rate                         : 543 kb/s
Movie name                               : Zoneminder Security Recording
Writing application                      : Lavf57.83.100

Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : High@L4.1
Format settings                          : CABAC / 1 Ref Frames
Format settings, CABAC                   : Yes
Format settings, ReFrames                : 1 frame
Format settings, GOP                     : M=1, N=20
Codec ID                                 : avc1
Codec ID/Info                            : Advanced Video Coding
Duration                                 : 12 s 900 ms
Bit rate mode                            : Variable
Bit rate                                 : 2 975 kb/s
Maximum bit rate                         : 4 910 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate mode                          : Constant
Frame rate                               : 10.000 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 0.143
Stream size                              : 4.58 MiB (99%)
Color range                              : Full
Color primaries                          : BT.709
Transfer characteristics                 : BT.709
Matrix coefficients                      : BT.709

Audio
ID                                       : 2
Format                                   : AAC
Format/Info                              : Advanced Audio Codec
Format profile                           : LC
Codec ID                                 : mp4a-40-2
Duration                                 : 1 min 11 s
Duration_FirstFrame                      : -341 ms
Duration_LastFrame                       : -341 ms
Bit rate mode                            : Variable
Bit rate                                 : 5 260 b/s
Maximum bit rate                         : 32.8 kb/s
Channel(s)                               : 2 channels
Channel(s)_Original                      : 1 channel
Channel positions                        : Front: C
Sampling rate                            : 16.0 kHz
Frame rate                               : 15.625 FPS (1024 SPF)
Compression mode                         : Lossy
Stream size                              : 45.9 KiB (1%)
Default                                  : Yes
Alternate group                          : 1
```

connortechnology, 2019-08-25:
So I took another look, and I've added some more debug and a few other things.  A new log would be appreciated. 
I'm having trouble wrapping my head around it all, the logs don't show any degative duration frames being written, but the ffprobe is complaining about them.  
The other thing I note is that each input audio frame has a duration of 6480.. but the aac encoder is outputting frames with a duration of 1024.  
Maybe for each input frame I need to allow for multiple output frames.. you'd think they would get output when flushing though. I did make a small fix on flushing...
So hence the request for a new debug log.

neillbell, 2019-08-29:
[zmc_m9.log](https://github.com/ZoneMinder/zoneminder/files/3553550/zmc_m9.log)
```
General
Complete name                            : 394575-video.mp4
Format                                   : MPEG-4
Format profile                           : Base Media
Codec ID                                 : isom (isom/iso2/avc1/mp41)
File size                                : 3.87 MiB
Duration                                 : 1 min 1 s
Overall bit rate mode                    : Variable
Overall bit rate                         : 530 kb/s
Movie name                               : Zoneminder Security Recording
Writing application                      : Lavf57.83.100

Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : High@L4.1
Format settings                          : CABAC / 1 Ref Frames
Format settings, CABAC                   : Yes
Format settings, ReFrames                : 1 frame
Format settings, GOP                     : M=1, N=10
Codec ID                                 : avc1
Codec ID/Info                            : Advanced Video Coding
Duration                                 : 11 s 0 ms
Bit rate mode                            : Variable
Bit rate                                 : 2 922 kb/s
Maximum bit rate                         : 4 910 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate mode                          : Constant
Frame rate                               : 10.000 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 0.141
Stream size                              : 3.83 MiB (99%)
Color range                              : Full
Color primaries                          : BT.709
Transfer characteristics                 : BT.709
Matrix coefficients                      : BT.709

Audio
ID                                       : 2
Format                                   : AAC
Format/Info                              : Advanced Audio Codec
Format profile                           : LC
Codec ID                                 : mp4a-40-2
Duration                                 : 1 min 1 s
Duration_FirstFrame                      : -341 ms
Duration_LastFrame                       : -341 ms
Bit rate mode                            : Variable
Bit rate                                 : 5 254 b/s
Maximum bit rate                         : 32.8 kb/s
Channel(s)                               : 2 channels
Channel(s)_Original                      : 1 channel
Channel positions                        : Front: C
Sampling rate                            : 16.0 kHz
Frame rate                               : 15.625 FPS (1024 SPF)
Compression mode                         : Lossy
Stream size                              : 39.3 KiB (1%)
Default                                  : Yes
Alternate group                          : 1
```

inpos, 2019-09-08:
Is there any workaround for 1.32.3 ?
Debian repo contains only that version and latest build crash with segfault on h264 passthrough

connortechnology, 2019-09-08:
Why are you hijacking other threads?  We will address the crashing in it's own issue.

inpos, 2019-09-08:
Because question is:
`Is there any workaround for 1.32.3 ?`
Workaround for a/v duration sync

connortechnology, 2019-09-08:
If you are in fact having THIS problem, then no there is no work around.

inpos, 2019-09-09:
For those who have THIS issue (audio/video duration unsync).
My camera send stream with 15 FPS and when I specify **Maximum FPS** and **Alarm Maximum FPS** with 30 FPS (debian version specify them by default in UI) there is a/v duration sync issue.
When I clear these parameters audio and video have the same duration.

connortechnology, 2019-09-11:
@neillbell can you confirm that your maxfps fields are empty?
I don't suppose I could get access to your camera?
Can I get the .mp4 that is generated?
I may just buy one of these cameras but it will likely take some time for it to get to me.

neillbell, 2019-09-11:
Yes, the maxfps fields are empty. I can definitely send you the video. I'll send the 394575-video.mp4 file to your email.

I'm sure I can set up access to the camera for you. Let me see about setting it up and I'll email you.

connortechnology, 2019-09-11:
Having the file helps.  I'll see what I can figure out from it + logs.  Then try getting access to camera.  That is one messed up file. When I use ffprobe, the frames look ok, but ffplay can't handle it either.  

The one thing I still can't figure out is that we are putting in audio frames at 6840 for duration and getting out frames that are 1024 duration... So somehow losing audio time.  Also, not having audio until way after you've made the sound.... there are audio packets at an appropriate pts... but it doesn't play them till much later.  I'm very confused.  But I'll get it figured out eventually.

PR #2646 (2019-06-24): Revert audio sync fixes
Revert the work I did to srt packets by dts after changing their time_base.  It doesn't work.  Instead since dts doesn't actually matter to playback, only ffmpeg storing, munge it to whatever is neccessary to make ffmpeg happy.
Hopefully fixes #2621
closed 2019-09-15 by connortechnology