ZoneMinder/zoneminder #2631: ZoneMinder 1.34.0 release

SteveGilvarry, 2019-06-03:
Bugs/Fixes for Release
- [ ] Separate source path into component parts for FFmpeg @SteveGilvarry 
- [ ] Ensure a note about Maximum FPS is included in release notes and possibly clear Max FPS  during upgrade
- [ ] Stop writing passwords to logs

Pre-Release Standard Actions
- [ ] Review Documentation, Install Guides etc. Any new specific functions to be documented?

knight-of-ni, 2019-06-12:
- Fix rpmspecfile so ZoneMinder builds on redhat based distros
- bring in the the two new git submodules statically, rather than as git submodules (reduces the amount of work for packagers)

knight-of-ni, 2019-06-23:
Rpm builds are currently failing due to the inclusion of header and library files needed to build the new bcrypt stuff. These files are not needed for runtime and could potentially conflict with bcrypt packages that may already exist on the target machine. While these files are needed to build, they should not install on the filesystem. I'll be looking to fix this in the coming days.

```
Checking for unpackaged file(s): /usr/lib/rpm/check-files /builddir/build/BUILDROOT/zoneminder-1.33.9-1.fc29.x86_64
error: Installed (but unpackaged) file(s) found:
   /usr/include/bcrypt/BCrypt.hpp
   /usr/include/bcrypt/bcrypt.h
   /usr/include/bcrypt/crypt.h
   /usr/include/bcrypt/crypt_blowfish.h
   /usr/include/bcrypt/crypt_gensalt.h
   /usr/include/bcrypt/ow-crypt.h
   /usr/include/bcrypt/winbcrypt.h
   /usr/lib64/libbcrypt.a

RPM build errors:
    Installed (but unpackaged) file(s) found:
   /usr/include/bcrypt/BCrypt.hpp
   /usr/include/bcrypt/bcrypt.h
   /usr/include/bcrypt/crypt.h
   /usr/include/bcrypt/crypt_blowfish.h
   /usr/include/bcrypt/crypt_gensalt.h
   /usr/include/bcrypt/ow-crypt.h
   /usr/include/bcrypt/winbcrypt.h
   /usr/lib64/libbcrypt.a
ERROR: Exception(/home/abauer/rpmbuild/SRPMS/zoneminder-1.33.9-1.fc29.src.rpm) Config(fedora-29-x86_64) 6 minutes 49 seconds
```

knight-of-ni, 2019-06-23:
And I fixed it with commit: 82672d9b8605dc725a550e4f5afe6ab5ed8edbf4

GusevVictor, 2020-01-12:
which version of zoneminder you recomend use for production now?

abishai, 2020-01-18:
I fast tried to build new ZM under FreeBSD 12.1 and found issue with bcrypt libraries.
```
In file included from /wrkdirs/usr/ports/multimedia/zoneminder/work/zoneminder-1.34.0/src/libbcrypt/src/bcrypt.c:36:
/wrkdirs/usr/ports/multimedia/zoneminder/work/zoneminder-1.34.0/src/libbcrypt/include/bcrypt/ow-crypt.h:27:14: error: conflicting types for 'crypt_r'
extern char *crypt_r(__const char *key, __const char *setting, void *data);
```
Probably, because of this https://www.freebsd.org/cgi/man.cgi?query=crypt_r&apropos=0&sektion=0&manpath=FreeBSD+12-current&arch=default&format=html

Looks like this define is needed https://github.com/ZoneMinder/zoneminder/blob/00183b535adf638f61e3b0f6f5859324df8e68a8/src/libbcrypt/include/bcrypt/ow-crypt.h#L25 , but only if FreeBSD >=12

I think, I'll just patch CMakeLists.txt conditionally as FreeBSD 11 is not going to EOL in short term.

closed 2021-03-10 by connortechnology