ZoneMinder/zoneminder #2632: Montage Review shows "No Data" for bulk frames under Centos 7 due to default PHP MySQL driver deficiencies

djast, 2019-06-03:
**Describe Your Environment**
- zoneminder-1.32.3-1.el7.x86_64
- RPM Fusion
- CentOS Linux release 7.6.1810 (Core)

**Describe the bug**
Zoneminder assumes that the PHP MySQL driver will return query results of the types as declared in the database.  However, the default php-mysql package under CentOS 7 returns strings for all queries, even for integer types.

One of the ways this manifests itself is that when bulk frames are in use, "Montage Review" for past events does not work correctly, and displays "No Data" after the first frame.

**To Reproduce**
Steps to reproduce the behavior:
1. Install and configure ZoneMinder 1.32.2 on a fresh CentOS 7 installation from RPM Fusion.  Observe that the dependencies will cause the php-mysql package to be installed (rather than the php-mysqlnd package, which uses the native driver).
2. Configure at least one camera in a mode which uses bulk frames
3. Click on "Montage Review"
4. Click "1 Hour"
5. Observe that after the first frame, all cameras will show "No Data".  If you "View source" on the page, you will observe that the values for the FrameID and EventID attributes in the events{} object are double-quoted; e.g. "FrameId" : "11960" instead of "FrameId" : 11960 .

**Expected behavior**
Frames should be displayed correctly in the montage.

**Debug Logs**
```
Jun  3 11:07:42 vid-c web_php[39465]: FAT [No Frame found for event(39856) and frame id(380063)]
Jun  3 11:07:43 vid-c web_php[39492]: FAT [No Frame found for event(39856) and frame id(380082)]
Jun  3 11:07:44 vid-c web_php[39439]: FAT [No Frame found for event(39856) and frame id(39002)]
Jun  3 11:07:45 vid-c web_php[39492]: FAT [No Frame found for event(39856) and frame id(390021)]
Jun  3 11:07:46 vid-c web_php[39439]: FAT [No Frame found for event(39856) and frame id(390041)]
Jun  3 11:07:47 vid-c web_php[39465]: FAT [No Frame found for event(39856) and frame id(390060)]
Jun  3 11:07:48 vid-c web_php[39957]: FAT [No Frame found for event(39856) and frame id(390080)]
Jun  3 11:07:49 vid-c web_php[39439]: FAT [No Frame found for event(39856) and frame id(40000)]
Jun  3 11:07:50 vid-c web_php[39465]: FAT [No Frame found for event(39856) and frame id(400019)]

```

Note that the Frame IDs are out of range.  This is because rather than adding the frame offset numerically to the base Frame ID, the Javascript is concatenating it.

**Additional Information**
There are several approaches that could be taken to address this:

A workaround for CentOS 7 users affected by this issue is to swap out the default driver package for the native driver, via "yum swap php-mysql php-mysqlnd" (followed by "apachectl graceful" to reload the web server with the new driver).

While it might appear that changing the package dependencies to require the php-mysqlnd package instead of php-mysql could address the issue, doing so would introduce dependency problems for existing users, since the php-mysql and php-mysqlnd packages are incompatible; I don't believe the package manager is clever enough to replace php-mysql if it is already installed, so the documentation would need to explain clearly how to ensure that php-mysqlnd is installed in preference to php-mysql (e.g., by installing php-mysqlnd explicitly prior to or alongside zoneminder).

A more complete solution, which would also apply if ZoneMinder needs to run in other environments with similarly-broken PHP MySQL drivers which return strings for columns containing integers, might be to convert the data type--e.g., something in the vein of '$this->{"FrameID"} = (int)$this->{"FrameID"}' in the constructor for the Frame class, or somehow adding a hack one level up in database.php.

Perhaps the ideal fix would be to persuade Red Hat to deprecate the php-mysql package in favour of php-mysqlnd, but I suspect that approach is not likely to be feasible.

djast, 2019-06-03:
More information on this at https://forums.zoneminder.com/viewtopic.php?f=38&t=27953&p=110699 .

connortechnology, 2019-06-10:
The presence of quotes around the FrameId values is actually valid json syntax.
We need a parseInt on line 112
Let's try changing
frame_id = Frame.FrameId + parseInt( (Frame.NextFrameId-Frame.FrameId) * ( time-Frame.TimeStampSecs )/duration );
to
frame_id = parseInt(Frame.FrameId) + parseInt( (Frame.NextFrameId-Frame.FrameId) * ( time-Frame.TimeStampSecs )/duration );

djast, 2019-06-10:
Yes, it's valid json syntax, but it misrepresents the intended type of the value.  There could be other places in the code where Javascript is generated that expects the various Id fields to be numeric--it would be cleaner to fix the object definition if possible rather than trying to identify all places where addition is done with Id, FrameId, EventId, etc.

Note that (Frame.NextFrameId-Frame.FrameId) technically works because Javascript converts strings to numbers for subtraction (whereas for the "+" operator it converts numbers to strings for concatenation; i.e., '5' - 3 = 2, but '5' + 3 = '53'), but it would be better form to cast the fields explicitly.

The proposed fix would work, but since the string value is already an integer, the unary + operator would be more concise and efficient than parseInt(), i.e.
frame_id = +Frame.FrameId + parseInt( (+Frame.NextFrameId - +Frame.FrameId) * ( time-Frame.TimeStampSecs )/duration );

connortechnology, 2019-06-11:
Please test latest master. Along with fixing this I made some improvements which should improve speed.  Also fixed a few oddities like if you set the speed to 0 it won't start back up after setting back to 1x

closed 2019-08-17 by stale[bot]