ZoneMinder/zoneminder #2636: Storage not working properly on size limited tmpfs (1 folder = 2 storage pools)

BlueMax, 2019-06-08:
Zoneminder v1.32.3
Alpine Linux edge x86_64 (rolling release)

I've created a 128MB tmpfs storage for testing purpose. Its intended to use a larger tmpfs folder later.
`mount -t tmpfs nosuid,nodev,size=128M tmpfs /tmp/zonestorage`

The storage diskspace shown on the webui topbar lists two storages though.
- the 128MB limited folder
- and the underlying root partition (58%)

The purge filters won't kick-in and the storage runs full. 
It looks like it needs some additional sanity checks.

[image]

connortechnology, 2019-06-08:
You have to tell the purge filter which storage area to work on.

connortechnology, 2019-06-08:
This is a valid complaint, which needs to be fixed...

BlueMax, 2019-06-08:
That 2nd `default` storage pool (having the same name) seems not related to the tmpfs storage pool though. The tmpfs storage was created by editing the existing `default` entry.
I've removed `ZM_DIR_EVENTS=` and `ZM_DIR_IMAGES=` from `zm.conf`, re-setup the database from scratch and cleared all `storage` entries in webui but this one `default` storage (that seems to point to /var/lib/zoneminder/events here) seems always shown in topbar. Not sure though if that's an Alpine Linux specific issue.

connortechnology, 2019-08-08:
ZM will n longer auto-add an entry for the default storage area in the header.

closed 2019-08-08 by connortechnology