ZoneMinder/zoneminder #2642: Stored Cross Site Scripting - Filters

joeylane, 2019-06-18:
- ZoneMinder version 1.32.3
- Installed from PPA (Ubuntu Xenial)
- Ubuntu 16.04 Xenial

The 'Filters' page in ZoneMinder 1.32.3 is vulnerable to stored cross site scripting due to improper input sanitation in the 'Name' field.  This is the latest version available in the Ubuntu 16.04 PPA.  Please see attached screenshots.

[image]
[image]

Steps to reproduce the behavior:
1. Go to the "Filters" page.
2. In the "Name" field paste in a javascript payload such as "<script>alert('Vulnerable')</script>"
3. Refresh the page to verify that the payload executes and remains stored through browser sessions.

Expected behavior:
The user input fields should be sanitized.  Special characters should be limited or escaped to prevent injection of malicious code.

connortechnology, 2019-06-18:
This has already been fixed in master.

closed 2019-06-18 by connortechnology

rakeshsv, 2019-06-30:
> This has already been fixed in master

Please can you give commit #

galaktipus, 2019-12-26:
> This has already been fixed in master.

Can we get a commit please?

SteveGilvarry, 2019-12-26:
Do you one better the original issue and closing commit. https://github.com/ZoneMinder/zoneminder/issues/2455