ZoneMinder/zoneminder #2648: undefined reference to symbol 'dlclose@@GLIBC_2.2.5'

JohnAZoidberg, 2019-06-27:
**Describe Your Environment**
- Building from the release tarball on GitHub of 1.32.3
- Modules
  - https://github.com/ZoneMinder/crud at 3.1.0-zm
  - https://github.com/ZoneMinder/CakePHP-Enum-Behavior at 1.0-zm
- Addons
  - [Xiaomi.pem](https://gist.githubusercontent.com/joshstrange/73a2f24dfaf5cd5b470024096ce2680f/raw/e964270c5cdbf95e5b7f214f7f0fc6113791530e/Xiaomi.pm)
- Distribution: NixOS unstable
- Build File: https://github.com/nixos/nixpkgs/blob/master/pkgs/servers/zoneminder/default.nix (You're probably unfamiliar with this build tool but maybe you can spot a mistake)
- Browser name and version (if this is an issue with the web interface)

Yes, I read the README "Don't build from source" but I'm trying to fix the build for the package manager of NixOS.

**Describe the bug**
Build process tries fails to link becaues the symbol `dlclose` is not provided because it's not explicitly linked with the dynamic linker library libdl.

**To Reproduce**
Try to build from source. Then it complains about misssing symbol `dlclose`.
When I add `-ldl` to the compiler options it builds fine.

**Expected behavior**
Build without error.

**Debug Logs**
```
[ 95%] Linking CXX executable zma                                                                                      
Manifying 24 pod documents                                                                                             
[ 96%] Linking CXX executable zms                                                                                      
[ 98%] Linking CXX executable zmu
/nix/store/qw8rhqk004qmqgnkvkpj85l1s194ang1-binutils-2.31.1/bin/ld: /nix/store/xm8fav5cy5kmjqzqj1l7yqhnikj4i6ff-mariadb-10.3.15-dev/lib/libmysqlclient.a(ma_client_plugin.c.o): undefined reference to symbol 'dlclose@@GLIBC_2.2.5'
/nix/store/qw8rhqk004qmqgnkvkpj85l1s194ang1-binutils-2.31.1/bin/ld: /nix/store/avr2x43njlq4kyb1a9zgrh6fih5fq4si-glibc-2.27/lib/libdl.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/zma.dir/build.make:101: src/zma] Error 1
make[1]: *** [CMakeFiles/Makefile2:269: src/CMakeFiles/zma.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Manifying 23 pod documents
[100%] Linking CXX executable zmc
/nix/store/qw8rhqk004qmqgnkvkpj85l1s194ang1-binutils-2.31.1/bin/ld: /nix/store/xm8fav5cy5kmjqzqj1l7yqhnikj4i6ff-mariadb-10.3.15-dev/lib/libmysqlclient.a(ma_client_plugin.c.o): undefined reference to symbol 'dlclose@@GLIBC_2.2.5'
/nix/store/qw8rhqk004qmqgnkvkpj85l1s194ang1-binutils-2.31.1/bin/ld: /nix/store/avr2x43njlq4kyb1a9zgrh6fih5fq4si-glibc-2.27/lib/libdl.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/zms.dir/build.make:101: src/zms] Error 1
make[1]: *** [CMakeFiles/Makefile2:412: src/CMakeFiles/zms.dir/all] Error 2
/nix/store/qw8rhqk004qmqgnkvkpj85l1s194ang1-binutils-2.31.1/bin/ld: /nix/store/xm8fav5cy5kmjqzqj1l7yqhnikj4i6ff-mariadb-10.3.15-dev/lib/libmysqlclient.a(ma_client_plugin.c.o): undefined reference to symbol 'dlclose@@GLIBC_2.2.5'
/nix/store/qw8rhqk004qmqgnkvkpj85l1s194ang1-binutils-2.31.1/bin/ld: /nix/store/avr2x43njlq4kyb1a9zgrh6fih5fq4si-glibc-2.27/lib/libdl.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/zmu.dir/build.make:101: src/zmu] Error 1
make[1]: *** [CMakeFiles/Makefile2:306: src/CMakeFiles/zmu.dir/all] Error 2
Manifying 22 pod documents                                 
/nix/store/qw8rhqk004qmqgnkvkpj85l1s194ang1-binutils-2.31.1/bin/ld: /nix/store/xm8fav5cy5kmjqzqj1l7yqhnikj4i6ff-mariadb-10.3.15-dev/lib/libmysqlclient.a(ma_client_plugin.c.o): undefined reference to symbol 'dlclose@@GLIBC_2.2.5'
/nix/store/qw8rhqk004qmqgnkvkpj85l1s194ang1-binutils-2.31.1/bin/ld: /nix/store/avr2x43njlq4kyb1a9zgrh6fih5fq4si-glibc-2.27/lib/libdl.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/zmc.dir/build.make:101: src/zmc] Error 1
make[1]: *** [CMakeFiles/Makefile2:343: src/CMakeFiles/zmc.dir/all] Error 2
```

connortechnology, 2019-06-27:
I think I have seen this, although it has been a long time and I think it was building on freebsd. CMake is suppose to auto detect this.  

Let's take a look at the output during cmake configuration and CmakeFiles/CMakeOutput.log

JohnAZoidberg, 2019-06-27:
Our cmake flags:
```
-DCMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY=ON -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=ON -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_BUILD_TYPE=Release -DCMAKE_SKIP_BUILD_RPATH=ON -DBUILD_TESTING=OFF -DCMAKE_INSTALL_IN
CLUDEDIR=/nix/store/wspgbm1aawqfdryv82vrszcl4qs80qcy-zoneminder-1.32.3/include -DCMAKE_INSTALL_LIBDIR=/nix/store/wspgbm1aawqfdryv82vrszcl4qs80qcy-zoneminder-1.32.3/lib -DCMAKE_INSTALL_NAME_DIR=/nix/store/wspgbm1aawqfdryv82vrszcl4qs80qcy-zo
neminder-1.32.3/lib -DCMAKE_POLICY_DEFAULT_CMP0025=NEW -DCMAKE_OSX_DEPLOYMENT_TARGET= -DCMAKE_OSX_SYSROOT= -DCMAKE_FIND_FRAMEWORK=last -DCMAKE_STRIP=/nix/store/qw8rhqk004qmqgnkvkpj85l1s194ang1-binutils-2.31.1/bin/strip -DCMAKE_RANLIB=/nix/
store/qw8rhqk004qmqgnkvkpj85l1s194ang1-binutils-2.31.1/bin/ranlib -DCMAKE_AR=/nix/store/qw8rhqk004qmqgnkvkpj85l1s194ang1-binutils-2.31.1/bin/ar -DCMAKE_C_COMPILER=gcc -DCMAKE_CXX_COMPILER=g++ -DCMAKE_INSTALL_PREFIX=/nix/store/wspgbm1aawqfd
ryv82vrszcl4qs80qcy-zoneminder-1.32.3 -DWITH_SYSTEMD=ON -DZM_LOGDIR=/var/log/zoneminder -DZM_RUNDIR=/run/zoneminder -DZM_SOCKDIR=/run/zoneminder -DZM_TMPDIR=/tmp/zoneminder -DZM_CONFIG_DIR=/nix/store/wspgbm1aawqfdryv82vrszcl4qs80qcy-zonemi
nder-1.32.3/etc/zoneminder -DZM_WEB_USER=zoneminder -DZM_WEB_GROUP=zoneminder
```                                        

<details>

```
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /nix/store/gzx96ys6wczvafmg3p57s9zc6q4xpkzz-gcc-wrapper-7.4.0/bin/gcc
-- Check for working C compiler: /nix/store/gzx96ys6wczvafmg3p57s9zc6q4xpkzz-gcc-wrapper-7.4.0/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /nix/store/gzx96ys6wczvafmg3p57s9zc6q4xpkzz-gcc-wrapper-7.4.0/bin/g++
-- Check for working CXX compiler: /nix/store/gzx96ys6wczvafmg3p57s9zc6q4xpkzz-gcc-wrapper-7.4.0/bin/g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test HAVE_SENDFILE4_SUPPORT
-- Performing Test HAVE_SENDFILE4_SUPPORT - Success
-- Sendfile support: Linux/Solaris sendfile()
-- Looking for libv4l1-videodev.h
-- Looking for libv4l1-videodev.h - not found
-- Looking for linux/videodev.h
-- Looking for linux/videodev.h - not found
-- Looking for linux/videodev2.h
-- Looking for linux/videodev2.h - found
-- Looking for execinfo.h
-- Looking for execinfo.h - found
-- Looking for backtrace
-- Looking for backtrace - found
-- Looking for backtrace_symbols
-- Looking for backtrace_symbols - found
-- Looking for ucontext.h
-- Looking for ucontext.h - found
-- Looking for sys/sendfile.h
-- Looking for sys/sendfile.h - found
-- Looking for sys/syscall.h
-- Looking for sys/syscall.h - found
-- Looking for syscall
-- Looking for syscall - found
-- Looking for sendfile
-- Looking for sendfile - found
-- Looking for posix_memalign
-- Looking for posix_memalign - found
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of siginfo_t
-- Check size of siginfo_t - done
-- Check size of ucontext_t
-- Check size of ucontext_t - done
-- Looking for clock_gettime in rt
-- Looking for clock_gettime in rt - found
-- Found ZLIB: /nix/store/2fy10zrw6hprl3nhzxy9q6im1ajy5jm8-zlib-1.2.11/lib/libz.so (found version "1.2.11")
-- Looking for zlib.h
-- Looking for zlib.h - found
-- Found CURL: /nix/store/7300bqcyi8zm937zqq70pbc1pnf744b1-curl-7.65.0/lib/libcurl.so (found version "7.65.0")
[... 62 lines cut ...]
-- Checking for one of the modules 'polkit-gobject-1;polkit'
-- Checking for module 'glib-2.0'
--   Found glib-2.0, version 2.60.3
-- Found GLIB2: /nix/store/v62cbvfzi4k04iaylnm0rycssj9jj8lg-glib-2.60.3/lib/libglib-2.0.so
-- Found Polkit: /nix/store/rhxg2vxvxb3208g8yy14fnwrhjcmi7n4-polkit-0.116/lib/libpolkit-gobject-1.so
-- Using mootools core file: mootools-core-1.6.0-compat.js
-- Using mootools more file: mootools-more-1.6.0.js
-- Optional libraries found: zlib cURL OpenSSL PCRE x264 mp4v2 AVFormat AVCodec AVDevice AVUtil SWScale AVResample
-- Optional libraries not found: GCrypt GnuTLS libVLC
-- Running ZoneMinder configuration generator
Generating 'src/zm_config_defines.h'
Updating 'db/zm_create.sql'
-- ZoneMinder configuration generator completed successfully
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    BUILD_TESTING
    CMAKE_EXPORT_NO_PACKAGE_REGISTRY

-- Build files have been written to: /build/source/build
```

</details>

connortechnology, 2019-06-27:
Is there a reason for all the extra CMAKE flags?  This is what we use on debian...
cmake -DCMAKE_INSTALL_PREFIX=/usr \
  -DCMAKE_SKIP_RPATH=ON \
  -DCMAKE_VERBOSE_MAKEFILE=OFF \
  -DCMAKE_COLOR_MAKEFILE=ON \
  -DZM_RUNDIR=/var/run/zm \
  -DZM_SOCKDIR=/var/run/zm \
  -DZM_TMPDIR=/var/tmp/zm \
  -DZM_LOGDIR=/var/log/zm \
  -DZM_WEBDIR=/usr/share/zoneminder/www \
  -DZM_CONTENTDIR=/var/cache/zoneminder \
  -DZM_CGIDIR=/usr/lib/zoneminder/cgi-bin \
  -DZM_CACHEDIR=/var/cache/zoneminder/cache \
  -DZM_WEB_USER=www-data \
  -DZM_WEB_GROUP=www-data \
  -DCMAKE_INSTALL_SYSCONFDIR=etc/zm \
  -DZM_CONFIG_DIR=/etc/zm

connortechnology, 2019-06-27:
https://stackoverflow.com/questions/33678965/need-to-link-cmake-project-to-dl-library shed some light but we don't actually set target_link_libraries... maybe we should.

JohnAZoidberg, 2019-06-27:
Yup, that works!
Thanks for responding so quickly!

We've got all those flags because they are set for all cmake projects we package for the distro.

PR #2649 (2019-06-27): Explicitly link with libdl
Otherwise on some systems linking would fail with

  undefined reference to symbol 'dlclose@@GLIBC_2.2.5'

Fixes #2648
closed 2019-06-27 by connortechnology