ZoneMinder/zoneminder #2651: Zoneminder not respecting Storage Areas

externo6, 2019-07-03:
**Describe Your Environment**
- zoneminder 1.33.10
- zmrepo - testing
- CentOS 7.6
- Chrome stable - 75.0.3770.100

**Describe the bug**
Zoneminder is not respecting non-default storage area with alarm frames and looking in the default location (/var/lib/zoneminder/events/). Normal frames are fine.
(maybe related) Zoneminder is not showing only alarm frames when clicking on the Alarm Frames number, it infact starts the index at 1

Frames and alarm frames have the same URL

**To Reproduce**
Steps to reproduce the behavior:
1. Define a storage area
2. Set cameras to use said storage area
3. Create alarms
4. Click on events > click on alarm frame number > index starts at 1 > find an alarm frame > not shown

**Expected behavior**

Zoneminder to show the alarm frame when clicked and when clicking alarm frames to only show alarm frames.

**Debug Logs**
```
Jul 03 11:22:25 zoneminder. web_php[14587]: ERR [Got addWidth but no passed width when getting popup size for tag 'frames' - getPopupSize()]
Jul 03 11:22:25 zoneminder. web_php[14587]: ERR [Got addHeight but no passed height when getting popup size for tag 'frames' - getPopupSize()]
Jul 03 11:22:34 zoneminder. web_php[12041]: ERR [Got addWidth but no passed width when getting popup size for tag 'frame' - getPopupSize()]
Jul 03 11:22:34 zoneminder. web_php[15286]: ERR [Can't create frame images from video because there is no video file for event  at /var/lib/zoneminder/events//]
Jul 03 11:22:34 zoneminder. web_php[12041]: WAR [Adjusting to minimum width when getting popup size for tag 'frame' - getPopupSize()]
Jul 03 11:22:34 zoneminder. web_php[15286]: ERR [Capture file does not exist at /var/lib/zoneminder/events///01317-capture.jpg]
Jul 03 11:22:34 zoneminder. web_php[12041]: ERR [Got addHeight but no passed height when getting popup size for tag 'frame' - getPopupSize()]
Jul 03 11:22:34 zoneminder. web_php[15286]: FAT [Frame 1317 Not Found]

```

externo6, 2019-07-04:
Just an update, mounted my storage onto /var/lib/zoneminder/events/ to see if it would fix it, however it still shows the above logs. Somethings not quite right here.
```Capture file does not exist at /var/lib/zoneminder/events///00055-capture.jpg```
The file exists in /var/lib/zoneminder/events/4/2019-07-04/138049
```-rw-r--r-- 1 apache apache 351458 Jul  4 11:57 /var/lib/zoneminder/events/4/2019-07-04/138049/00055-capture.jpg
```

externo6, 2019-07-06:
Further testing, downgrading to package:
zoneminder-httpd-1.33.8-1.15.20190522gitgbdbbab0d3.el7.x86_64 fixes this issue. 
Upgrading to: 
zoneminder-httpd-1.33.9-1.241.20190624gitg82672d9b8.el7.x86_64 starts the problems above.

externo6, 2019-07-07:
Further testing, in httpd error logs:
``` PHP Notice:  Undefined property: ZM\\Event::$Scheme in /usr/share/zoneminder/www/includes/Event.php on line 135, referer: https://zoneminder***/zm/index.php?view=frames&page=1&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Battr%5D=FramesEventId&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bop%5D=%3D&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bval%5D=724&sort_field=FramesType&sort_asc=0&limit=
```

connortechnology, 2019-07-07:
There some very wrong stuff happening here.  It basically looks like it isn't loading the event, so various things like MonitorId, Scheme, etc aren't populated.

externo6, 2019-07-07:
Yes, I've also created a fresh Centos install and the same occured. 
I dont believe its my setup as 1.33.8 is fine, upgrading to 1.33.9 breaks things. 
Perhaps a missing dependency?

connortechnology, 2019-07-07:
I doubt it fixes it, but I did fix one small issues in the frames view.

It might help to see what the contents of the url to image are.

externo6, 2019-07-07:
Ok, I re-upgraded to 1.33.11, and I found that the scale is set to 'Scale to fit' which is the default for my monitors. Changing it to Actual fixes the problem. It seems setting a scale causes the above errors.

externo6, 2019-07-07:
> > I doubt it fixes it, but I did fix one small issues in the frames view.
> > It might help to see what the contents of the url to image are.
> 
> Ok, I re-upgraded to 1.33.11, and I found that the scale is set to 'Scale to fit' which is the default for my monitors. Changing it to Actual allows an image to show, however it shows random alarms frames from all monitors. URL after switching from scale to Actual: https://zoneminder.***/zm/index.php?view=frame&eid=&scale=100&show=capt&fid=1603
> 
> Switching from Actual to 3/4x = no image = https://zoneminder.***/zm/index.php?view=frame&eid=&fid=1613&popup=1
> 
> I'll implement your fix to see if it fixes the issues.

Ok, scratch what I said above, it shows an image but shows the wrong monitor, I think it was just pot luck I selected a frame that existed for another monitor?

I included your fix but didnt seem to help.

connortechnology, 2019-07-07:
You put me on the right path.  When scale = scale to fit, the width => 0.  So the image is there, just zero width.  This has led me down the rabbit hole a bit.  Lots of problems.  I'll have it sorted by tomorrow.

connortechnology, 2019-07-07:
although the scaling issue might be unrelated

externo6, 2019-08-03:
Hey hows the fixes coming along?

hankintosh, 2019-08-06:
While I haven't had issues with scaling, I've been investigating an issue similar to this where the eid GET variable is blank in the URL for the frames, resulting in the following error:
```
Notice: Undefined index: eid in /usr/share/zoneminder/www/skins/classic/views/frames.php on line 27
```
The trigger appears to be the sortHeader() call.  Before clicking any column to sort, the eid is present in each frame's URL and opening any frame works.  Sorting any column results in the eid missing from each frame URL when the page reloads.

I tested a "fix" by just explicitly passing the eid GET variable.  However, this isn't probably the real fix, as it looks like the FramesEventId is being passed via the filter[Query] array in the URL (I'm a little fuzzy on how this is supposed to work).

Here's what fixed my issue, and I'm hoping it leads you closer to a permanent resolution.
```
diff --git a/web/includes/functions.php b/web/includes/functions.php
index 867861ffe..05db4e881 100644
--- a/web/includes/functions.php
+++ b/web/includes/functions.php
@@ -1472,7 +1472,7 @@ function getPagination( $pages, $page, $maxShortcuts, $query, $querySep='&amp;'

 function sortHeader( $field, $querySep='&amp;' ) {
   global $view;
-  return '?view='.$view.$querySep.'page=1'.$_REQUEST['filter']['query'].$querySep.'sort_field='.$field.$querySep.'sort_asc='.($_REQUEST['sort_field'] == $field?!$_REQUEST['sort_asc']:0).$querySep.'limit='.validInt($_REQUEST['limit']);
+  return '?view='.$view.$querySep.'page=1'.$querySep.'eid='.$_REQUEST['eid'].$querySep.$_REQUEST['filter']['query'].$querySep.'sort_field='.$field.$querySep.'sort_asc='.($_REQUEST['sort_field'] == $field?!$_REQUEST['sort_asc']:0).$querySep.'limit='.validInt($_REQUEST['limit']);
 }

 function sortTag( $field ) {
```

This line of code hasn't changed much since 2016, so I'm probably far off, but if you think this is the issue I have a PR ready to go.

connortechnology, 2019-08-08:
I think you are right on.  I'm pushing that fix.  Please confirm the status of this issue once that is done.

hankintosh, 2019-08-09:
Confirmed [your commit](https://github.com/ZoneMinder/zoneminder/commit/df285006d20a7c172679db956a6170b2a550acf9) resolved my issue.  Thank you!

@externo6, hoping this helped you as well.