ZoneMinder/zoneminder #2668: Can't export or download events

APTISteve, 2019-07-22:
Using I7 intel cpu 12G RAM and 500G SSD running Fedora 29 latest kernel.
Zoneminder 1.33.12
Installed using the Repo
Fedora 29 Kernel 5.1.16-200
Firefox being used with latest update

**Describe the bug**
I am not able to export or download video. The export button does nothing and download button askes if I want tar or zip but after selection I get blank screen that never leaves. Only exporting one event that it says would be 13 meg

**To Reproduce**
Steps to reproduce the behavior:
1. From console click on one of the event counts such as under week or month
2. Place a check mark next to the desired event
3. Click export (nothing happens)
4. Click download
5. select "zip" then "generate download"
6. watch the pretty white screen for the rest of the day.
7. No error messages generated so nothing to show under debug.

**Expected behavior**
I would expect under export to be asked the normal how to arrange and package the download. under download I would be expecting to tell it where to download.

**Debug Logs**
```

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

closed 2019-07-24 by connortechnology

APTISteve, 2019-07-24:
thanks for closing this but where is the fix?

connortechnology, 2019-07-24:
If you look up a bit there is a link to the PR with the code.

commit 547e80b1 (2019-07-24): Fix download and export button from events list. Fixes #2668 (#2670)