ZoneMinder/zoneminder #2673: 1.32.3: cURL allow to capture 1 frame per minute

onlyjob, 2019-07-28:
Some online traffic cameras refresh every minute.
However ZoneMinder (1.32.3 | from-source | Debian 10 "Buster") is unable(?) to slow-capture e.g. 1 frame per minute when FPS is set to `0.017` (1/60):

~~~~
zms: Terminating, last frame sent time 45.004968 secs more than maximum of 10.000000
~~~~

Why maximum of `10` is enforced? Would it be reasonable to override it?

**To Reproduce**
Source type: `cURL`
FPS: `0.017`  (1/60)

closed 2019-07-31 by connortechnology

onlyjob, 2019-08-31:
Thank you very much @connortechnology.

IMHO this warrants a minor release. I can not cherry pick onto current release (1.32.3) due to DB schema changes...

connortechnology, 2019-08-31:
There are a lot of fixes since 1.32.3 that warrant a release.  However that release will be 1.34.

I am simply not capable of doing everything that needs to be done.  So I am focussing on getting 1.34 out which I might not be capable of doing either.

PR #2675 (2019-07-31): Fix 2673
Fixes #2673 
increases max_last_send_frame_time to handle cameras with <1fps.