ZoneMinder/zoneminder #2674: apt upgrade error

joedaimarg, 2019-07-31:
**Describe Your Environment**
- Version of ZoneMinder [v1.33.12]
- How you installed ZoneMinder [PPA]
- Full name and version of OS Linux Mint 19.1 Cinnamon
- Linux Kernel - 5.0.0-20-generic

**Describe the bug**
An error is produced when running apt upgrade

**To Reproduce**
Steps to reproduce the behavior:
1. apt update
2. apt upgrade
3. the error will show up every time even if there's no update for zoneminder

**Debug Logs**
```

Errors were encountered while processing:
 zoneminder
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

connortechnology, 2019-07-31:
I think there are previous errors that you didn't see or are not reporting.  I believe the db did not get updated.

run
sudo zmupdate.pl
sudo zmupdate.pl -f
sudo apt-get -f install

joedaimarg, 2019-08-05:
@connortechnology found this error after running the code

```
Setting up zoneminder (1.33.13~20190731171807-bionic) ...
Detected db service is mariadb.service
mysqladmin: connect to server at 'localhost' failed
error: 'Access denied for user 'root'@'localhost' (using password: NO)'
Creating zm db
ERROR 1045 (28000): Access denied for user 'root'@'localhost' (using password: NO)
Error creating db.
dpkg: error processing package zoneminder (--configure):
 installed zoneminder package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 zoneminder
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
Is there a zoneminder config that I can update to fix this? or should I always remove my database password every time I update zm?

connortechnology, 2019-08-16:
The issue here is that it can't connect to the database. Are you using a remote database server?  
On debian based systems, there is typically a file in /etc/mysql/debian.cnf which allows root to configure the db.  It appears that you have disabled that file or perhaps mint doesn't use it.  Long story short though, if root can't talk to the db, then it can't do the update.

closed 2019-08-22 by joedaimarg