ZoneMinder/zoneminder #2677: suggested repository results in package issues

dabeegmon, 2019-08-04:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]

I never achieved a successful installation - - - - my fault as I really would prefer to use Postgresql than Mariadb so I did not complete the installation.

- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]

used:
https://wiki.zoneminder.com/Debian_10_Buster_with_Zoneminder_1.32.3_the_Easy_Way
for directions

- Full name and version of OS

Above instructions suggest using the deb-multimedia repository for the Debian packages. 
Doing this can create issues especially when its time to upgrade (I found the issue when I was upgrading my system from Buster to Bullseye).
The packages in the mentioned repository use a 'dmo' suffix in their names. 
When its time to update the package the general repository does not connect and because a package is present in 'dmo' version no update occurs. This results in a twisted system. 

Correcting this meant removing all 'dmo' packages (took some time cause i'm no real programmer!) and then I reinstalled versions of the removed programs from the general repository. 

These difficulties make it feel like the deb-multimedia repository isn't really 'Debian' and that is problematic imo. Now to a very experienced sysadmin this is likely hardly worth the time of day but I'd bet that not everyone installing is an experienced sysadmin.

knight-of-ni, 2019-08-07:
Personally, I completely agree with you that the deb-multimedia repo should be avoided. That repo is not an official Debian repo. 

However, the issue you describe concerns the ZoneMinder Wiki rather than our content on github or readthedocs. The ZoneMinder Wiki is maintained by the community and writeable by anyone who wishes to add content to it. Please discuss any issues concerning the wiki either in the user forum or contact the users who created or last edit the wiki page in question. Feel free to make changes to it.

closed 2019-08-07 by knight-of-ni

dabeegmon, 2019-08-08:
On Wed, Aug 7, 2019 at 6:25 PM Andrew Bauer <notifications@github.com>
wrote:

> Personally, I completely agree with you that the deb-multimedia repo
> should be avoided. That repo is not an official Debian repo.
>
> However, the issue you describe concerns the ZoneMinder Wiki rather than
> our content on github or readthedocs. The ZoneMinder Wiki is maintained by
> the community and writeable by anyone who wishes to add content to it.
> Please discuss any issues concerning the wiki either in the user forum or
> contact the users who created or last edit the wiki page in question. Feel
> free to make changes to it.
>
>
> Would love to do that but upon attempt I was locked out of the forums so
thought of letting the community know that there are issues with the
presented information.
Oh well - - - - guess that someone else gets to also learn not to use
deb-multimedia as a repo the hard way.

Regards

knight-of-ni, 2019-08-08:
What is your user forum account name?
Were you locked out by our spam filter? That happens from time to time.

dabeegmon, 2019-08-08:
On Wed, Aug 7, 2019 at 8:02 PM Andrew Bauer <notifications@github.com>
wrote:

> What is your user forum account name?
>
ajoeiam

> Were you locked out by our spam filter? That happens from time to time.
>
I start getting confused because of the huge number of variants in what I'm
allowed to use for a password.
Sometimes no letters (only numeric), sometimes only letters and numbers no
caps, sometimes caps included, sometimes a small set of special characters
and sometimes all the characters on a 'normal' (US 105) keyboard are
allowed.

Just know that it keeps telling that I exceeded the maximum number of
attempts and then I am supposed to do recatpcha and then it never works so
dunno what the issue is.
Makes it real easy to forget that I wanted to use the software when I can't
access the forum board associated with it.

Regards

>
>

knight-of-ni, 2019-08-08:
I have emailed you a new password.

As a sysadmin, I am sure you are familiar with our password complexity rules. The rules are 6-100 characters, mixed case, with at least one number. That's fairly typical these days.

If you run into any future problems related to the user forum, please either email us (if you can) or  reach out to us in our our Slack channel.