ZoneMinder/zoneminder #2682: Monitor segfault (11) when image buffer increased

hgregorian, 2019-08-10:
**Environment Info.**
- zoneminder:1.33.14 installed via PPA
- Memory: 192 GB RAM
- CPU: Dual CPU Xeon (16 cores each, 32 total)
- OS:
```
NAME="Ubuntu"
VERSION="18.04.1 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.1 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
```

**Camera / Monitor Info.**
- Hikvision DS2CD2185FWD-I
- 20 fps
- 3840x2160
- RTSP (TCP)

**Bug / Reproduction**
ZM 1.33.x introduces analyzation of stream and provides a recommendation (via log warnings) to increase image buffer based on FPS and keyframe frequency.  Following recommendations, a stable monitor is made unstable/unusable when image buffer is greater than ~120, given camera details above.  To be fair, when I had initially started to use ZM about 9 months ago (1.32.x), I experimented with image buffer and found that the monitor became unstable/unusable under the same circumstances.  This issue was initially mentioned in the following forum post: https://forums.zoneminder.com/viewtopic.php?f=39&t=28502

**Expected behavior**
Stable monitor so long as adequate /dev/shm is available.

**Debug Logs**
[zm_debug.log.00422](https://github.com/ZoneMinder/zoneminder/files/3488704/zm_debug.log.00422.txt) - expected monitor behavior (image buffer set to 60)
[zm_debug.log.02003](https://github.com/ZoneMinder/zoneminder/files/3488703/zm_debug.log.02003.txt) - crashing monitor (image buffer set to 160)
[zm_debug.log.02068](https://github.com/ZoneMinder/zoneminder/files/3488702/zm_debug.log.02068.txt) - crashing monitor (image buffer set to 160)
[zm_debug.log.02077](https://github.com/ZoneMinder/zoneminder/files/3488701/zm_debug.log.02077.txt) - crashing monitor (image buffer set to 160)
[zm_debug.log.02140](https://github.com/ZoneMinder/zoneminder/files/3488700/zm_debug.log.02140.txt) - crashing monitor (image buffer set to 160)

connortechnology, 2019-08-10:
SO I was investigating the line size issue, turns out it is a pointer to an array, not an integer so that explains why it is different. I will be pushing a fix for that in a moment.
Just a note, your zmc.mmap will be over 5Gb which given the amount of ram shouldn't be an issue.
I have cranked up the buffers on my system here, but I only have 720p cameras so I've had to crank them pretty high to achieve a similar ram pressure.  No luck so far.
Next step might be running it under valgrind.  Are you up for that? Here are the commands to do so
# 1 stop normal zmc process
sudo zmdc.pl stop zmwatch.pl
sudo zmdc.pl stop zmc -m 16
# 2 start our own under valgrind
sudo su -s/bin/bash -c"valgrind -v zmc -m 16" www-data

hgregorian, 2019-08-10:
Image Buffer 60 (user terminated after ~2 minutes)
```
==7487== Memcheck, a memory error detector
==7487== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==7487== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==7487== Command: zmc -m 16
==7487== 
--7487-- Valgrind options:
--7487--    -v
--7487-- Contents of /proc/version:
--7487--   Linux version 3.10.0-957.12.1.el7.x86_64 (mockbuild@kbuilder.bsys.centos.org) (gcc version 4.8.5 20150623 (Red Hat 4.8.5-36) (GCC) ) #1 SMP Mon Apr 29 14:59:59 UTC 2019
--7487-- 
--7487-- Arch and hwcaps: AMD64, LittleEndian, amd64-cx16-lzcnt-rdtscp-sse3-avx-avx2-bmi
--7487-- Page sizes: currently 4096, max supported 4096
--7487-- Valgrind library directory: /usr/lib/valgrind
--7487-- Reading syms from /usr/bin/zmc
--7487--    object doesn't have a symbol table
--7487-- Reading syms from /lib/x86_64-linux-gnu/ld-2.27.so
--7487--   Considering /lib/x86_64-linux-gnu/ld-2.27.so ..
--7487--   .. CRC mismatch (computed 1b7c895e wanted 2943108a)
--7487--   Considering /usr/lib/debug/lib/x86_64-linux-gnu/ld-2.27.so ..
--7487--   .. CRC is valid
--7487-- Reading syms from /usr/lib/valgrind/memcheck-amd64-linux
--7487--   Considering /usr/lib/valgrind/memcheck-amd64-linux ..
--7487--   .. CRC mismatch (computed c25f395c wanted 0a9602a8)
--7487--    object doesn't have a symbol table
--7487--    object doesn't have a dynamic symbol table
--7487-- Scheduler: using generic scheduler lock implementation.
--7487-- Reading suppressions file: /usr/lib/valgrind/default.supp
==7487== embedded gdbserver: reading from /tmp/vgdb-pipe-from-vgdb-to-7487-by-www-data-on-477b23dada95
==7487== embedded gdbserver: writing to   /tmp/vgdb-pipe-to-vgdb-from-7487-by-www-data-on-477b23dada95
==7487== embedded gdbserver: shared mem   /tmp/vgdb-pipe-shared-mem-vgdb-7487-by-www-data-on-477b23dada95
==7487== 
==7487== TO CONTROL THIS PROCESS USING vgdb (which you probably
==7487== don't want to do, unless you know exactly what you're doing,
==7487== or are doing some strange experiment):
==7487==   /usr/lib/valgrind/../../bin/vgdb --pid=7487 ...command...
==7487== 
==7487== TO DEBUG THIS PROCESS USING GDB: start GDB like this
==7487==   /path/to/gdb zmc
==7487== and then give GDB the following command
==7487==   target remote | /usr/lib/valgrind/../../bin/vgdb --pid=7487
==7487== --pid is optional if only one valgrind process is running
==7487== 
--7487-- REDIR: 0x401f2f0 (ld-linux-x86-64.so.2:strlen) redirected to 0x58060901 (???)
--7487-- REDIR: 0x401f0d0 (ld-linux-x86-64.so.2:index) redirected to 0x5806091b (???)
--7487-- Reading syms from /usr/lib/valgrind/vgpreload_core-amd64-linux.so
--7487--   Considering /usr/lib/valgrind/vgpreload_core-amd64-linux.so ..
--7487--   .. CRC mismatch (computed 4b63d83e wanted 670599e6)
--7487--    object doesn't have a symbol table
--7487-- Reading syms from /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so
--7487--   Considering /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so ..
--7487--   .. CRC mismatch (computed a4b37bee wanted 8ad4dc94)
--7487--    object doesn't have a symbol table
==7487== WARNING: new redirection conflicts with existing -- ignoring it
--7487--     old: 0x0401f2f0 (strlen              ) R-> (0000.0) 0x58060901 ???
--7487--     new: 0x0401f2f0 (strlen              ) R-> (2007.0) 0x04c32db0 strlen
--7487-- REDIR: 0x401d360 (ld-linux-x86-64.so.2:strcmp) redirected to 0x4c33ee0 (strcmp)
--7487-- REDIR: 0x401f830 (ld-linux-x86-64.so.2:mempcpy) redirected to 0x4c374f0 (mempcpy)
--7487-- Reading syms from /lib/x86_64-linux-gnu/librt-2.27.so
--7487--   Considering /lib/x86_64-linux-gnu/librt-2.27.so ..
--7487--   .. CRC mismatch (computed 16979484 wanted f9e041e3)
--7487--   Considering /usr/lib/debug/lib/x86_64-linux-gnu/librt-2.27.so ..
--7487--   .. CRC is valid
--7487-- Reading syms from /lib/x86_64-linux-gnu/libz.so.1.2.11
--7487--    object doesn't have a symbol table
--7487-- Reading syms from /usr/lib/x86_64-linux-gnu/libcurl-gnutls.so.4.5.0
--7487--    object doesn't have a symbol table
--7487-- Reading syms from /usr/lib/x86_64-linux-gnu/libjpeg.so.8.1.2
--7487--    object doesn't have a symbol table
[... 892 lines cut ...]
==8034== 
==8034== LEAK SUMMARY:
==8034==    definitely lost: 0 bytes in 0 blocks
==8034==    indirectly lost: 0 bytes in 0 blocks
==8034==      possibly lost: 28,038 bytes in 70 blocks
==8034==    still reachable: 105,767,816 bytes in 1,530 blocks
==8034==                       of which reachable via heuristic:
==8034==                         newarray           : 1,536 bytes in 16 blocks
==8034==         suppressed: 0 bytes in 0 blocks
==8034== Rerun with --leak-check=full to see details of leaked memory
==8034== 
==8034== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
==8034== 
==8034== 1 errors in context 1 of 1:
==8034== Invalid write of size 8
==8034==    at 0x8BD3622: ??? (in /usr/lib/x86_64-linux-gnu/libswscale.so.4.8.100)
==8034==    by 0x8BAB2BD: sws_scale (in /usr/lib/x86_64-linux-gnu/libswscale.so.4.8.100)
==8034==    by 0x2070D3: FfmpegCamera::transfer_to_image(Image&, AVFrame*, AVFrame*) (in /usr/bin/zmc)
==8034==    by 0x209BBA: FfmpegCamera::CaptureAndRecord(Image&, timeval, char*) (in /usr/bin/zmc)
==8034==    by 0x156438: Monitor::Capture() (in /usr/bin/zmc)
==8034==    by 0x12D049: main (in /usr/bin/zmc)
==8034==  Address 0x964c5000 is not stack'd, malloc'd or (recently) free'd
==8034== 
==8034== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
```

connortechnology, 2019-08-10:
Good news!  I can reproduce it!  All I have to do is set my res that high with buffers that high and it crashes!  1920x1080 doesn't do it.  Seems to be totally resolution dependent.  I can set my buffers to over 1000 and it doesn't crash.

connortechnology, 2019-08-10:
Also, it seems to do it reliably at frame 32.

hgregorian, 2019-08-10:
Hey hey, that’s great!  Thanks for looking into this.

connortechnology, 2019-08-12:
So I figured it out.  Integer overflow when calculating the shared mem size.  camera->ImageSize was an int, so the additions were done as 32bit int.  We were allocating 1Gb when we needed 5Gb.

closed 2019-08-12 by connortechnology

hgregorian, 2019-08-12:
Looks good on my end.  Thanks so much!

laviddichterman, 2020-01-19:
Howdy! I'm getting a similar issue when the image buffer is increased on my Dahua IPC-HDBW4631R-S

I get the following warning:
`2020-01-18 05:36:26.026153 zmc_m4[25213].WAR-zm_ffmpeg_camera.cpp/951 [ImageBufferCount 20 is too small. Needs to be at least 751. Either increase it or decrease time between keyframes]`

Here's my ffmpeg info:
`ffmpeg version N-96402-gab9627223e Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.2.1-9ubuntu2)
  configuration: --enable-nonfree --enable-nvenc --enable-libx264 --enable-gpl --enable-cuda --enable-cuvid --enable-cuda-nvcc --enable-libnpp
  libavutil      56. 38.100 / 56. 38.100
  libavcodec     58. 66.100 / 58. 66.100
  libavformat    58. 35.103 / 58. 35.103
  libavdevice    58.  9.103 / 58.  9.103
  libavfilter     7. 71.100 /  7. 71.100
  libswscale      5.  6.100 /  5.  6.100
  libswresample   3.  6.100 /  3.  6.100
  libpostproc    55.  6.100 / 55.  6.100
Hyper fast Audio and Video encoder
`

So I bumped the "Image Buffer Size (frames)" to 40 and the following happens when I tried to follow the above debugging advice.
Here's the valgrind output:
`lavid@wcp-zeus:/var/log/zm$ zmc --version
1.34.0
lavid@wcp-zeus:/var/log/zm$ sudo zmdc.pl stop zmwatch.pl
'zmwatch.pl' sending stop to pid 21961 at 20/01/18 21:32:18
lavid@wcp-zeus:/var/log/zm$ sudo zmdc.pl stop zmc -m 4
Command 'zmc -m 4' removed from pending list at 20/01/18 21:32:36
lavid@wcp-zeus:/var/log/zm$ sudo su -s/bin/bash -c"valgrind -v zmc -m 4" www-data
==28061== Memcheck, a memory error detector
==28061== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==28061== Using Valgrind-3.15.0-608cb11914-20190413 and LibVEX; rerun with -h for copyright info
==28061== Command: zmc -m 4
==28061== 
--28061-- Valgrind options:
--28061--    -v
--28061-- Contents of /proc/version:
--28061--   Linux version 5.3.0-26-generic (buildd@lgw01-amd64-013) (gcc version 9.2.1 20191008 (Ubuntu 9.2.1-9ubuntu2)) #28-Ubuntu SMP Wed Dec 18 05:37:46 UTC 2019
--28061-- 
--28061-- Arch and hwcaps: AMD64, LittleEndian, amd64-cx16-lzcnt-rdtscp-sse3-ssse3-avx-avx2-bmi-f16c-rdrand
--28061-- Page sizes: currently 4096, max supported 4096
--28061-- Valgrind library directory: /usr/lib/x86_64-linux-gnu/valgrind
--28061-- Reading syms from /usr/bin/zmc
--28061--    object doesn't have a symbol table
--28061-- Reading syms from /lib/x86_64-linux-gnu/ld-2.30.so
--28061--   Considering /lib/x86_64-linux-gnu/ld-2.30.so ..
--28061--   .. CRC mismatch (computed ff246dee wanted d35d27d8)
--28061--   Considering /lib/x86_64-linux-gnu/ld-2.30.so ..
--28061--   .. CRC mismatch (computed ff246dee wanted d35d27d8)
--28061--   Considering /usr/lib/debug/lib/x86_64-linux-gnu/ld-2.30.so ..
--28061--   .. CRC is valid
--28061-- Reading syms from /usr/lib/x86_64-linux-gnu/valgrind/memcheck-amd64-linux
--28061--    object doesn't have a symbol table
--28061--    object doesn't have a dynamic symbol table
--28061-- Scheduler: using generic scheduler lock implementation.
--28061-- Reading suppressions file: /usr/lib/x86_64-linux-gnu/valgrind/default.supp
==28061== embedded gdbserver: reading from /tmp/vgdb-pipe-from-vgdb-to-28061-by-www-data-on-???
==28061== embedded gdbserver: writing to   /tmp/vgdb-pipe-to-vgdb-from-28061-by-www-data-on-???
==28061== embedded gdbserver: shared mem   /tmp/vgdb-pipe-shared-mem-vgdb-28061-by-www-data-on-???
==28061== 
==28061== TO CONTROL THIS PROCESS USING vgdb (which you probably
==28061== don't want to do, unless you know exactly what you're doing,
==28061== or are doing some strange experiment):
==28061==   /usr/lib/x86_64-linux-gnu/valgrind/../../bin/vgdb --pid=28061 ...command...
==28061== 
==28061== TO DEBUG THIS PROCESS USING GDB: start GDB like this
==28061==   /path/to/gdb zmc
==28061== and then give GDB the following command
==28061==   target remote | /usr/lib/x86_64-linux-gnu/valgrind/../../bin/vgdb --pid=28061
==28061== --pid is optional if only one valgrind process is running
[... 448 lines cut ...]
==28061==    by 0x189CBB: Monitor::Load(unsigned int, bool, Monitor::Purpose) (in /usr/bin/zmc)
==28061==    by 0x15C6CA: main (in /usr/bin/zmc)
--28061-- Discarding syms at 0x48505c0-0x4856a1c in /lib/x86_64-linux-gnu/libnss_files-2.30.so (have_dinfo 1)
==28061== 
==28061== HEAP SUMMARY:
==28061==     in use at exit: 25,717,020 bytes in 7,699 blocks
==28061==   total heap usage: 9,523 allocs, 1,824 frees, 26,237,090 bytes allocated
==28061== 
==28061== Searching for pointers to 7,699 not-freed blocks
==28061== Checked 809,830,912 bytes
==28061== Skipped 363,130,880 bytes due to read errors
==28061== 
==28061== LEAK SUMMARY:
==28061==    definitely lost: 0 bytes in 0 blocks
==28061==    indirectly lost: 0 bytes in 0 blocks
==28061==      possibly lost: 77,224 bytes in 49 blocks
==28061==    still reachable: 25,639,796 bytes in 7,650 blocks
==28061==                       of which reachable via heuristic:
==28061==                         newarray           : 1,536 bytes in 16 blocks
==28061==         suppressed: 0 bytes in 0 blocks
==28061== Rerun with --leak-check=full to see details of leaked memory
==28061== 
==28061== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
Bus error (core dumped)
`

commit c1984ad7 (2019-08-12): Fix problem calculating mem_size using an int from ImageSize.  With camera resolutions going up, width*height*colour could exceed 32bits.  So use a guarnteed 53bit type, which fixes the memsize calculations.  Fixes #2682