ZoneMinder/zoneminder #2683: Zoneminder starts in stopped state

bhearsum, 2019-08-10:
**Describe Your Environment**
- 1.32.3
- PPA
- Ubuntu 19.04

**Describe the bug**
Whenever Zoneminder is restarted, it starts in a stopped state. I can't tell if it's actually the `Stopped` state, or the `default` state, but in either case, none of my Monitors are capturing anything until I manually go in and switch the state.

**To Reproduce**
Steps to reproduce the behavior:
1. Restart Zoneminder (or the computer it's running on)
2. Observe that Monitors are all stopped

**Expected behavior**
Zoneminder should start in the same state it was in when it was shut down.

connortechnology, 2019-10-10:
If you have installed zoneminder following the directions, then you will have done 
sudo systemctl enable zoneminder
Which will start zoneminder at system boot and systemd will attempt to keep it going.

bhearsum, 2019-10-11:
It's not the Zoneminder service itself that's stopped -- all of the processes are running. It's the "State" of Zoneminder. It's in "Stopped" (as in, Monitors are all off), and I have to use either the web interface or zmninja to switch it to a state that's actually got Monitors enabled.

connortechnology, 2019-10-11:
The ZM web UI gets it's status by talk to the running zmdc.pl process through a socket.  This socket in a single server install lives at /run/zm/zmdc.sock.  In a multi-server install it might be at /run/zm/zmdc1.sock or zmdc2.sock or whatever.  The number corresponds to the Id in the Servers table. 

Not being able to talk to zmdc.pl through this file can happen due to a misconfigured multi-server install, or an install with old zm files in places like /usr/local/ where the various bits and pieces are looking in different places.  

In any event you will want to upgrade to the latest master, because that will be the next release.

sudo add-apt-repository ppa:iconnor/zoneminder-master
sudo apt-get update
sudo apt-get dist-upgrade

closed 2019-11-04 by connortechnology

acsawdey, 2022-03-21:
I was hitting this problem and the solution turned out to be `systemctl enable zoneminder` ... it seems if it is not started at boot, the web UI will show "Stopped" as the state and you can start it from the web UI.