ZoneMinder/zoneminder #2684: View of event do not work

tomasrudh, 2019-08-11:
Zoneminder was installed on Raspbian 9 (Debian) that was later upgraded to Raspbian 10, and that is when the problems started.

**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
1.32.3-2
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
Following: https://zoneminder.readthedocs.io/en/1.32.3/installationguide/debian.html#easy-way-debian-jessie
- Full name and version of OS
Linux 4.19.58-v7+ #1245 SMP Fri Jul 12 17:25:51 BST 2019 armv7l GNU/Linux
Raspbian GNU/Linux 10 (buster)

**Describe the bug**
When clicking on an event does it take a long time before a picture shows, and that is a still, none of the buttons below work. Live stream works.

**To Reproduce**
Steps to reproduce the behavior:
1. View an event

**Expected behavior**
Should show a video of the event, without errors in the log

**Debug Logs**
```
08/10/19 17:35:22.287111 zmc_m1[3953].DB4-zm_remote_camera_http.cpp/276 [ÿؿā^]
08/10/19 17:35:22.287139 zmc_m1[3953].DB4-zm_remote_camera_http.cpp/993 [getting more data]
08/10/19 17:35:22.317405 zmc_m1[3953].DB3-zm_remote_camera_http.cpp/252 [Just getting 9048]
08/10/19 17:35:22.317639 zmc_m1[3953].DB3-zm_remote_camera_http.cpp/255 [Expecting 9048 bytes]
08/10/19 17:35:22.317915 zmc_m1[3953].DB3-zm_remote_camera_http.cpp/271 [Read 9048 bytes]
08/10/19 17:35:22.317994 zmc_m1[3953].DB4-zm_remote_camera_http.cpp/276 [ÿؿā^]
08/10/19 17:35:22.318032 zmc_m1[3953].DB3-zm_remote_camera_http.cpp/1001 [Got end of image by length, content-length = 61154]
08/10/19 17:35:22.318063 zmc_m1[3953].DB3-zm_remote_camera_http.cpp/1060 [Returning 61154 bytes, buffer size: (61176) bytes of captured content]
08/10/19 17:35:22.328441 zmc_m1[3953].DB4-zm_monitor.cpp/2400 [Return from Capture (1)]
08/10/19 17:35:22.328644 zmc_m1[3953].DB6-zm_remote_camera_http.cpp/905 [20: --video boundary--
Content-type: image/jpeg]
08/10/19 17:35:22.328680 zmc_m1[3953].DB4-zm_remote_camera_http.cpp/913 [Got boundary subheader '--video boundary--
Content-type: image/jpeg']
08/10/19 17:35:22.328711 zmc_m1[3953].DB3-zm_remote_camera_http.cpp/951 [Unable to extract subheader from stream, retrying]
08/10/19 17:35:22.347680 zmc_m1[3953].DB3-zm_remote_camera_http.cpp/252 [Just getting 104]
08/10/19 17:35:22.347865 zmc_m1[3953].DB3-zm_remote_camera_http.cpp/255 [Expecting 104 bytes]
08/10/19 17:35:22.348015 zmc_m1[3953].DB3-zm_remote_camera_http.cpp/271 [Read 104 bytes]
08/10/19 17:35:22.348164 zmc_m1[3953].ERR-zm_signal.cpp/50 [Got signal 11 (Segmentation fault), crashing]
```

rhaamo, 2019-09-05:
It's the case with 1.33.14 too.

I'm getting one but on recording events, which is probably your case too rather than playback, the segfault causes the .mp4 to be unreadable, then the event cannot be played back at all (except the dynamic jpeg thumbnail):
```
root@fenouil:~/zm# cat /var/log/zm/zmc_m5.log
09/05/19 07:17:00.443958 zmc_m5[24965].INF-zm_ffmpeg.cpp/81 [Not enabling ffmpeg logs, as LOG_FFMPEG and/or LOG_DEBUG is disabled in options, or this monitor not part of your debug targets]
09/05/19 07:17:00.796123 zmc_m5[24965].INF-zmc.cpp/223 [Starting Capture version 1.33.14]
09/05/19 07:17:00.859478 zmc_m5[24965].INF-zm_ffmpeg_camera.cpp/205 [Priming capture from rtsp://192.168.10.21:554/user=admin_password=tlJwpbo6_channel=1_stream=0.sdp?real_stream]
09/05/19 07:17:02.870190 zmc_m5[24965].INF-zm_monitor.cpp/2539 [Entree porte bois: images:1 - Capturing at 0.50 fps, capturing bandwidth 134923bytes/sec]
09/05/19 07:17:46.954599 zmc_m5[24965].INF-zm_monitor.cpp/2539 [Entree porte bois: images:1000 - Capturing at 22.73 fps, capturing bandwidth 310185bytes/sec]
09/05/19 07:18:32.393503 zmc_m5[24965].INF-zm_monitor.cpp/2539 [Entree porte bois: images:2000 - Capturing at 21.74 fps, capturing bandwidth 303298bytes/sec]
09/05/19 07:19:17.834842 zmc_m5[24965].INF-zm_monitor.cpp/2539 [Entree porte bois: images:3000 - Capturing at 22.22 fps, capturing bandwidth 311427bytes/sec]
09/05/19 07:20:03.309104 zmc_m5[24965].INF-zm_monitor.cpp/2539 [Entree porte bois: images:4000 - Capturing at 21.74 fps, capturing bandwidth 296962bytes/sec]
09/05/19 07:20:48.764024 zmc_m5[24965].INF-zm_monitor.cpp/2539 [Entree porte bois: images:5000 - Capturing at 22.22 fps, capturing bandwidth 308356bytes/sec]
09/05/19 07:21:34.191384 zmc_m5[24965].INF-zm_monitor.cpp/2539 [Entree porte bois: images:6000 - Capturing at 21.74 fps, capturing bandwidth 302206bytes/sec]
09/05/19 07:22:19.689787 zmc_m5[24965].INF-zm_monitor.cpp/2539 [Entree porte bois: images:7000 - Capturing at 22.22 fps, capturing bandwidth 311571bytes/sec]
09/05/19 07:23:05.106753 zmc_m5[24965].INF-zm_monitor.cpp/2539 [Entree porte bois: images:8000 - Capturing at 21.74 fps, capturing bandwidth 298110bytes/sec]
09/05/19 07:23:50.589710 zmc_m5[24965].INF-zm_monitor.cpp/2539 [Entree porte bois: images:9000 - Capturing at 22.22 fps, capturing bandwidth 308985bytes/sec]
09/05/19 07:24:36.029273 zmc_m5[24965].INF-zm_monitor.cpp/2539 [Entree porte bois: images:10000 - Capturing at 21.74 fps, capturing bandwidth 305934bytes/sec]
09/05/19 07:24:49.994632 zmc_m5[24965].INF-zm_videostore.cpp/61 [Opening video storage stream /var/cache/zoneminder/events/5/2019-09-05/232105/232105-video.mp4 format: mp4]
09/05/19 07:24:51.536266 zmc_m5[24965].ERR-zm_signal.cpp/52 [Got signal 11 (Segmentation fault), crashing]
09/05/19 07:24:51.588825 zmc_m5[24965].ERR-zm_signal.cpp/83 [Signal address is 0x10, from 0x7f864751bfbf]
09/05/19 07:24:51.708078 zmc_m5[24965].ERR-zm_signal.cpp/104 [Backtrace 0: /usr/bin/zmc(_Z14zm_die_handleriP9siginfo_tPv+0x7a) [0x56147bd7ef8a]]
09/05/19 07:24:51.755697 zmc_m5[24965].ERR-zm_signal.cpp/104 [Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0x13f40) [0x7f8649bb7f40]]
09/05/19 07:24:51.839361 zmc_m5[24965].ERR-zm_signal.cpp/104 [Backtrace 2: /lib/x86_64-linux-gnu/libgcc_s.so.1(_Unwind_Resume+0xcf) [0x7f864751bfbf]]
09/05/19 07:24:51.881104 zmc_m5[24965].ERR-zm_signal.cpp/104 [Backtrace 3: /usr/bin/zmc(+0x3a3bb) [0x56147bcff3bb]]
09/05/19 07:24:51.922620 zmc_m5[24965].ERR-zm_signal.cpp/104 [Backtrace 4: /usr/bin/zmc(_ZN10VideoStore21writeVideoFramePacketEP8AVPacket+0x1a4) [0x56147bd95214]]
09/05/19 07:24:51.972639 zmc_m5[24965].ERR-zm_signal.cpp/104 [Backtrace 5: /usr/bin/zmc(_ZN12FfmpegCamera16CaptureAndRecordER5Image7timevalPc+0x1e3d) [0x56147bddcadd]]
09/05/19 07:24:52.014301 zmc_m5[24965].ERR-zm_signal.cpp/104 [Backtrace 6: /usr/bin/zmc(_ZN7Monitor7CaptureEv+0x4d9) [0x56147bd3fa49]]
09/05/19 07:24:52.056087 zmc_m5[24965].ERR-zm_signal.cpp/104 [Backtrace 7: /usr/bin/zmc(main+0xb9a) [0x56147bd1b1ea]]
09/05/19 07:24:52.097662 zmc_m5[24965].ERR-zm_signal.cpp/104 [Backtrace 8: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xeb) [0x7f8647346b6b]]
09/05/19 07:24:52.181143 zmc_m5[24965].ERR-zm_signal.cpp/104 [Backtrace 9: /usr/bin/zmc(_start+0x2a) [0x56147bd1d2aa]]
09/05/19 07:24:52.272819 zmc_m5[24965].INF-zm_signal.cpp/111 [Backtrace complete, please execute the following command for more information]
09/05/19 07:24:52.356266 zmc_m5[24965].INF-zm_signal.cpp/112 [addr2line -e /usr/bin/zmc 0x56147bd7ef8a 0x7f8649bb7f40 0x7f864751bfbf 0x56147bcff3bb 0x56147bd95214 0x56147bddcadd 0x56147bd3fa49 0x56147bd1b1ea 0x7f8647346b6b 0x56147bd1d2aa]
```

ffprobe on file:
```
root@fenouil:~/zm# ffprobe /var/cache/zoneminder/events/5/2019-09-05/232105/232105-video.mp4
ffprobe version 4.1.3-0ubuntu1 Copyright (c) 2007-2019 the FFmpeg developers
  built with gcc 8 (Ubuntu 8.3.0-6ubuntu1)
  configuration: --prefix=/usr --extra-version=0ubuntu1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enab [... cut ...]
  libavcodec     58. 35.100 / 58. 35.100
  libavformat    58. 20.100 / 58. 20.100
  libavdevice    58.  5.100 / 58.  5.100
  libavfilter     7. 40.101 /  7. 40.101
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  3.100 /  5.  3.100
  libswresample   3.  3.100 /  3.  3.100
  libpostproc    55.  3.100 / 55.  3.100
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x5597da2ddfc0] moov atom not found
/var/cache/zoneminder/events/5/2019-09-05/232105/232105-video.mp4: Invalid data found when processing input
root@fenouil:~/zm#
```

inpos, 2019-09-08:
@rhaamo, same issue.
Debian 10 amd64. Zoneminder from git (2019/09/07)

inpos, 2019-09-08:
If I enable "x264 encode" there is no audio stream.
```
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/var/cache/zoneminder/events/1/2019-09-08/4/4-video.mp4':
  Metadata:
    major_brand     : mp42
    minor_version   : 0
    compatible_brands: mp42isom
    creation_time   : 2019-09-08T10:57:58.000000Z
  Duration: 00:10:00.05, start: 0.310000, bitrate: 385 kb/s
    Stream #0:0(und): Video: h264 (Main) (avc1 / 0x31637661), yuv420p, 1280x720, 382 kb/s, 15 fps, 16.67 tbr, 1k tbn, 2k tbc (default)
    Metadata:
      creation_time   : 2019-09-08T10:57:58.000000Z
      encoder         : JVT/AVC Coding
```

connortechnology, 2019-09-08:
Nor should there be.

The next step as always is turn on debug logging!

I have spent the weekend setting up a debian 10 dev environment, I should be able to figure this out soon.

connortechnology, 2019-09-08:
I believe I have fixed the crash on debian buster.  Please update and let us know.

inpos, 2019-09-09:
No, segfault again:
```
сен 09 08:14:03 zm zmdc[1524]: INF [Starting pending process, zmc -m 1]
сен 09 08:14:03 zm zmdc[13776]: INF ['zmc -m 1' started at 19/09/09 08:14:03]
сен 09 08:14:03 zm zmdc[1524]: INF ['zmc -m 1' starting at 19/09/09 08:14:03, pid = 13776]
сен 09 08:14:03 zm zmc_m1[13776]: INF [zmc_m1] [Not enabling ffmpeg logs, as LOG_FFMPEG and/or LOG_DEBUG is disabled in options, or this monitor not part of your debug targets]
сен 09 08:14:03 zm zma_m1[13770]: INF [zma_m1] [cam04: image_count:20 - Closing event 5, shutting down]
сен 09 08:14:03 zm zmdc[1524]: INF ['zma -m 1' exited normally]
сен 09 08:14:03 zm zmc_m1[13776]: INF [zmc_m1] [Starting Capture version 1.33.14]
сен 09 08:14:03 zm zmc_m1[13776]: INF [zmc_m1] [Priming capture from rtsp://user:pass@10.89.89.101:554/cam/realmonitor?channel=1&subtype=0]
сен 09 08:14:05 zm zmc_m1[13776]: INF [zmc_m1] [cam04: images:1 - Capturing at 0.50 fps, capturing bandwidth 44779bytes/sec]
сен 09 08:14:08 zm zmdc[1524]: INF [Starting pending process, zma -m 1]
сен 09 08:14:10 zm zmdc[13777]: INF ['zma -m 1' started at 19/09/09 08:14:10]
сен 09 08:14:10 zm zmdc[1524]: INF ['zma -m 1' starting at 19/09/09 08:14:10, pid = 13777]
сен 09 08:14:11 zm zmc_m1[13776]: INF [zmc_m1] [cam04: images:100 - Capturing at 20.00 fps, capturing bandwidth 246656bytes/sec]
сен 09 08:14:11 zm zma_m1[13777]: INF [zma_m1] [In mode 4/1, warming up]
сен 09 08:14:11 zm zma_m1[13777]: INF [zma_m1] [cam04: 001 - Opening new event 6, section start]
сен 09 08:14:11 zm zmc_m1[13776]: INF [zmc_m1] [Opening video storage stream /var/cache/zoneminder/events/1/2019-09-09/6/6-video.mp4 format: mp4]
сен 09 08:14:11 zm zmc_m1[13776]: ERR [zmc_m1] [Got signal 11 (Segmentation fault), crashing]
сен 09 08:14:11 zm zmc_m1[13776]: ERR [zmc_m1] [Signal address is 0x10, from 0x7ff48f1b665f]
сен 09 08:14:11 zm zmc_m1[13776]: ERR [zmc_m1] [Backtrace 0: /usr/bin/zmc(_Z14zm_die_handleriP9siginfo_tPv+0x77) [0x556cece1f0a7]]
сен 09 08:14:11 zm zmc_m1[13776]: ERR [zmc_m1] [Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0x12730) [0x7ff4915d6730]]
сен 09 08:14:11 zm zmc_m1[13776]: ERR [zmc_m1] [Backtrace 2: /lib/x86_64-linux-gnu/libgcc_s.so.1(_Unwind_Resume+0xcf) [0x7ff48f1b665f]]
сен 09 08:14:11 zm zmc_m1[13776]: ERR [zmc_m1] [Backtrace 3: /usr/bin/zmc(+0x3a583) [0x556cecd9f583]]
сен 09 08:14:11 zm zmc_m1[13776]: ERR [zmc_m1] [Backtrace 4: /usr/bin/zmc(_ZN10VideoStore21writeVideoFramePacketEP8AVPacket+0x1a4) [0x556cece35314]]
сен 09 08:14:11 zm zmc_m1[13776]: ERR [zmc_m1] [Backtrace 5: /usr/bin/zmc(_ZN12FfmpegCamera16CaptureAndRecordER5Image7timevalPc+0x1e3d) [0x556cece7c86d]]
сен 09 08:14:11 zm zmc_m1[13776]: ERR [zmc_m1] [Backtrace 6: /usr/bin/zmc(_ZN7Monitor7CaptureEv+0x4a9) [0x556cecddf979]]
сен 09 08:14:11 zm zmc_m1[13776]: ERR [zmc_m1] [Backtrace 7: /usr/bin/zmc(main+0xb9a) [0x556cecdbb47a]]
сен 09 08:14:11 zm zmc_m1[13776]: ERR [zmc_m1] [Backtrace 8: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xeb) [0x7ff48f00609b]]
сен 09 08:14:11 zm zmc_m1[13776]: ERR [zmc_m1] [Backtrace 9: /usr/bin/zmc(_start+0x2a) [0x556cecdbd51a]]
сен 09 08:14:11 zm zmc_m1[13776]: INF [zmc_m1] [Backtrace complete, please execute the following command for more information]
сен 09 08:14:11 zm zmc_m1[13776]: INF [zmc_m1] [addr2line -e /usr/bin/zmc 0x556cece1f0a7 0x7ff4915d6730 0x7ff48f1b665f 0x556cecd9f583 0x556cece35314 0x556cece7c86d 0x556cecddf979 0x556cecdbb47a 0x7ff48f00609b 0x556cecdbd51a]
сен 09 08:14:11 zm zmdc[1524]: ERR ['zmc -m 1' exited abnormally, exit status 11]
```

rhaamo, 2019-09-09:
Just updated, triggering an alarm still causes a segfault, I believe this is the right debug log:
```log
09/09/19 08:04:42.998632 zmc_m5[4843].DB1-zm_logger.cpp/251 [LogOpts: level=DB1/DB1, screen=OFF, database=INF, logfile=DB1->/var/log/zm/zm_debug.log.04843, syslog=INF]
09/09/19 08:04:42.998702 zmc_m5[4843].DB1-zm_utils.cpp/287 [Detected a x86\x86-64 processor with SSE4.2]
09/09/19 08:04:42.999748 zmc_m5[4843].DB1-zm_monitor.cpp/2089 [Capture Delay!? 0.000]
09/09/19 08:04:42.999774 zmc_m5[4843].DB1-zm_monitor.cpp/2111 [Got 1 for v4l_captures_per_frame]
09/09/19 08:04:42.999823 zmc_m5[4843].INF-zm_ffmpeg.cpp/79 [Enabling ffmpeg logs, as LOG_DEBUG+LOG_FFMPEG are enabled in options]
09/09/19 08:04:43.043286 zmc_m5[4843].DB1-zm_monitor.cpp/416 [monitor purpose=1]
09/09/19 08:04:43.043331 zmc_m5[4843].DB1-zm_monitor.cpp/430 [mem.size(8) SharedData=600 TriggerData=560 VideoStoreData=4128 timestamps=1424 images=89x2764800 = 246067200 total=246073976]
09/09/19 08:04:43.043354 zmc_m5[4843].DB1-zm_storage.cpp/94 [No id passed to Storage constructor.  Using default path /var/cache/zoneminder/events instead]
09/09/19 08:04:43.043373 zmc_m5[4843].DB1-zm_monitor.cpp/434 [Storage path: /var/cache/zoneminder/events]
09/09/19 08:04:43.044085 zmc_m5[4843].DB1-zm_monitor.cpp/584 [Unable to map file /dev/shm/zm.mmap.5 (246073976 bytes) to locked memory, trying unlocked]
09/09/19 08:04:43.044114 zmc_m5[4843].DB1-zm_monitor.cpp/587 [Mapped file /dev/shm/zm.mmap.5 (246073976 bytes) to unlocked memory]
09/09/19 08:04:43.217572 zmc_m5[4843].DB1-zm_monitor.cpp/507 [Monitor Entree porte bois has function 3,
label format = '%N - %d/%m/%y %H:%M:%S', label X = 0, label Y = 0, label size = 1,
image buffer count = 89, warmup count = 25, pre-event count = 25, post-event count = 25, alarm frame count = 1,
fps report interval = 1000, ref blend percentage = 12, alarm ref blend percentage = 6, track motion = 0]
09/09/19 08:04:43.218212 zmc_m5[4843].DB1-zm_zone.cpp/884 [Got 2 zones for monitor Entree porte bois]
09/09/19 08:04:43.227805 zmc_m5[4843].DB1-zm_monitor.cpp/2384 [Loaded monitor 5(Entree porte bois), 2 zones]
09/09/19 08:04:43.227836 zmc_m5[4843].INF-zmc.cpp/223 [Starting Capture version 1.33.14]
09/09/19 08:04:43.269963 zmc_m5[4843].INF-zm_ffmpeg_camera.cpp/205 [Priming capture from rtsp://192.168.10.21:554/user=admin_password=tlJwpbo6_channel=1_stream=0.sdp?real_stream]
09/09/19 08:04:43.311063 zmc_m5[4843].DB1-zm_ffmpeg_camera.cpp/358 [Calling avformat_open_input for rtsp://192.168.10.21:554/user=admin_password=tlJwpbo6_channel=1_stream=0.sdp?real_stream]
09/09/19 08:04:44.103489 zmc_m5[4843].WAR-zm_ffmpeg.cpp/67 [Too short data for FU-A H.264 RTP packet
]
09/09/19 08:04:44.813863 zmc_m5[4843].DB1-zm_ffmpeg_camera.cpp/451 [Failed to find decoder (h264_mmal)]
09/09/19 08:04:44.813926 zmc_m5[4843].DB1-zm_ffmpeg.cpp/331 [Dumping stream index i(0) index(0)]
09/09/19 08:04:44.813953 zmc_m5[4843].DB1-zm_ffmpeg.cpp/341 [    Stream #0:0]
09/09/19 08:04:44.813973 zmc_m5[4843].DB1-zm_ffmpeg.cpp/352 [frames:28, frame_size:0 stream timebase: 1/90000]
09/09/19 08:04:44.814018 zmc_m5[4843].DB1-zm_ffmpeg.cpp/354 [codec: Video: h264 (Main), 1 reference frame, yuv420p(progressive, left), 1280x720, 1/44]
09/09/19 08:04:44.814043 zmc_m5[4843].DB1-zm_ffmpeg.cpp/284 [22 fps]
09/09/19 08:04:44.814068 zmc_m5[4843].DB1-zm_ffmpeg.cpp/286 [90k stream tb numerator]
09/09/19 08:04:44.814293 zmc_m5[4843].DB1-zm_ffmpeg.cpp/325 [Dumping codec_context codec_type(0) codec_id(27) width(1280) height(720)  timebase(1/44) format(yuv420p)
gop_size 12 max_b_frames 0 me_cmp 0 me_range 0 qmin 2 qmax 31]
09/09/19 08:04:44.814324 zmc_m5[4843].DB1-zm_ffmpeg_camera.cpp/553 [HWACCEL not in use]
09/09/19 08:04:44.857705 zmc_m5[4843].DB1-zm_ffmpeg_camera.cpp/968 [raw frame from decoder: format 0 yuv420p 1280x720 linesize:1280x640 pts: 21110]
09/09/19 08:04:44.858396 zmc_m5[4843].DB1-zm_ffmpeg_camera.cpp/1098 [Setup conversion context for 1280x720 yuv420p to 1280x720 rgb24]
09/09/19 08:04:44.859941 zmc_m5[4843].INF-zm_monitor.cpp/2539 [Entree porte bois: images:1 - Capturing at 1.00 fps, capturing bandwidth 279219bytes/sec]
09/09/19 08:04:44.915147 zmc_m5[4843].DB1-zm_ffmpeg_camera.cpp/968 [raw frame from decoder: format 0 yuv420p 1280x720 linesize:1280x640 pts: 3600]
09/09/19 08:04:44.918526 zmc_m5[4843].DB1-zm_ffmpeg_camera.cpp/968 [raw frame from decoder: format 0 yuv420p 1280x720 linesize:1280x640 pts: 7200]
09/09/19 08:04:44.922017 zmc_m5[4843].DB1-zm_ffmpeg_camera.cpp/968 [raw frame from decoder: format 0 yuv420p 1280x720 linesize:1280x640 pts: 10800]
09/09/19 08:04:44.925907 zmc_m5[4843].DB1-zm_ffmpeg_camera.cpp/968 [raw frame from decoder: format 0 yuv420p 1280x720 linesize:1280x640 pts: 14400]
09/09/19 08:04:44.933276 zmc_m5[4843].DB1-zm_ffmpeg_camera.cpp/968 [raw frame from decoder: format 0 yuv420p 1280x720 linesize:1280x640 pts: 18000]
09/09/19 08:04:44.938990 zmc_m5[4843].DB1-zm_ffmpeg_camera.cpp/968 [raw frame from decoder: format 0 yuv420p 1280x720 linesize:1280x640 pts: 21600]
09/09/19 08:04:44.943890 zmc_m5[4843].DB1-zm_ffmpeg_camera.cpp/968 [raw frame from decoder: format 0 yuv420p 1280x720 linesize:1280x640 pts: 25200]
09/09/19 08:04:44.948341 zmc_m5[4843].DB1-zm_ffmpeg_camera.cpp/968 [raw frame from decoder: format 0 yuv420p 1280x720 linesize:1280x640 pts: 28800]
09/09/19 08:04:44.955016 zmc_m5[4843].DB1-zm_ffmpeg_camera.cpp/968 [raw frame from decoder: format 0 yuv420p 1280x720 linesize:1280x640 pts: 32400]
[... cut ...]
09/09/19 08:05:01.462999 zmc_m5[4843].INF-zm_signal.cpp/111 [Backtrace complete, please execute the following command for more information]
09/09/19 08:05:01.496400 zmc_m5[4843].INF-zm_signal.cpp/112 [addr2line -e /usr/bin/zmc 0x55e679b9bfca 0x7f4c51e0df40 0x7f4c4f5b9ed7 0x7f4c4f59b535 0x7f4c4f602726 0x7f4c4f60959a 0x7f4c4f60b3cc 0x55e679b1c3b3 0x55e679bb2274 0x55e679bf988d 0x55e679b5ca89 0x55e679b3822a 0x7f4c4f59cb6b 0x55e679b3a2ea]
```

Just to mention, the addr2line only returns `??:0` and is totally useless.

connortechnology, 2019-09-09:
Debug level needs to be 3.
Yeah addr2line hasn't been useful for years to do address space randomization.
I'm surprised that this is happening on ubuntu 19.04, as that is what I run on, and it has been rock solid for me there.

rhaamo, 2019-09-09:
yeah, as said in the other thread, my previous zm version worked fine.

will run another trace with debug level at 3 tomorrow if that can help.

rhaamo, 2019-09-10:
@connortechnology so hmm, I've just rebuilt ZM, and now it works. I guess your commits between my last build and now fixed it. so for now my case seems solved. will see if there is any event broken tomorrow.

connortechnology, 2019-09-10:
Awesome! I was hoping that would be the case.  So for those who want to know... the problem was that in the VideoStore object we had a member called ret which we used to store the return value from ffmpeg functions.  In one method write_packet, I also defined it as a local variable.  Apparently this confused gcc.  You would think the compile could at least detect and warn.  Maybe it does if we use the right compiler flags.  Anyways, ret really shouldn't be an object member so now it isn't and the crash doesn't happen anymore.

closed 2019-09-11 by connortechnology