ZoneMinder/zoneminder #2685: MySQL 8 reserved names (follow up of #2659)

EdwinGH, 2019-08-13:
I'm using Fedora 29 with MySQL Community 8.0 and Zoneminder 1.33.14 from https://zmrepo.zoneminder.com/f/29/x86_64/

I upgraded to Fedora 29 and then MySQL upgraded from 5.7 to 8.0 and Zoneminder stopped working. I got many SQL errors, and one was related to the use of 'Group', which is a reserved word in MySQL 8 (see https://dev.mysql.com/doc/refman/8.0/en/keywords.html )
This was handled in issue #2659 

The next error I get is related to 'Function' being a reserved keyword according to https://dev.mysql.com/doc/refman/8.0/en/keywords.html 
I get the following errors:
SQL-ERR 'SQLSTATE[42000]: Syntax error or access violation: 1064 You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'Function' at line 1', statement was 'describe Monitors Function' params:

SQL-ERR dbFetchOne no result, statement was 'describe Monitors Function'

I see Function is used as a column name in the Monitors table...
Can this be changed to a different name (pre or postfixing may solve) or isolated by back-tics in the code?
For example:
mysql> describe Monitors Function;
ERROR 1064 (42000): You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'Function' at line 1
mysql> describe Monitors `Function`;
+----------+------------------------------------------------------------+------+-----+---------+-------+
| Field    | Type                                                       | Null | Key | Default | Extra |
+----------+------------------------------------------------------------+------+-----+---------+-------+
| Function | enum('None','Monitor','Modect','Record','Mocord','Nodect') | NO   |     | Monitor |       |
+----------+------------------------------------------------------------+------+-----+---------+-------+
1 row in set (0.01 sec)

EdwinGH, 2019-08-14:
After manually adding backticks to database.php, I get to the next error (which I can't edit as I'm not compiling from source myself):
Can't run query: You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'Function+0, Enabled, LinkedMonitors, AnalysisFPSLimit, AnalysisUpdateDelay, MaxF' at line 1 | zm_db.cpp | 102
Next use of reserved keyword Function...

connortechnology, 2019-08-14:
I have added backticks to the c source.  Give it a try.

EdwinGH, 2019-08-15:
Thank you for implementing the changes for my setup. I'll have to wait until the zmrepo makes a new build for my system. I will check this evening, and let you know if it works.Thanks again! Edwin 

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/ZoneMinder/zoneminder/issues/2685?email_source=notifications\u0026email_token=AIYSTHSNVIOKACZ5GC6IPY3QERSHHA5CNFSM4ILK2TOKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD4J7UIA#issuecomment-521402912",
"url": "https://github.com/ZoneMinder/zoneminder/issues/2685?email_source=notifications\u0026email_token=AIYSTHSNVIOKACZ5GC6IPY3QERSHHA5CNFSM4ILK2TOKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD4J7UIA#issuecomment-521402912",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

EdwinGH, 2019-08-16:
Next issue with reserved name Function:
Can't run query: You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'Function != 'None' AND Type = 'Local' AND Device='/dev/video0'' at line 1 | zm_db.cpp | 102

connortechnology, 2019-08-17:
I just fixed up a few more that I missed, like that one.

EdwinGH, 2019-08-25:
We're almost there; I have the system running again.
Still found few more issues with Function:

2019-08-25 16:00:56 | zmpkg |   | 13025 | FAT | Can't execute: You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'Function='None', Enabled='0' WHERE Id='1'' at line 1 | zmpkg.pl

2019-08-25 16:00:36 | web_php |   | 8263 | ERR | SQL-ERR 'SQLSTATE[42000]: Syntax error or access violation: 1064 You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'Function=?, Enabled=? WHERE Id=?' at line 1', statement was 'UPDATE Monitors SET Function=?, Enabled=? WHERE Id=?' params:None,1,2 | /usr/share/zoneminder/www/includes/database.php | 158

EdwinGH, 2019-09-16:
Isaac, others, I still have the issue with MySQL keyword violation when  enabling or disabling monitors:

2019-09-16 16:37:10 | web_php |   | 25948 | ERR | SQL-ERR 'SQLSTATE[42000]: Syntax error or access violation: 1064 You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'Function=?, Enabled=? WHERE Id=?' at line 1', statement was 'UPDATE Monitors SET Function=?, Enabled=? WHERE Id=?' params:Modect,1,1 | /usr/share/zoneminder/www/includes/database.php | 158

I checked the error location, but database.php line 158 is just the error reporting location. The real problem seems to be in Monitor.php  line 459, but my limited PHP skills do not enable me to do a right array explode with the required back-ticks...
I found https://stackoverflow.com/questions/51044040/adding-backtick-to-column-name-when-building-mysql-query-from-a-string but cannot get it to work in the Monitor,php...
Can someone please help?

connortechnology, 2019-09-16:
Well that one is easy enough to fix.  The cakephp enums are still on the todo list.

dsl, 2019-10-21:
What's left to do to close this one off?

connortechnology, 2019-10-21:
cakephp doesn't handle enums.  We use a module to handle them but it isn't doing the escaping right.

bhearsum, 2019-10-24:
FYI: This issue currently makes it very difficult to run Zoneminder on Ubuntu 19.10, which has no easy way of using MySQL 5.7.

bkjaya2020, 2019-11-03:
I have managed to install zoneminder master latest packge with mariadb  on ubuntu 19.10  please  refer www.bkjaya.wordpress.com

connortechnology, 2019-11-30:
I believe I have fixed the remaining issue to do with enums.  Please test and close this if everytihng is indeed working.

EdwinGH, 2019-12-08:
Hi Isaac, the state change is still not working as indicated in my last comment: https://github.com/ZoneMinder/zoneminder/issues/2685#issuecomment-531811017 .
The system is operational, but I use the state changes to enable and disable cameras when at home or not. This state change is not working; neither in the perl files nor the web interface.

connortechnology, 2019-12-08:
Ok try latest master I pushed a change for that

EdwinGH, 2019-12-08:
Just tried it and your fix works. Next issue with Function I found in state.php line 34:
$sql = 'SELECT Id,Function,Enabled FROM Monitors ORDER BY Id';
I changed that in my local copy with the backquotes and the system now seems to be running nicely with state changes:
$sql = 'SELECT `Id`,`Function`,`Enabled` FROM Monitors ORDER BY `Id`';

connortechnology, 2019-12-09:
Ok, applied those fixes as well

EdwinGH, 2019-12-30:
System seems fully operational now on MySQL 8. Thank you!

closed 2019-12-30 by EdwinGH

bkjaya2020, 2020-01-02:
Hi EdwinGH
If you have successfully installed , could you kindly publish  a tutorial in zoneminder wiki  ,on using MySQL 8. with zoneminder ?

bartbk, 2020-05-15:
I still have an error on Ubuntu 20.04 with mysql  Ver 8.0.20-0ubuntu0.20.04.1 for Linux on x86_64 ((Ubuntu)) just by adding group:
web_php		154854	ERR	SQL-ERR 'SQLSTATE[42000]: Syntax error or access violation: 1064 You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'Groups (`Id`, `Name`, `ParentId`) VALUES (?, ?, ?)' at line 1', statement was 'INSERT INTO Groups (`Id`, `Name`, `ParentId`) VALUES (?, ?, ?)' params:,Przodek,	/usr/share/zoneminder/www/includes/database.php	152

bkjaya2020, 2020-05-17:
I have managed to install Zoneminder v1.34.12. with mysql
8.0.20-0ubuntu0.20.04.1 on Ubuntu 20.04 , without any problem
Please see the following link for more details on installation
https://bkjaya.wordpress.com/2020/05/10/how-to-install-zoneminder-v1-34-12-on-ubuntu-20-04-lts-focal-fossa/

bartbk, 2020-05-17:
Hi,

 

Can you add Groups with your config?

 

Kind regards,

Bart

 

https://bkjaya.wordpress.com/2020/05/10/how-to-install-zoneminder-v1-34-12-on-ubuntu-20-04-lts-focal-fossa/

bkjaya2020, 2020-05-18:
Hi Bart
I haven't used the groups option earlier . It seems ,adding groups does not
work in my configuration.

 I can install Zoneminder v1.34.12. with mysql 8.0.20-0ubuntu0.20.04.1 on
Ubuntu 20.04 ,without any  problem .
But still there is an error (SQL-ERR 'SQLSTATE[42S22] ) when you try to add
a new monitor to Zoneminder -master v1.35.4
Best regards
B.K.J

bartbk, 2020-05-18:
My installation works completely OK – apart this adding of Groups – I think that is Error caused by reserved name in MySQL.

I do not have any errors when I tray to add monitors at v. 1.34.14