ZoneMinder/zoneminder #2689: Browser does not remained logged into console

neillbell, 2019-08-29:
**Describe Your Environment**
- Version of ZoneMinder: 1.33.14 (1.33.14~20190828122014-bionic)
- How you installed ZoneMinder: PPA
- Ubuntu 18.04.3 LTS
- Browser name and version: Chromium Version 76.0.3809.100

**Describe the bug**
The browser does not remain logged into the console indefinitely.

**To Reproduce**
Steps to reproduce the behavior:
1. Log into the console
2. Wait a while (It took a little over an hour last time, but I noticed that the last two times it happened, I saw an ERR in web_php.log at 9 and 12 local time)
3. Return to browser and see it is at the login screen

**Expected behavior**
Previously, the browser would remain logged into the console indefinitely.

**Debug Logs**
First time today:
```
08/29/19 07:38:13.189686 web_php[11846].INF [Login successful for user "neill"] at /usr/share/zoneminder/www/includes/actions/login.php line 78
08/29/19 07:56:49.790315 web_php[23457].ERR [Please do not use refresh tokens for this operation] at /usr/share/zoneminder/www/includes/auth.php line 133
08/29/19 07:56:49.794963 web_php[23457].INF [Creating token for "neill"] at /usr/share/zoneminder/www/api/app/Controller/HostController.php line 121
08/29/19 08:28:27.389660 web_php[1761].ERR [Socket /var/run/zm/zms-801496s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.] at /usr/share/zoneminder/www/includes/functions.php line 2170
08/29/19 08:56:49.829939 web_php[9559].ERR [Please do not use refresh tokens for this operation] at /usr/share/zoneminder/www/includes/auth.php line 133
08/29/19 08:56:49.833667 web_php[9559].INF [Creating token for "neill"] at /usr/share/zoneminder/www/api/app/Controller/HostController.php line 121
08/29/19 09:00:18.559392 web_php[2137].ERR [Unable to authenticate user from auth hash '087868fea2f1946b2be36f75df1917da'] at /usr/share
```

Second time today:
```
08/29/19 10:52:05.336173 web_php[18671].INF [Login successful for user "neill"] at /usr/share/zoneminder/www/includes/actions/login.php line 78
08/29/19 10:56:36.919580 web_php[14909].ERR [Please do not use refresh tokens for this operation] at /usr/share/zoneminder/www/includes/auth.php line 133
08/29/19 10:56:36.926015 web_php[14909].INF [Creating token for "neill"] at /usr/share/zoneminder/www/api/app/Controller/HostController.php line 121
08/29/19 11:56:36.965234 web_php[30620].ERR [Please do not use refresh tokens for this operation] at /usr/share/zoneminder/www/includes/auth.php line 133
08/29/19 11:56:36.970533 web_php[30620].INF [Creating token for "neill"] at /usr/share/zoneminder/www/api/app/Controller/HostController.php line 121
08/29/19 12:00:04.555051 web_php[9061].ERR [Unable to authenticate user from auth hash '16b337ca0b2c0450257b22af6eb0cea8'] at /usr/share/zoneminder/www/includes/auth.php line 195

```

pliablepixels, 2019-09-01:
Do you have zmNinja running as well? The token part should never be called by the web console.

neillbell, 2019-09-01:
Yes. I probably should have turned it off to avoid confusion.

neillbell, 2019-09-01:
Here's a log without zmNinja running. There are still some requests for notification images.

[web_php.log](https://github.com/ZoneMinder/zoneminder/files/3563784/web_php.log)

pliablepixels, 2019-09-01:
Assuming it is not a system/PHP/DB time issue, you seem to be getting logged out every few minutes. this almost looks like we are inadvertently destroying the session somewhere (PHP layer or API layer, doesn't matter, as its a super global). I'm not sure if we are debug logging all invocations of session destroy, but we should. @connortechnology thoughts?

pliablepixels, 2019-09-06:
This should be fixed in master. Isaac put in various fixes and it seems to be behaving for me. Can you confirm?

neillbell, 2019-09-07:
This seems to be fixed. I'm closing it. Thanks Isaac!

closed 2019-09-07 by neillbell

synthead, 2023-12-19:
I am still experiencing this on 1.36.33.  It seems like Firefox logs me out of the front-end after a couple minutes.