ZoneMinder/zoneminder #2691: Frequent "Unable to send packet Resource temporarily unavailable, continuing" messages

hgregorian, 2019-08-30:
**Environment Info.**
- zoneminder:1.33.14 installed via PPA
- ffmpeg:
```
ffmpeg version 3.4.6-0ubuntu0.18.04.1 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 7 (Ubuntu 7.3.0-16ubuntu3)
  configuration: --prefix=/usr --extra-version=0ubuntu0.18.04.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-librsvg --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared
  libavutil      55. 78.100 / 55. 78.100
  libavcodec     57.107.100 / 57.107.100
  libavformat    57. 83.100 / 57. 83.100
  libavdevice    57. 10.100 / 57. 10.100
  libavfilter     6.107.100 /  6.107.100
  libavresample   3.  7.  0 /  3.  7.  0
  libswscale      4.  8.100 /  4.  8.100
  libswresample   2.  9.100 /  2.  9.100
  libpostproc    54.  7.100 / 54.  7.100
Hyper fast Audio and Video encoder
usage: ffmpeg [options] [[infile options] -i infile]... {[outfile options] outfile}...
```
- Memory: 192 GB RAM
- CPU: Dual CPU Xeon (16 cores each, 32 total)
- OS:
```
NAME="Ubuntu"
VERSION="18.04.1 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.1 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
```

**Camera / Monitor Info.**
- Hikvision DS2CD2185FWD-I
- 20 fps
- 3840x2160
- RTSP (TCP)
- bitrate: 13500 Kbps

**Bug / Reproduction**
Given the above camera setup, intermittent warnings/errors relating to "Resource temporarily unavailable/Unable to receive frame" are encountered.  Reducing the bitrate to ~8000 Kbps nearly eliminates these messages, and further reducing to ~6000 Kbps appears to eliminate them completely, however this comes at a cost of quality.  Errors continue to occur if this is the only monitor that's enabled, so that appears to eliminate resource constraints.  I'm also unable to reproduce anything remotely like these errors in CLI capture of the stream using ffmpeg.

**Expected behavior**
Error free monitor capture.

**Debug Logs**
```
08/29/19 23:53:18.769537 zmc_m21[445].DB1-zm_ffmpeg_camera.cpp/957 [raw frame from decoder: format 12 yuvj420p 3840x2160 linesize:3840x1920 pts: 197491770]
08/29/19 23:53:18.875177 zmc_m21[445].DB1-zm_ffmpeg_camera.cpp/957 [raw frame from decoder: format 12 yuvj420p 3840x2160 linesize:3840x1920 pts: 197496270]
08/29/19 23:53:18.959284 zmc_m21[445].DB1-zm_ffmpeg_camera.cpp/957 [raw frame from decoder: format 12 yuvj420p 3840x2160 linesize:3840x1920 pts: 197500770]
08/29/19 23:53:19.026586 zmc_m21[445].WAR-zm_ffmpeg.cpp/67 [error while decoding MB 159 115, bytestream -7
]
08/29/19 23:53:19.031509 zmc_m21[445].DB1-zm_ffmpeg.cpp/67 [concealing 4690 DC, 4690 AC, 4690 MV errors in P frame
]
08/29/19 23:53:19.146874 zmc_m21[445].DB1-zm_ffmpeg_camera.cpp/957 [raw frame from decoder: format 12 yuvj420p 3840x2160 linesize:3840x1920 pts: 197505270]
08/29/19 23:53:19.211765 zmc_m21[445].ERR-zm_ffmpeg.cpp/491 [Unable to send packet Resource temporarily unavailable, continuing]
08/29/19 23:53:19.214354 zmc_m21[445].WAR-zm_ffmpeg_camera.cpp/946 [Unable to receive frame 32636: Resource temporarily unavailable. error count is 0]
08/29/19 23:53:19.272351 zmc_m21[445].ERR-zm_ffmpeg.cpp/491 [Unable to send packet Resource temporarily unavailable, continuing]
08/29/19 23:53:19.275059 zmc_m21[445].WAR-zm_ffmpeg_camera.cpp/946 [Unable to receive frame 32636: Resource temporarily unavailable. error count is 0]
08/29/19 23:53:19.321906 zmc_m21[445].ERR-zm_ffmpeg.cpp/491 [Unable to send packet Resource temporarily unavailable, continuing]
08/29/19 23:53:19.324549 zmc_m21[445].WAR-zm_ffmpeg_camera.cpp/946 [Unable to receive frame 32636: Resource temporarily unavailable. error count is 0]
08/29/19 23:53:19.381503 zmc_m21[445].ERR-zm_ffmpeg.cpp/491 [Unable to send packet Resource temporarily unavailable, continuing]
08/29/19 23:53:19.384215 zmc_m21[445].WAR-zm_ffmpeg_camera.cpp/946 [Unable to receive frame 32636: Resource temporarily unavailable. error count is 0]
08/29/19 23:53:19.434149 zmc_m21[445].ERR-zm_ffmpeg.cpp/491 [Unable to send packet Resource temporarily unavailable, continuing]
08/29/19 23:53:19.436902 zmc_m21[445].WAR-zm_ffmpeg_camera.cpp/946 [Unable to receive frame 32636: Resource temporarily unavailable. error count is 0]
[... 25 lines cut ...]
08/29/19 23:54:06.184913 zmc_m21[445].ERR-zm_ffmpeg.cpp/491 [Unable to send packet Resource temporarily unavailable, continuing]
08/29/19 23:54:06.187695 zmc_m21[445].WAR-zm_ffmpeg_camera.cpp/946 [Unable to receive frame 33285: Resource temporarily unavailable. error count is 0]
08/29/19 23:54:06.261887 zmc_m21[445].ERR-zm_ffmpeg.cpp/491 [Unable to send packet Resource temporarily unavailable, continuing]
08/29/19 23:54:06.264438 zmc_m21[445].WAR-zm_ffmpeg_camera.cpp/946 [Unable to receive frame 33285: Resource temporarily unavailable. error count is 0]
08/29/19 23:54:06.325681 zmc_m21[445].ERR-zm_ffmpeg.cpp/491 [Unable to send packet Resource temporarily unavailable, continuing]
08/29/19 23:54:06.328370 zmc_m21[445].WAR-zm_ffmpeg_camera.cpp/946 [Unable to receive frame 33285: Resource temporarily unavailable. error count is 0]
08/29/19 23:54:06.384466 zmc_m21[445].ERR-zm_ffmpeg.cpp/491 [Unable to send packet Resource temporarily unavailable, continuing]
08/29/19 23:54:06.387182 zmc_m21[445].WAR-zm_ffmpeg_camera.cpp/946 [Unable to receive frame 33285: Resource temporarily unavailable. error count is 0]
08/29/19 23:54:06.441584 zmc_m21[445].ERR-zm_ffmpeg.cpp/491 [Unable to send packet Resource temporarily unavailable, continuing]
08/29/19 23:54:06.444434 zmc_m21[445].WAR-zm_ffmpeg_camera.cpp/946 [Unable to receive frame 33285: Resource temporarily unavailable. error count is 0]
08/29/19 23:54:06.493175 zmc_m21[445].ERR-zm_ffmpeg.cpp/491 [Unable to send packet Resource temporarily unavailable, continuing]
08/29/19 23:54:06.495733 zmc_m21[445].WAR-zm_ffmpeg_camera.cpp/946 [Unable to receive frame 33285: Resource temporarily unavailable. error count is 0]
08/29/19 23:54:06.545838 zmc_m21[445].ERR-zm_ffmpeg.cpp/491 [Unable to send packet Resource temporarily unavailable, continuing]
08/29/19 23:54:06.548837 zmc_m21[445].WAR-zm_ffmpeg_camera.cpp/946 [Unable to receive frame 33285: Resource temporarily unavailable. error count is 0]
08/29/19 23:54:06.597066 zmc_m21[445].ERR-zm_ffmpeg.cpp/491 [Unable to send packet Resource temporarily unavailable, continuing]
08/29/19 23:54:06.604130 zmc_m21[445].WAR-zm_ffmpeg_camera.cpp/946 [Unable to receive frame 33285: Resource temporarily unavailable. error count is 0]
08/29/19 23:54:06.668685 zmc_m21[445].DB1-zm_ffmpeg_camera.cpp/957 [raw frame from decoder: format 12 yuvj420p 3840x2160 linesize:3840x1920 pts: 201919770]
08/29/19 23:54:06.745547 zmc_m21[445].DB1-zm_ffmpeg_camera.cpp/957 [raw frame from decoder: format 12 yuvj420p 3840x2160 linesize:3840x1920 pts: 201924270]
08/29/19 23:54:06.810626 zmc_m21[445].DB1-zm_ffmpeg_camera.cpp/957 [raw frame from decoder: format 12 yuvj420p 3840x2160 linesize:3840x1920 pts: 201928770]
08/29/19 23:54:06.868563 zmc_m21[445].DB1-zm_ffmpeg_camera.cpp/957 [raw frame from decoder: format 12 yuvj420p 3840x2160 linesize:3840x1920 pts: 201933270]
08/29/19 23:54:06.931940 zmc_m21[445].DB1-zm_ffmpeg_camera.cpp/957 [raw frame from decoder: format 12 yuvj420p 3840x2160 linesize:3840x1920 pts: 201937770]
08/29/19 23:54:06.989819 zmc_m21[445].DB1-zm_ffmpeg_camera.cpp/957 [raw frame from decoder: format 12 yuvj420p 3840x2160 linesize:3840x1920 pts: 201942270]
08/29/19 23:54:07.058878 zmc_m21[445].DB1-zm_ffmpeg_camera.cpp/957 [raw frame from decoder: format 12 yuvj420p 3840x2160 linesize:3840x1920 pts: 201946770]
08/29/19 23:54:07.150061 zmc_m21[445].DB1-zm_ffmpeg_camera.cpp/957 [raw frame from decoder: format 12 yuvj420p 3840x2160 linesize:3840x1920 pts: 201951270]
```

connortechnology, 2019-09-02:
Do you have values in maxFPS and alarm  max fps fields?  If so clear them.  They used to be necessary but now only cause problems by starving the codecs for cpu.  

You seem to have debug on, but not ffmpeg debug logs.  Maybe try turning them on.

hgregorian, 2019-09-02:
I haven't had the FPS fields populated for some time after seeing your same explanation in other threads.

How do I enable FFmpeg debug logs?  I've got LOG_FFMPEG enabled, but I'm not seeing anything that looks like FFmpeg debug in the LOGGING options.  Could you point me in the right direction?

connortechnology, 2019-09-02:
LOG_FFMPEG should be the option directly above LOG_DEBUG in the options

hgregorian, 2019-09-02:
I do have LOG_FFMPEG enabled, perhaps I need to bump up the LOG_DEBUG_LEVEL?

hgregorian, 2019-09-03:
Bumped LOG_DEBUG_LEVEL to 5...

[zm_debug.log.00460.gz](https://github.com/ZoneMinder/zoneminder/files/3567720/zm_debug.log.00460.gz)

connortechnology, 2019-09-03:
This isn't a ZM problem per se, not much I'm going to be able to do about it. 

We should test with a newer ffmpeg.

hgregorian, 2019-11-23:
Mostly an FYI I suppose, but just today, I've tested latest master snapshot (20191022) of zoneminder along with the following ffmpeg 4.x version:

```
root@4bf5f44dc797:/# ffmpeg -version
ffmpeg version 4.2.1-0york0~18.04.1 Copyright (c) 2000-2019 the FFmpeg developers
built with gcc 7 (Ubuntu 7.4.0-1ubuntu1~18.04.1)
configuration: --prefix=/usr --extra-version='0york0~18.04.1' --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-nonfree --enable-libfdk-aac --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
libavutil      56. 31.100 / 56. 31.100
libavcodec     58. 54.100 / 58. 54.100
libavformat    58. 29.100 / 58. 29.100
libavdevice    58.  8.100 / 58.  8.100
libavfilter     7. 57.100 /  7. 57.100
libavresample   4.  0.  0 /  4.  0.  0
libswscale      5.  5.100 /  5.  5.100
libswresample   3.  5.100 /  3.  5.100
libpostproc    55.  5.100 / 55.  5.100
```

So far so good.  The issue I had previously observed, while not completely gone, appears to be much improved.

Brunes, 2019-11-27:
Hi,
I have similar problem, my ffmpeg version:
[image]

My log
[image]

and My settings
[image]
[image]

have you some idea how to fix it?
Thanks

closed 2020-02-02 by stale[bot]