ZoneMinder/zoneminder #2692: Can't add monitor via API with new version of mariadb

stas-ab, 2019-08-30:
**Describe Your Environment**
- Version of ZoneMinder: zoneminder_1.32.3-2_amd64.deb
- How you installed ZoneMinder: `apt install ./zoneminder_1.32.3-2_amd64.deb default-mysql-server`
- Full name and version of OS: Debian 10 Buster
- mariadb-server-10.3
- Cakephp 2.10.11 from Debian Buster

**Describe the bug**
Can't add monitor via API. Got HTTP 500: SQLSTATE[01000]: Warning: 1265 Data truncated for column &#039;Orientation&#039; at row 1.
The problem is that produced SQL query contains 0 instead of '0'.
This works for mysql 5.5 in Debian Jessie (our old servers), but doesn't work in mariadb 10.3 (don't know about any other versions). Here is an example of SQL query from version 1.29 in Jessie:

```
INSERT INTO `zm`.`Monitors` (`Type`, `Function`, `Enabled`, `Channel`, `Format`, `Width`, `Height`, `Colours`, `Palette`, `Orientation`, `Deinterlacing`, `RTSPDescribe`, `Brightness`, `Contrast`, `Hue`, `Colour`, `EventPrefix`, `LabelFormat`, `LabelX`, `LabelY`, `LabelSize`, `ImageBufferCount`, `WarmupCount`, `PreEventCount`, `PostEventCount`, `StreamReplayBuffer`, `AlarmFrameCount`, `SectionLength`, `FrameSkip`, `MotionFrameSkip`, `AnalysisUpdateDelay`, `FPSReportInterval`, `RefBlendPerc`, `AlarmRefBlendPerc`, `Controllable`, `ControlId`, `TrackMotion`, `TrackDelay`, `ReturnLocation`, `ReturnDelay`, `DefaultView`, `DefaultRate`, `DefaultScale`, `SignalCheckColour`, `WebColour`, `Exif`, `Name`, `Protocol`, `Method`, `Host`, `Port`, `Path`) VALUES ('Local', 'Modect', 1, 0, 0, 704, 480, 4, 0, 0, 0, '0', -1, -1, -1, -1, 'Event-', '%N - %y/%m/%d %H:%M:%S', 0, 0, 1, 100, 25, 10, 10, 1000, 1, 600, 0, 0, 0, 250, 6, 6, 0, 0, 0, 0, -1, 0, 'Events', 100, 100, '#0000BE', 'red', '0', 'Cliff-Burton', 'http', 'simple', 'usr:pass@192.168.11.20', '80', '/mjpg/video.mjpg');
```
It works in mysql 5.5, but it doesn't work in mariadb 10.3. If I change 0 to '0' for Orientation it will work in both.

**To Reproduce**
Request from doc:
```
curl -XPOST http://localhost/zm/api/monitors.json -d "Monitor[Name]=Cliff-Burton\
&Monitor[Function]=Modect\
&Monitor[Protocol]=http\
&Monitor[Method]=simple\
&Monitor[Host]=usr:pass@192.168.11.20\
&Monitor[Port]=80\
&Monitor[Path]=/mjpg/video.mjpg\
&Monitor[Width]=704\
&Monitor[Height]=480\
&Monitor[Colours]=4"
```
Monitor isn't added in Debian Buster with MariaDB 10.3.

**Expected behavior**
Monitor should be added.

**Debug Logs**
```
{
    "success": false,
    "data": {
        "code": 500,
        "name": "SQLSTATE[01000]: Warning: 1265 Data truncated for column &#039;Orientation&#039; at row 1",
        "message": "SQLSTATE[01000]: Warning: 1265 Data truncated for column &#039;Orientation&#039; at row 1",
        "url": "\/zm\/api\/monitors.json",
        "error": {
            "errorInfo": [
                "01000",
                1265,
                "Data truncated for column 'Orientation' at row 1"
            ],
            "queryString": "INSERT INTO `zm`.`Monitors` (`StorageId`, `Type`, `Function`, `Enabled`, `Device`, `Channel`, `Format`, `Width`, `Height`, `Colours`, `Palette`, `Orientation`, `Deinterlacing`, `SaveJPEGs`, `VideoWriter`, `RecordAudio`, `Brightness`, `Contrast`, `Hue`, `Colour`, `EventPrefix`, `LabelFormat`, `LabelX`, `LabelY`, `LabelSize`, `ImageBufferCount`, `WarmupCount`, `PreEventCount`, `PostEventCount`, `StreamReplayBuffer`, `AlarmFrameCount`, `SectionLength`, `FrameSkip`, `MotionFrameSkip`, `AnalysisUpdateDelay`, `FPSReportInterval`, `RefBlendPerc`, `AlarmRefBlendPerc`, `Controllable`, `TrackMotion`, `ReturnLocation`, `DefaultView`, `DefaultRate`, `DefaultScale`, `SignalCheckPoints`, `SignalCheckColour`, `WebColour`, `Exif`, `ZoneCount`, `Name`, `Protocol`, `Method`, `Host`, `Port`, `Path`) VALUES (0, 'Local', 'Modect', 1, '\\'\\'', 0, 0, 704, 480, 4, 0, 0, 0, 3, 0, 0, -1, -1, -1, -1, 'Event-', '%N - %y\/%m\/%d %H:%M:%S', 0, 0, 1, 100, 25, 10, 10, 1000, 1, 600, 0, 0, 0, 250, 6, 6, 0, 0, -1, 'Events', 100, 100, 0, '#0000BE', 'red', '0', 0, 'Cliff-Burton', 'http', 'simple', 'usr:pass@192.168.11.20', '80', '\/mjpg\/video.mjpg')"
        },
        "exception": {
            "class": "PDOException",
            "code": "01000",
            "message": "SQLSTATE[01000]: Warning: 1265 Data truncated for column 'Orientation' at row 1",
            "trace": [
                "#0 \/usr\/share\/php\/Cake\/Model\/Datasource\/DboSource.php(472): PDOStatement->execute(Array)",
                "#1 \/usr\/share\/php\/Cake\/Model\/Datasource\/DboSource.php(438): DboSource->_execute('INSERT INTO `zm...', Array)",
                "#2 \/usr\/share\/php\/Cake\/Model\/Datasource\/DboSource.php(1092): DboSource->execute('INSERT INTO `zm...')",
                "#3 \/usr\/share\/php\/Cake\/Model\/Model.php(1942): DboSource->create(Object(Monitor), Array, Array)",
                "#4 \/usr\/share\/php\/Cake\/Model\/Model.php(1760): Model->_doSave(Array, Array)",
                "#5 \/usr\/share\/zoneminder\/www\/api\/app\/Controller\/MonitorsController.php(130): Model->save(Array)",
                "#6 [internal function]: MonitorsController->add()",
                "#7 \/usr\/share\/php\/Cake\/Controller\/Controller.php(491): ReflectionMethod->invokeArgs(Object(MonitorsController), Array)",
                "#8 \/usr\/share\/zoneminder\/www\/api\/app\/Plugin\/Crud\/Lib\/CrudControllerTrait.php(31): Controller->invokeAction(Object(CakeRequest))",
                "#9 \/usr\/share\/php\/Cake\/Routing\/Dispatcher.php(193): AppController->invokeAction(Object(CakeRequest))",
                "#10 \/usr\/share\/php\/Cake\/Routing\/Dispatcher.php(167): Dispatcher->_invoke(Object(MonitorsController), Object(CakeRequest))",
                "#11 \/usr\/share\/zoneminder\/www\/api\/app\/webroot\/index.php(107): Dispatcher->dispatch(Object(CakeRequest), Object(CakeResponse))",
                "#12 {main}"
            ]
        },
        "queryLog": {
            "default": {
                "log": [
                    {
                        "query": "BEGIN",
                        "params": [],
                        "affected": false,
                        "numRows": false,
                        "took": false
                    },
                    {
                        "query": "ROLLBACK",
                        "params": [],
                        "affected": false,
                        "numRows": false,
                        "took": false
                    }
                ],
                "count": 2,
                "time": 0
            }
        }
    }
```

stas-ab, 2019-09-03:
Looks like I just didn't know about Strict Mode in MariaDB/MySQL. So the difference between two DBs is defaults for this mode (STRICT_TRANS_TABLES).
But I still consider this as an bug.

knight-of-ni, 2019-09-03:
This has already been fixed in master. Please test using latest master branch.

connortechnology, 2019-09-03:
I don't think this has been fixed.  I can recreate it.  

It has to do with our use of enum's..  cakephp can't handle it.  I'm tempted to get rid of them all.

stas-ab, 2019-09-06:
> This has already been fixed in master. Please test using latest master branch.
Just to be sure I checked api directory from master (not whole web directory). It's the same.
The problem as I see it is string-numbers in enum. I can add monitors with 'vert' or 'hori' orientation, but not with 0/90/180/270. If I disable strict mode, than I can add monitor, but any of 0/90/180/270 values will be truncated to empty value (which works like normal orientation).

Dinsmoor, 2019-09-10:
Hate to be +1, but +1.
This does affect the most recent package versions in the Debian Buster repos.
Thanks for all your hard work.

PR #2767 (2019-11-29): Fix 2692
Fix #2692 

This PR replaces the numeric enum values with strings similar to the defines used in our C++ code.
This fixes the quoting problems that cakephp has, and is good practice.
closed 2019-11-29 by connortechnology