ZoneMinder/zoneminder #2693: Various playback and events errors after update

rhaamo, 2019-09-02:
**Describe Your Environment**
- Version of ZoneMinder: 1.33.14~20190902131017-disco
- How you installed ZoneMinder: do_debian_package.sh
- Full name and version of OS: ubuntu 19.04
- `zmupdate.pl --interactive` have already been ran, no errors

**Describe the bug**
After updating from 1.33.12 to 1.33.14 (do_debian_package both) all events pages (week, day, month,... whole or per monitor) reports like 28 events for this day on one monitor, clicking the link only list two events, and both cannot playback.

**Debug Logs**
This is one recurring error it seems, but I can see a `/var/cache/zoneminder/events` with Id 0 in Options->Storage.
Clicking that entry spawn a popup with name "NewStorage" and empty path, is that expected or something related to the DB ?
```
==> /var/log/zm/zmaudit.log <==
09/02/2019 22:33:59.680859 zmaudit[24699].ERR [ZoneMinder::Event:140] [Storage area for 0 no longer exists in db.]

==> /var/log/zm/zmdc.log <==
Use of uninitialized value $dir in opendir at /usr/bin/zmaudit.pl line 1064.
Use of uninitialized value $dir in concatenation (.) or string at /usr/bin/zmaudit.pl line 1065.

==> /var/log/zm/zmaudit.log <==
09/02/2019 22:33:59.746670 zmaudit[24699].ERR [main:1065] [Can't open directory '': No such file or directory]
```

Navigating to the daily events list of one monitor (home says 28, listed are two):
<details>
<summary>very long one, tl;dr: Unknown function call Object->Monitorid from Array</summary>

```
==> /var/log/zm/web_php.log <==
09/02/19 22:38:26.048942 web_php[25947].WAR [Unknown function call Object->Monitorid from Array
(
    [0] => Array
        (
            [file] => /usr/share/zoneminder/www/skins/classic/views/events.php
            [line] => 195
            [function] => __call
            [class] => ZM\ZM_Object
            [object] => ZM\Event Object
                (
                    [defaults:protected] => Array
                        (
                            [Id] =>
                            [Name] =>
                            [MonitorId] =>
                            [StorageId] =>
                            [SecondaryStorageId] =>
                            [Cause] =>
                            [StartTime] =>
                            [EndTime] =>
                            [Width] =>
                            [Height] =>
                            [Length] =>
                            [Frames] =>
                            [AlarmFrames] =>
                            [DefaultVideo] =>
                            [SaveJPEGs] => 0
                            [TotScore] => 0
                            [AvgScore] => 0
                            [MaxScore] => 0
                            [Archived] => 0
                            [Videoed] => 0
                            [Uploaded] => 0
                            [Emailed] => 0
                            [Messaged] => 0
                            [Executed] => 0
                            [Notes] =>
                            [StateId] => 0
                            [Orientation] => 0
                            [DiskSpace] =>
[... 440 lines cut ...]

For the playback issue, the last dir of the event show jpegs, and a .mp4 file, but the mp4 seems invalid:
```
root@fenouil:~# du -sh /var/cache/zoneminder/events/3/2019-09-02/232090/232090-video.mp4
512	/var/cache/zoneminder/events/3/2019-09-02/232090/232090-video.mp4
root@fenouil:~# ffprobe /var/cache/zoneminder/events/3/2019-09-02/232090/232090-video.mp4
ffprobe version 4.1.3-0ubuntu1 Copyright (c) 2007-2019 the FFmpeg developers
  built with gcc 8 (Ubuntu 8.3.0-6ubuntu1)
  configuration: --prefix=/usr --extra-version=0ubuntu1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 22.100 / 56. 22.100
  libavcodec     58. 35.100 / 58. 35.100
  libavformat    58. 20.100 / 58. 20.100
  libavdevice    58.  5.100 / 58.  5.100
  libavfilter     7. 40.101 /  7. 40.101
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  3.100 /  5.  3.100
  libswresample   3.  3.100 /  3.  3.100
  libpostproc    55.  3.100 / 55.  3.100
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x564b67408fc0] moov atom not found
/var/cache/zoneminder/events/3/2019-09-02/232090/232090-video.mp4: Invalid data found when processing input
root@fenouil:~#
```
This event was created way after the ZM upgrade.

I can do tests if needed, I just don't have a database backup before the upgrade because I'm an idiot.

rhaamo, 2019-09-02:
I've downgraded to: `zoneminder_1.33.12~20190728210053-disco_amd64.deb zoneminder-dbg_1.33.12~20190728210053-disco_amd64.deb zoneminder-doc_1.33.12~20190728210053-disco_all.deb`
And ran: `update Config set Value="1.33.12" where Name="ZM_DYN_DB_VERSION";`

An event triggered, it can be played, the file is valid.

I stopped ZM, reset the Value to 1.33.14, and installed again 1.33.14.

Triggered an event (outside of ZM, real movement), the event reproduce the errors previously stated (file broken, events list trigger giant error).

However the downgrade to 1.33.12 still have the problem with the count of events being wrong (I guess after upgrade the count somehow got mangled, or some records disappeared or whatever).

meldogger1972, 2019-09-03:
I opened an issue?

fennec622, 2019-09-04:
Same bug do you know command for build zoneminder_1.33.12 with do_debian_package

Thanks

rhaamo, 2019-09-04:
You should be able to either fork ZoneMinder and create a branch old enough to be a 1.33.12 and pass the args `--branch=branch_name --fork=your_github_username` or edit line 169 so that instead of using `$BRANCH` it uses an older sha commit, perhaps `--branch=git_sha_commit` would work too.

I still had the .deb lying around so I haven't tried that.

connortechnology, 2019-09-04:
Okay, so there is an issue with zmaudit resetting starttime, I will have to look into that.
Please note that I have fixed some of the issues listed here, please update.

Also please note that running zmaudit is no longer recommended.

connortechnology, 2019-09-04:
Re the Storage Area with Id 0:  The idea was on update from 1.30, since we now have Storage Areas to automatically add one for the old Default storage area, since the Monitors and Events would all default to 0 for their Storage Id. We probably should have done something smarter.
Anyways, I think we no longer do that, so you can delete that entry if you want. (I think). I think the other purpose was to retain the Deep storage scheme on update.  The new default is Medium, so we added an entry for Default storage area with type set to Deep.   You might want to switch to Medium but you don't have to.

rhaamo, 2019-09-04:
@connortechnology thanks !
I'm rebuilding ZM and will upgrade very soon and report back.

Ok for zmaudit, to be honest I didn't really checked on each updates if there was a release note about it. If I uncheck `RUN_AUDIT`, there is other means already in place to do the same thing right ?

As for the storage I don't really understand, my actual one already have a `Scheme=Medium` so that should be good ?
(The doc doesn't mention the Storage tab at all, also if I click on the entry in the WEB UI it opens the new storage popup instead of editing, dunno if that's expected)

```
| Id | Path                         | Name    | Type  | Url  | DiskSpace  | Scheme | ServerId | DoDelete |
+----+------------------------------+---------+-------+------+------------+--------+----------+----------+
|  0 | /var/cache/zoneminder/events | Default | local | NULL | 5611483433 | Medium |     NULL |        1 |
```

fennec622, 2019-09-04:
> Re the Storage Area with Id 0: The idea was on update from 1.30, since we now have Storage Areas to automatically add one for the old Default storage area, since the Monitors and Events would all [...]

I make update date is ok in web

but there is always bug with zmfilter

no argument can pass on script

> 04/09/2019 17:30:32.777306 zmfilter[32582].INF [main:177] [Scanning for events using filter 'pushalarme']
04/09/2019 17:30:32.783998 zmfilter[32582].ERR [ZoneMinder::Event:140] [Storage area for 0 no longer exists in db.]
Use of uninitialized value $event_path in concatenation (.) or string at /usr/bin/zmfilter.pl line 1005.
04/09/2019 17:30:32.784167 zmfilter[32582].INF [main:1008] [Executing '/home/fennec/scripts/videozm.py ']
Traceback (most recent call last):
  File "/home/fennec/scripts/videozm.py", line 72, in <module>
    events=argu[1]
IndexError: list index out of range
04/09/2019 17:30:33.110173 zmfilter[32582].ERR [main:1016] [Command '/home/fennec/scripts/videozm.py ' exited with status: 1]

>

rhaamo, 2019-09-04:
Ok so I've updated, I don't see spurious errors in the logs.
I've ran my SQL query to fix the StartTime, the archived list properly and I can playback an event without issue.

re the date issue. yesterday I had 19391 events with that date (1970-01-01), today only 149, which is the count of the archived ones. Every other columns are 0.

can a filter have caused the deletions ? or the audit ?
looks like the events are only missing from the database, they seems still present on filesystem

AndreasW79, 2019-09-04:
Same or nearly same error here with 
"Storage area for 0 no longer exists in db."
And in addition after removing monitor_8 zmfilter_8 still throws a lot of errors:

[image]

There is no monitor with number 8.

Storage area 0 exists and is mapped to all monitors.

[image]

zmeventnotificaton also breaks with the storage problem
[image]

Ubuntu 18.04.03 LTS
Zoneminder 1.33.14 (zmupdate.pl --interactive Database already at version 1.33.14, update skipped.)

PS: I lost a lot of events after a reboot because of some changes to db like mentioned above and cant reproduce it and have to make a large restore... dont know what exactly happened because i havent done any update before the reboot.

rhaamo, 2019-09-04:
So, I've deactivated the RUN_AUDIT.
New events seems to show in the count, and the list, and disappears after a few seconds.
Still seeing some:
```
==> /var/log/zm/zmfilter_4.log <==
09/04/2019 19:12:32.121687 zmfilter_4[7067].WAR [ZoneMinder::Event:534] [DiskSpace: Event does not exist at :ZoneMinder::Event: AlarmFrames => 4 AvgScore => 2 Cause => Motion DefaultVideo => 232362-video.mp4 EndTime => 2019-09-04 19:12:05 Frames => 29 Height => 720 Id => 232362 Length => 2.01 MaxScore => 3 MonitorId => 3 Name => Event- 232362 Notes => Motion: porte entree SaveJPEGs => 3 Scheme => Medium StartTime => 2019-09-04 19:12:03 StateId => 1 TotScore => 9 Width => 1280]

==> /var/log/zm/zmfilter_3.log <==
09/04/2019 19:12:32.634539 zmfilter_3[7062].ERR [ZoneMinder::Event:140] [Storage area for 0 no longer exists in db.]

==> /var/log/zm/zmdc.log <==
Use of uninitialized value in concatenation (.) or string at /usr/share/perl5/ZoneMinder/Event.pm line 373.
```

The event got created:
```
09/04/19 19:12:22.470360 zma_m3[8328].INF-zm_monitor.cpp/679 [Entree: image_count:56 - Closing event 232365, shutting down]
```
And deleted by the filter:
```
09/04/2019 19:12:34.277662 zmfilter_3[7062].ERR [ZoneMinder::Event:140] [Storage area for 0 no longer exists in db.]
09/04/2019 19:12:34.310798 zmfilter_3[7062].INF [ZoneMinder::Event:373] [Deleting event 232365 from Monitor 3 StartTime:2019-09-04 19:12:13 from ]
```

In fact they all got deleted after triggering (and I know the events I've triggered by walking in front of the camera should have been kept because motion):
```
MariaDB [zm]> select * from Events where Archived=false;
Empty set (0.00 sec)
```

connortechnology, 2019-09-04:
I'm a little concerned by that first line where it prints out all the values of the fields in the event.  It doesn't list StorageId.  It should.  It should say StorageId=>0
I have pushed a fix to allow for StorageId to be 0.  
You will likely have to run zmaudit (manually) to delete the events that have been deleted from the db.

rhaamo, 2019-09-05:
I've rebuilt ZM, stopped it, dropped the zm database, restored a 2 september dump.
That DB now has 19346 events, which should likely match 90% of what's on the filesystem.
I've deactivated RUN_AUDIT in the DB:
```
MariaDB [zm]> UPDATE Config set Value=0 where Name="ZM_RUN_AUDIT";
Query OK, 1 row affected (0.03 sec)
Rows matched: 1  Changed: 1  Warnings: 0
```
And fixed the dates with previous SQL Query (19332 events fixed).
Installed new ZM debs and it started.

For now, I'm seeing zmfilter removing slowly old events, but deleting them from the filesystem this time, which is expected.

Wait and see now.

rhaamo, 2019-09-05:
So far everything "looks good", I will close this issue if nothing related to it occurs soon.
Getting other issues, commented in one and will open another bug report.

AndreasW79, 2019-09-05:
With the latest master branch for me the error with not existing storage id 0 is gone too.
But the error with the deleted monitor is still here - maybe some caching issues?
`Sep 05 14:04:36 zoneminder zmfilter_8[17514]: WAR [DiskSpace: Event does not exist at /var/cache/zoneminder/events/2/2019-09-03/451437:ZoneMinder::Event: AlarmFrames => 24 AvgScore => 255
Sep 05 14:04:36 zoneminder zmfilter_8[17514]: WAR [DiskSpace: Event does not exist at /var/cache/zoneminder/events/2/2019-09-03/451436:ZoneMinder::Event: AlarmFrames => 3 AvgScore => 6 Cau
Sep 05 14:04:36 zoneminder zmfilter_8[17514]: WAR [DiskSpace: Event does not exist at /var/cache/zoneminder/events/2/2019-09-03/451435:ZoneMinder::Event: AlarmFrames => 20 AvgScore => 255
Sep 05 14:04:36 zoneminder zmfilter_8[17514]: WAR [DiskSpace: Event does not exist at /var/cache/zoneminder/events/2/2019-09-03/451434:ZoneMinder::Event: AlarmFrames => 21 AvgScore => 255
Sep 05 14:04:36 zoneminder zmfilter_8[17514]: WAR [DiskSpace: Event does not exist at /var/cache/zoneminder/events/4/2019-08-10/451432:ZoneMinder::Event: AlarmFrames => 21 AvgScore => 24 C
Sep 05 14:04:36 zoneminder zmfilter_8[17514]: WAR [DiskSpace: Event does not exist at /var/cache/zoneminder/events/4/2019-08-10/451431:ZoneMinder::Event: AlarmFrames => 5 AvgScore => 14 Ca
Sep 05 14:04:36 zoneminder zmfilter_8[17514]: WAR [DiskSpace: Event does not exist at /var/cache/zoneminder/events/4/2019-08-10/451430:ZoneMinder::Event: AlarmFrames => 66 AvgScore => 24 C
Sep 05 14:04:36 zoneminder zmfilter_8[17514]: WAR [DiskSpace: Event does not exist at /var/cache/zoneminder/events/4/2019-08-10/451429:ZoneMinder::Event: AlarmFrames => 23 AvgScore => 32 C`

Another thing i found is that the website shows the service as stopped but it is actually started.
Any hint to debug this?

[image]

connortechnology, 2019-09-10:
The website not showing as running is generally due to there being old zm files around somewhere causing the socket to be created/looked for in different places.  In a non-multiserver setup zmdc.pl puts it's socket at /run/zm/zmdc.sock.  In a multi-server setup it will be /run/zm/zmdc1.sock (or whatever number is associated with the server).

AndreasW79, 2019-09-10:
Hi @connortechnology,

thanks for your reply.
There is only one file in /run/zm/
`654 4,0K -rw-r--r--  1 www-data www-data   5 Sep 10 15:40 zm.pid`
Just renamed the file and restarted zoneminder:
`657 4,0K -rw-r--r--  1 www-data www-data   5 Sep 10 15:57 zm.pid
654 4,0K -rw-r--r--  1 www-data www-data   5 Sep 10 15:40 zm.pid.old`

Zoneminder ist installed with apt repository from your ppa-master.

Brgds
Andreas

closed 2019-10-10 by connortechnology