ZoneMinder/zoneminder #2701: PTZ conrtols does not work from "Console"

petterkatt, 2019-09-09:
**Describe Your Environment**
- Version of ZoneMinder: 1.32.3
- How you installed ZoneMinder DEB repo
- Full name and version of OS: Ubuntu 18.04.3
- Browser name and version: Firefox 69.0 / Chrome 76.0.3809.132

**Describe the bug**
Im running ZM behind a reverse proxy. ZM is running on port 80, but the reverse proxy is using SSL.
ZM Server configuration is set up to use HTTP, port 80 but the host is set to the proxy.
This is redering PTZ controls from the camera view to not work at all, however PTZ Controls from the Montage view does work.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Console' and select a PTZ camera
2. Click on any of the control buttons

**Expected behavior**
I would expect the camera controls to operate

**Debug Logs**
Not able to provide any logs since the buttons dosn't seem to do anything or trigger any inputs.
```
```

connortechnology, 2019-09-11:
Going to have to request that you upgrade to latest master package and see if we have already fixed it.
Should be painless.
sudo add-apt-repository ppa:iconnor/zoneminder-master
sudo apt-get update
sudo apt-get install zoneminder

connortechnology, 2019-09-15:
I just pushed a little change that may have fixed it.  You may want to update and give it a try.

petterkatt, 2019-09-18:
Thanks, sorry it took a while for me to be able to test.

After the latest update, I no longer can control the camera from the Montage -> Control view.
For some reason the cameras is no longer listed in the drop down menu.

closed 2019-11-24 by stale[bot]