ZoneMinder/zoneminder #2702: ONVIF camera user/pass tries to auth with ZM users instead of writing to source path

Dinsmoor, 2019-09-11:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit] 1.33.14~20190906171559-buster 
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc] https://zmrepo.zoneminder.com/debian/master buster/
- Full name and version of OS Debian Buster
- Browser name and version (if this is an issue with the web interface) Brave (shields down) and MS Edge

**If the issue concerns a camera**
- Make and Model SV3C IPcam
- frame rate
- resolution
- ZoneMinder Source Type: FFMPEG/RTSP

**Describe the bug**
A clear and concise description of what the bug is.
When trying to add an IP cam via ONVIF, cameras will be selectable. After selecting the camera and entering it's user and password, the ZM web interface goes back to the ZM Login menu, but it allows you to view streams etc. like you are already logged in.
If you type a valid user's name and password, the UI will continue, but the source path's username and password will be incorrect for the camera, but correct for ZM.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...' Add Source
2. Click on '....' ONVIF at the top
3. Scroll down to '....' Detected Cameras, select your camera, enter user and pass (for the camera), press Next.
4. See error. (this error does not occur when using the ZM username/passphrase, but it needs to be changed in the Source Path under the Source tab later.)

**Expected behavior**
A clear and concise description of what you expected to happen.

Do NOT try to authenticate the username and password for the camera with the ZM users database. Just paste to source path.
**Debug Logs**
```

2019-09-10 17:29:51 | web_php |   | 2234 | ERR | Login denied for user "admin" | /usr/share/zoneminder/www/includes/auth.php | 286
-- | -- | -- | -- | -- | -- | -- | --

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

Dinsmoor, 2019-09-11:
Please note that this works properly if authentication is turned off in the ZM settings.

connortechnology, 2019-09-11:
I think I know what is going on here and should be able to fix it today.

Thanks for the clear report.

closed 2019-09-11 by connortechnology