ZoneMinder/zoneminder #2703: Event Deletion Error

Dinsmoor, 2019-09-11:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit] 1.33.14~20190906171559-buster
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc] https://zmrepo.zoneminder.com/debian/master buster/
- Full name and version of OS Debian Buster
- Browser name and version (if this is an issue with the web interface) Brave and Edge. Probably not a browser problem.

**Describe the bug**
A clear and concise description of what the bug is.

When I attempt to delete an event, it renders an error in the apache2 logs and it does NOT delete the event.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...' Events menu
2. Click on '....' the checkbox for any event, press 'Delete' Press OK on confirmation dialog.
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

Events should delete during the reproduction steps.

**Debug Logs**
```

In /var/log/apache2/error.log:
[Tue Sep 10 18:34:21.897938 2019] [php7:error] [pid 31211] [client 192.168.1.141:59493] PHP Fatal error:  Uncaught PDOException: There is already an active transaction in /usr/share/zoneminder/www/includes/Event.php:135\nStack trace:\n#0 /usr/share/zoneminder/www/includes/Event.php(135): PDO->beginTransaction()\n#1 /usr/share/zoneminder/www/includes/functions.php(442): ZM\\Event->delete()\n#2 /usr/share/zoneminder/www/includes/actions/events.php(49): deleteEvent(Object(ZM\\Event))\n#3 /usr/share/zoneminder/www/index.php(216): require_once('/usr/share/zone...')\n#4 {main}\n  thrown in /usr/share/zoneminder/www/includes/Event.php on line 135, referer: http://tercero-camera/zm/index.php?view=events&page=1&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Battr%5D=MonitorId&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bop%5D=%3D&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bval%5D=3 

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

Dinsmoor, 2019-09-11:
Please note that I ensured a complete removal of this package, and a complete reinstallation, including purge of databases just to make sure it wasn't some configuration I changed. The server is a fresh installation with Zoneminder installed as the first package.

connortechnology, 2019-09-11:
Your installed package is from Sept 6.  I fixed this on the 9th. 8103156436173db679dae579a41042eb69313e1e

closed 2019-09-11 by connortechnology

Dinsmoor, 2019-09-12:
Thank you very much, I did another build from master branch today, and it works well.
Expect a donation.