ZoneMinder/zoneminder #2707: Error when php does not have session.use_trans_sid

kifeo, 2019-09-16:
**Describe Your Environment**
- Version of ZoneMinder v1.32.3
- How you installed ZoneMinder Freebsd jail from Freenas
- Full name and version of OS : Freenas Jail 11.2
Using the library here for homeassistant zoneminder component.

**Describe the bug**
If the php.ini does have the session.use_trans_id set to 1, the following backtrace appears :
```

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/zoneminder/zm.py", line 93, in _zm_request
    return req.json()
  File "/usr/local/lib/python3.7/site-packages/requests/models.py", line 897, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/simplejson/__init__.py", line 518, in loads
    return _default_decoder.decode(s)
  File "/usr/local/lib/python3.7/site-packages/simplejson/decoder.py", line 370, in decode
    obj, end = self.raw_decode(s)
  File "/usr/local/lib/python3.7/site-packages/simplejson/decoder.py", line 400, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
simplejson.errors.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
```

The logs above the stacktrace does contain a lot of info and pollutes the disk.
This is like :
```

$defaults = array(
	&#039;cookie&#039; =&gt; &#039;CAKEPHP&#039;,
	&#039;timeout&#039; =&gt; (int) 240,
	&#039;ini&#039; =&gt; array(
		&#039;session.use_trans_sid&#039; =&gt; (int) 0,
		&#039;session.cookie_path&#039; =&gt; &#039;/&#039;
	)
)
$value = (int) 0
$setting = &#039;session.use_trans_sid&#039;</pre><pre class="stack-trace">ini_set - [internal], line ??
CakeSession::_configureSession() - CORE/Cake/Model/Datasource/CakeSession.php, line 581
CakeSession::_startSession() - CORE/Cake/Model/Datasource/CakeSession.php, line 732
CakeSession::start() - CORE/Cake/Model/Datasource/CakeSession.php, line 219
CakeSession::read() - CORE/Cake/Model/Datasource/CakeSession.php, line 404
SessionComponent::read() - CORE/Cake/Controller/Component/SessionComponent.php, line 69
AppController::beforeFilter() - APP/Controller/AppController.php, line 70
CakeEventManager::dispatch() - CORE/Cake/Event/CakeEventManager.php, line 243
Controller::startupProcess() - CORE/Cake/Controller/Controller.php, line 677
ExceptionRenderer::_getController() - CORE/Cake/Error/ExceptionRenderer.php, line 158
ExceptionRenderer::__construct() - CORE/Cake/Error/ExceptionRenderer.php, line 95
ErrorHandler::handleException() - CORE/Cake/Error/ErrorHandler.php, line 126
[main] - [internal], line ??</pre></div></pre>{
    "success": false,
    "data": {
        "code": 500,
        "name": "Unable to configure the session, setting session.use_trans_sid failed.",
        "message": "Unable to configure the session, setting session.use_trans_sid failed.",
        "url": "\/zm\/api\/host\/daemonCheck.json?p=%2Fhost%2FdaemonCheck.json",
        "exception": {
            "class": "CakeSessionException",
            "code": 500,
            "message": "Unable to configure the session, setting session.use_trans_sid failed.",
            "trace": [
                "#0 \/usr\/local\/www\/zoneminder\/api\/lib\/Cake\/Model\/Datasource\/CakeSession.php(619): CakeSession::_configureSession()",
                "#1 \/usr\/local\/www\/zoneminder\/api\/lib\/Cake\/Model\/Datasource\/CakeSession.php(632): CakeSession::_cookieName()",
                "#2 \/usr\/local\/www\/zoneminder\/api\/lib\/Cake\/Model\/Datasource\/CakeSession.php(404): CakeSession::_hasSession()",
                "#3 \/usr\/local\/www\/zoneminder\/api\/lib\/Cake\/Controller\/Component\/SessionComponent.php(69): CakeSession::read('user')",
                "#4 \/usr\/local\/www\/zoneminder\/api\/app\/Controller\/AppController.php(70): SessionComponent->read('user')",
                "#5 \/usr\/local\/www\/zoneminder\/api\/lib\/Cake\/Event\/CakeEventManager.php(243): AppController->beforeFilter(Object(CakeEvent))",
                "#6 \/usr\/local\/www\/zoneminder\/api\/lib\/Cake\/Controller\/Controller.php(677): CakeEventManager->dispatch(Object(CakeEvent))",
                "#7 \/usr\/local\/www\/zoneminder\/api\/lib\/Cake\/Routing\/Dispatcher.php(189): Controller->startupProcess()",
                "#8 \/usr\/local\/www\/zoneminder\/api\/lib\/Cake\/Routing\/Dispatcher.php(167): Dispatcher->_invoke(Object(HostController), Object(CakeRequest))",
                "#9 \/usr\/local\/www\/zoneminder\/api\/app\/webroot\/index.php(107): Dispatcher->dispatch(Object(CakeRequest), Object(CakeResponse))",
                "#10 {main}"
            ]
        },
        "queryLog": []
    }
}"
```

**To Reproduce**
- On freenas, deploy the zoneminder jail
- deploy a homeassistant and setup the zoneminder module
- the backatrace is generated.

**Expected behavior**
The error message should be JSON formatted without giving that much verbose output.

connortechnology, 2019-09-18:
Please check that in /usr/share/zoneminder/www/api/app/Configure/core.php on line 34 that you have 
Configure::write('debug', 0);
instead of 
Configure::write('debug', 2);

This should prevent the backtrace.

Further, in current master we have removed most use of sessions from cakephp.  So I suspect it won't be an issue going forward.

kifeo, 2019-09-19:
Thanks ! 
Indeed this resolved my issue.

closed 2019-09-19 by kifeo