ZoneMinder/zoneminder #2708: Unable to build ZoneMinder 1.33.14 using packpack

chssn, 2019-09-18:
**Describe Your Environment**
- Version of ZoneMinder: commit 17e00ce and 9889311 (v1.33.14)
- How you installed ZoneMinder: attempting to build using these instructions - [https://zoneminder.readthedocs.io/en/stable/installationguide/packpack.html#procedure](https://zoneminder.readthedocs.io/en/stable/installationguide/packpack.html#procedure)
- Full name and version of OS: Ubuntu Server 19.04 (Disco) x86_64

**Describe the bug**
Fails to build ZoneMinder using these instructions [https://zoneminder.readthedocs.io/en/stable/installationguide/packpack.html#procedure](https://zoneminder.readthedocs.io/en/stable/installationguide/packpack.html#procedure)

**To Reproduce**
Steps to reproduce the behavior:
```
git clone https://github.com/ZoneMinder/zoneminder
cd zoneminder
OS=ubuntu DIST=disco utils/packpack/startpackpack.sh
```

**Expected behavior**
A ZoneMinder build from the development branch for Ubuntu Server 19.04 (Disco) x86_64

**Debug Logs**
```
chris@snet-desktop:~/git-repos/zoneminder$ OS=ubuntu DIST=disco utils/packpack/startpackpack.sh 
Checking packpack github repo for changes...
From https://github.com/zoneminder/packpack
 * branch            master     -> FETCH_HEAD
Already up-to-date.
Found existing Crud 3.1.0-zm tarball...
Found existing CakePHP-Enum-Behavior 1.0-zm tarball...
Begin ubuntu disco build...

Packpack VERSION has been set to: 1.33.14~20190918.245
Packpack RELEASE has been set to: disco

Crud plugin already installed...
CakePHP-Enum-Behavior plugin already installed...
Starting packpack...
+ docker pull packpack/packpack:ubuntu-disco
ubuntu-disco: Pulling from packpack/packpack
Digest: sha256:ada6c3e5006484c7c85cbc8665e7507548a1c321aee844456e4d1382733122f3
Status: Image is up to date for packpack/packpack:ubuntu-disco
docker.io/packpack/packpack:ubuntu-disco
+ docker run --volume /home/chris/git-repos/zoneminder/packpack/pack:/pack:ro --volume /home/chris/git-repos/zoneminder:/source:ro --volume /home/chris/git-repos/zoneminder/build:/build --env-file /home/chris/git-repos/zoneminder/build/env --workdir /source --rm=true --tty=true --entrypoint=/build/userwrapper.sh -e XDG_CACHE_HOME=/cache -e CCACHE_DIR=/cache/ccache -e TMPDIR=/tmp --volume /home/chris/.cache/packpack:/cache packpack/packpack:ubuntu-disco make -f /pack/Makefile -C /source BUILDDIR=/build -j
useradd: user 'root' already exists
make: Entering directory '/source'
# empty
# empty
# empty
-------------------------------------------------------------------
Installing dependencies
-------------------------------------------------------------------
if [ -n "" ] && [ -n "" ]; then \
	curl -s https://packagecloud.io/install/repositories///script.deb.sh | sudo bash; \
fi
sudo rm -rf /var/lib/apt/lists/
sudo apt-get update > /dev/null
cd /build/zoneminder-1.33.14~20190918.245 && \
	sudo mk-build-deps -i --tool "apt-get --no-install-recommends -y" && \
	sudo rm -f *build-deps_*.deb \
@echo
dh_testdir
dh_testroot
dh_prep
dh_testdir
dh_testroot
dh_install
dh_installdocs
dh_installchangelogs
dh_compress
dh_fixperms
dh_installdeb
[... 89 lines cut ...]
Unpacking shared-mime-info (1.10-1) ...
Preparing to unpack .../5-libldap-2.4-2_2.4.47+dfsg-3ubuntu2.2_amd64.deb ...
Unpacking libldap-2.4-2:amd64 (2.4.47+dfsg-3ubuntu2.2) over (2.4.47+dfsg-3ubuntu2) ...
Preparing to unpack .../6-libcurl3-gnutls_7.64.0-2ubuntu1.2_amd64.deb ...
Unpacking libcurl3-gnutls:amd64 (7.64.0-2ubuntu1.2) over (7.64.0-2ubuntu1) ...
Setting up libexpat1:amd64 (2.2.6-1ubuntu0.19.5) ...
Setting up libglib2.0-0:amd64 (2.60.4-0ubuntu0.19.04.1) ...
No schema files found: doing nothing.
Setting up libssl1.1:amd64 (1.1.1b-1ubuntu2.1) ...
debconf: unable to initialize frontend: Dialog
debconf: (No usable dialog-like program is installed, so the dialog based frontend cannot be used. at /usr/share/perl5/Debconf/FrontEnd/Dialog.pm line 76.)
debconf: falling back to frontend: Readline
Setting up libldap-2.4-2:amd64 (2.4.47+dfsg-3ubuntu2.2) ...
Setting up libcurl3-gnutls:amd64 (7.64.0-2ubuntu1.2) ...
Setting up bzip2 (1.0.6-9ubuntu0.19.04.1) ...
Setting up libglib2.0-data (2.60.4-0ubuntu0.19.04.1) ...
Setting up shared-mime-info (1.10-1) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for libc-bin (2.29-0ubuntu1) ...
mk-build-deps: Unable to install zoneminder-build-deps at /usr/bin/mk-build-deps line 416.
mk-build-deps: Unable to install all build-dep packages
make: *** [/pack//deb.mk:121: /build/zoneminder_1.33.14~20190918.245-disco_amd64.changes] Error 1
make: Leaving directory '/source'
chris@snet-desktop:~/git-repos/zoneminder$ 
```

connortechnology, 2019-10-23:
the packpack docker images don't have everything needed.  You can use my packpack images by using 

OS=ubuntu DIST=disco DOCKER_REPO=iconzm/packpack ./utils/packpack/startpackpack.sh

closed 2019-12-29 by stale[bot]

ThellraAK, 2020-10-10:
Is there any other work around for this besides using @connortechnology thing?