ZoneMinder/zoneminder #2709: No video - mp4

nnneooo, 2019-09-20:
Zoneminder was installed on Debian10 buster.
I've seen updates - zoneminder-dbg_1.33.14 ~ 20190919.245-stretch but under Debian buster it is not.

Describe Your Environment

    Version of ZoneMinder [release version, development version, or commit]
    1.33.14~20190906171559-buster.
    How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
    Following: https://zoneminder.readthedocs.io/en/1.32.3/installationguide/debian.html#easy-way-debian-jessie
    Full name and version of OS
Linux  4.19.0-6-amd64 #1 SMP Debian 4.19.67-2 (2019-08-28) x86_64 GNU/Linux

Describe the bug

To Reproduce
Steps to reproduce the behavior:

    View an event

Expected behavior

2019-09-20 12:20:09.868963 zmc_m15[8913].ERR-zm_signal.cpp/104 [Backtrace 13: /usr/bin/zmc(_start+0x2a) [0x55cea5c722aa]]
2019-09-20 12:20:09.751718 zmc_m15[8913].ERR-zm_signal.cpp/104 [Backtrace 12: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xeb) [0x7fdb7dfac09b]]
2019-09-20 12:20:09.626075 zmc_m15[8913].ERR-zm_signal.cpp/104 [Backtrace 11: /usr/bin/zmc(main+0xb9a) [0x55cea5c701ea]]
2019-09-20 12:20:09.508481 zmc_m15[8913].ERR-zm_signal.cpp/104 [Backtrace 10: /usr/bin/zmc(_ZN7Monitor7CaptureEv+0x4d9) [0x55cea5c94a49]]
2019-09-20 12:20:09.383062 zmc_m15[8913].ERR-zm_signal.cpp/104 [Backtrace 9: /usr/bin/zmc(_ZN12FfmpegCamera16CaptureAndRecordER5Image7timevalPc+0x1e3d) [0x55cea5d31add]]
2019-09-20 12:20:09.383013 zmc_m4[4161].ERR-zmc.cpp/310 [Failed to pre-capture monitor 4 -1688031244 (1/1)]
2019-09-20 12:20:09.282480 zmc_m4[4161].ERR-zm_remote_camera_http.cpp/1063 [Unable to connect to camera]
2019-09-20 12:20:09.282423 zmc_m15[8913].ERR-zm_signal.cpp/104 [Backtrace 8: /usr/bin/zmc(_ZN10VideoStore21writeVideoFramePacketEP8AVPacket+0x1a4) [0x55cea5cea214]]
2019-09-20 12:20:09.246730 zmwatch[1454].ERR-zmwatch.pl [Last analyse time for 14 Balkon Nowy 14 was zero.]
2019-09-20 12:20:09.165261 zmc_m4[4161].ERR-zm_remote_camera_http.cpp/169 [Unable to connect to the remote camera, aborting]
2019-09-20 12:20:09.165243 zmc_m15[8913].ERR-zm_signal.cpp/104 [Backtrace 7: /usr/bin/zmc(+0x3a3b3) [0x55cea5c543b3]]
2019-09-20 12:20:09.039844 zmc_m15[8913].ERR-zm_signal.cpp/104 [Backtrace 6: /lib/x86_64-linux-gnu/libc.so.6(+0x8142c) [0x7fdb7e00942c]]
2019-09-20 12:20:08.916481 zmc_m15[8913].ERR-zm_signal.cpp/104 [Backtrace 5: /lib/x86_64-linux-gnu/libc.so.6(+0x7fc1a) [0x7fdb7e007c1a]]
2019-09-20 12:20:08.771579 zmc_m15[8913].ERR-zm_signal.cpp/104 [Backtrace 4: /lib/x86_64-linux-gnu/libc.so.6(+0x79508) [0x7fdb7e001508]]
2019-09-20 12:20:08.649787 zmc_m15[8913].ERR-zm_signal.cpp/104 [Backtrace 3: /lib/x86_64-linux-gnu/libc.so.6(abort+0x121) [0x7fdb7dfaa535]]
2019-09-20 12:20:08.343782 zmc_m15[8913].ERR-zm_signal.cpp/104 [Backtrace 2: /lib/x86_64-linux-gnu/libc.so.6(gsignal+0x10b) [0x7fdb7dfbf7bb]]
2019-09-20 12:20:08.118112 zmc_m15[8913].ERR-zm_signal.cpp/104 [Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0x12730) [0x7fdb80233730]]
2019-09-20 12:20:07.825843 zmc_m15[8913].ERR-zm_signal.cpp/104 [Backtrace 0: /usr/bin/zmc(_Z14zm_die_handleriP9siginfo_tPv+0x7a) [0x55cea5cd3f8a]]
2019-09-20 12:20:07.407885 zmc_m4[4161].ERR-zmc.cpp/310 [Failed to pre-capture monitor 4 -1688031244 (1/1)]
2019-09-20 12:20:07.307540 zmc_m15[8913].ERR-zm_signal.cpp/83 [Signal address is 0x21000022d1, from 0x7fdb7dfbf7bb]
2019-09-20 12:20:07.257166 zmc_m4[4161].ERR-zm_remote_camera_http.cpp/1063 [Unable to connect to camera]
2019-09-20 12:20:07.213315 zmc_m15[8913].ERR-zm_signal.cpp/52 [Got signal 6 (Aborted), crashing]
2019-09-20 12:20:07.156582 zmc_m4[4161].ERR-zm_remote_camera_http.cpp/169 [Unable to connect to the remote camera, aborting]
2019-09-20 12:20:05.933148 zmc_m4[4161].ERR-zmc.cpp/310 [Failed to pre-capture monitor 4 -1688031244 (1/1)]
2019-09-20 12:20:05.882842 zmc_m4[4161].ERR-zm_remote_camera_http.cpp/1063 [Unable to connect to camera]
2019-09-20 12:20:05.832772 zmc_m4[4161].ERR-zm_remote_camera_http.cpp/169 [Unable to connect to the remote camera, aborting]
2019-09-20 12:20:04.661651 zmc_m4[4161].ERR-zmc.cpp/310 [Failed to pre-capture monitor 4 -1688031244 (1/1)]
2019-09-20 12:20:04.561382 zmc_m4[4161].ERR-zm_remote_camera_http.cpp/1063 [Unable to connect to camera]
2019-09-20 12:20:04.427342 zmc_m4[4161].ERR-zm_remote_camera_http.cpp/169 [Unable to connect to the remote camera, aborting]
2019-09-20 12:20:03.196684 zmc_m4[4161].ERR-zmc.cpp/310 [Failed to pre-capture monitor 4 -1688031244 (1/1)]
2019-09-20 12:20:03.112943 zmc_m4[4161].ERR-zm_remote_camera_http.cpp/1063 [Unable to connect to camera]
2019-09-20 12:20:03.070984 zmc_m4[4161].ERR-zm_remote_camera_http.cpp/169 [Unable to connect to the remote camera, aborting]
2019-09-20 12:20:02.770200 zmdc[1138].ERR-zmdc.pl ['zmc -m 20' exited abnormally, exit status 6]
2019-09-20 12:20:02.534939 zmc_m20[8855].ERR-zm_signal.cpp/104 [Backtrace 13: /usr/bin/zmc(_start+0x2a) [0x563203c8e2aa]]
2019-09-20 12:20:02.367562 zmc_m20[8855].ERR-zm_signal.cpp/104 [Backtrace 12: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xeb) [0x7f1032c4a09b]]
2019-09-20 12:20:02.166654 zmc_m20[8855].ERR-zm_signal.cpp/104 [Backtrace 11: /usr/bin/zmc(main+0xb9a) [0x563203c8c1ea]]
2019-09-20 12:20:02.123630 zmdc[1138].ERR-zmdc.pl ['zmc -m 19' exited abnormally, exit status 6]
2019-09-20 12:20:02.064934 zmc_m20[8855].ERR-zm_signal.cpp/104 [Backtrace 10: /usr/bin/zmc(_ZN7Monitor7CaptureEv+0x4d9) [0x563203cb0a49]]
2019-09-20 12:20:01.933416 zmc_m20[8855].ERR-zm_signal.cpp/104 [Backtrace 9: /usr/bin/zmc(_ZN12FfmpegCamera16CaptureAndRecordER5Image7timevalPc+0x1e3d) [0x563203d4dadd]]
2019-09-20 12:20:01.858150 zmdc[1138].ERR-zmdc.pl ['zmc -m 22' exited abnormally, exit status 6]
2019-09-20 12:20:01.788661 zmc_m20[8855].ERR-zm_signal.cpp/104 [Backtrace 8: /usr/bin/zmc(_ZN10VideoStore21writeVideoFramePacketEP8AVPacket+0x1a4) [0x563203d06214]]
2019-09-20 12:20:01.788661 zmc_m4[4161].ERR-zmc.cpp/310 [Failed to pre-capture monitor 4 -1688031244 (1/1)]
2019-09-20 12:20:01.746643 zmc_m19[8839].ERR-zm_signal.cpp/104 [Backtrace 13: /usr/bin/zmc(_start+0x2a) [0x56248b9242aa]]
2019-09-20 12:20:01.704969 zmc_m4[4161].ERR-zm_remote_camera_http.cpp/1063 [Unable to connect to camera]
2019-09-20 12:20:01.704856 zmc_m20[8855].ERR-zm_signal.cpp/104 [Backtrace 7: /usr/bin/zmc(+0x3a3b3) [0x563203c703b3]]
2019-09-20 12:20:01.662976 zmc_m19[8839].ERR-zm_signal.cpp/104 [Backtrace 12: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xeb) [0x7f8b88fd909b]]
2019-09-20 12:20:01.621074 zmc_m20[8855].ERR-zm_signal.cpp/104 [Backtrace 6: /lib/x86_64-linux-gnu/libc.so.6(+0x8142c) [0x7f1032ca742c]]
2019-09-20 12:20:01.621073 zmc_m4[4161].ERR-zm_remote_camera_http.cpp/169 [Unable to connect to the remote camera, aborting]
2019-09-20 12:20:01.579078 zmc_m19[8839].ERR-zm_signal.cpp/104 [Backtrace 11: /usr/bin/zmc(main+0xb9a) [0x56248b9221ea]]
[... 122 lines cut ...]
2019-09-20 12:19:53.125733 zmc_m20[8718].ERR-zm_signal.cpp/104 [Backtrace 4: /lib/x86_64-linux-gnu/libc.so.6(+0x79508) [0x7f5d65cb5508]]
2019-09-20 12:19:53.082284 zmc_m19[8710].ERR-zm_signal.cpp/104 [Backtrace 6: /lib/x86_64-linux-gnu/libc.so.6(+0x8142c) [0x7fb2af37242c]]
2019-09-20 12:19:53.032509 zmc_m20[8718].ERR-zm_signal.cpp/104 [Backtrace 3: /lib/x86_64-linux-gnu/libc.so.6(abort+0x121) [0x7f5d65c5e535]]
2019-09-20 12:19:53.032045 zmc_m19[8710].ERR-zm_signal.cpp/104 [Backtrace 5: /lib/x86_64-linux-gnu/libc.so.6(+0x7fc1a) [0x7fb2af370c1a]]
2019-09-20 12:19:52.931652 zmc_m19[8710].ERR-zm_signal.cpp/104 [Backtrace 4: /lib/x86_64-linux-gnu/libc.so.6(+0x79508) [0x7fb2af36a508]]
2019-09-20 12:19:52.931570 zmc_m20[8718].ERR-zm_signal.cpp/104 [Backtrace 2: /lib/x86_64-linux-gnu/libc.so.6(gsignal+0x10b) [0x7f5d65c737bb]]
2019-09-20 12:19:52.805733 zmc_m19[8710].ERR-zm_signal.cpp/104 [Backtrace 3: /lib/x86_64-linux-gnu/libc.so.6(abort+0x121) [0x7fb2af313535]]
2019-09-20 12:19:52.805730 zmc_m20[8718].ERR-zm_signal.cpp/104 [Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0x12730) [0x7f5d67ee7730]]
2019-09-20 12:19:52.680212 zmc_m20[8718].ERR-zm_signal.cpp/104 [Backtrace 0: /usr/bin/zmc(_Z14zm_die_handleriP9siginfo_tPv+0x7a) [0x559220037f8a]]
2019-09-20 12:19:52.679799 zmc_m19[8710].ERR-zm_signal.cpp/104 [Backtrace 2: /lib/x86_64-linux-gnu/libc.so.6(gsignal+0x10b) [0x7fb2af3287bb]]
2019-09-20 12:19:52.629574 zmc_m4[4161].ERR-zmc.cpp/310 [Failed to pre-capture monitor 4 -1688031244 (1/1)]
2019-09-20 12:19:52.580733 zmc_m20[8718].ERR-zm_signal.cpp/83 [Signal address is 0x210000220e, from 0x7f5d65c737bb]
[... cut ...]
2019-09-20 12:19:50.993364 zmc_m4[4161].ERR-zm_remote_camera_http.cpp/1063 [Unable to connect to camera]
2019-09-20 12:19:50.909282 zmc_m4[4161].ERR-zm_remote_camera_http.cpp/169 [Unable to connect to the remote camera, aborting]
2019-09-20 12:19:50.775380 zmdc[1138].ERR-zmdc.pl ['zmc -m 14' exited abnormally, exit status 6]
2019-09-20 12:19:50.179078 zmc_m14[8674].ERR-zm_signal.cpp/104 [Backtrace 13: /usr/bin/zmc(_start+0x2a) [0x55d82e7242aa]]
2019-09-20 12:19:50.020137 zmc_m14[8674].ERR-zm_signal.cpp/104 [Backtrace 12: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xeb) [0x7fc10481d09b]]

connortechnology, 2019-09-20:
There is  zoneminder 1.33.14~20190920082028-buster available in zmrepo. Please update to it and report back.

nnneooo, 2019-09-21:
Hello,
after the update is ok :).
But there is another problem with zmNinja. No camera preview.

Waiting for authSession to have a value ..

connortechnology, 2019-09-28:
You will need to manually update your database.
mysql -u zmuser -p zm < /usr/share/zoneminder/db/zm_update-1.33.14.sql

closed 2019-09-28 by connortechnology