ZoneMinder/zoneminder #2716: Unable to batch edit monitors' function 1.32.3

theonemcdonald, 2019-10-03:
**Describe Your Environment**
- Version of ZoneMinder: RELEASE
- How you installed ZoneMinder: PPA
- Full name and version of OS: Ubuntu 18.04

**Describe the bug**
When editing multiple monitors simultaneously, I am unable to batch edit these monitors' function. 

**To Reproduce**
Steps to reproduce the behavior:
1. Select multiple monitors.
2. Click edit.
3. Change function.
4. Click save.

**Expected behavior**
A clear and concise description of what you expected to happen.
When clicking save, all cameras that were selected should have the new function applied.

**Debug Logs**
```

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

connortechnology, 2019-10-08:
I cannot reproduce this on current master.  Please update and confirm.
To update to latest master do:
sudo add-apt-repository ppa:iconnor/zoneminder-master
sudo apt-get update
sudo apt-get install zoneminder

theonemcdonald, 2019-10-08:
Same, master branch seems to work. Closing.

closed 2019-10-08 by theonemcdonald