ZoneMinder/zoneminder #2717: Some PTZ icons disappeared

derKaY, 2019-10-03:
- Version of ZoneMinder [**v1.33.14**]
- How you installed ZoneMinder [**ZM Repro Master**]
- Debian 10.1
- Chrome/Firefox (latest)

**If the issue concerns a camera**
- DAHUA SD59230U-HNI
- 20 fps
- 1920x1080
- ffmpeg

**Describe the bug**
Some PTZ control icons (direction arrows, presets) are disappeared since the last update.
Focus, Zoom and Iris still there. / I think this is not a path problem

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '**Console**'
2. Click on '**Camera**'
3. Scroll down to '**PTZ controls**'
4. See error '**no arrows**'

**Expected behavior**
Is not affected by browser type ... but the function is still possible in my app eg. "zm_ninja"

**Screenshot Error Log**
[image]

**Debug Logs**
```

2019-10-03 14:26:08 | web_php |   | 13325 | ERR | Control: ZM\Control Object( [defaults:protected] => Array ( [Id] => [CanMove] => 0 [CanMoveDiag] => 0 [CanMoveMap] => 0 [CanMoveAbs] => 0 [CanMoveRel] => 0 [CanMoveCon] => 0 [CanPan] => 0 [CanReset] => 0 [CanReboot] => 0 [CanSleep] => 0 [CanWake] => 0 [MinPanRange] => [MaxPanRange] => [MinPanStep] => [MaxPanStep] => [HasPanSpeed] => 0 [MinPanSpeed] => [MaxPanSpeed] => [HasTurboPan] => 0 [TurboPanSpeed] => [CanTilt] => 0 [MinTiltRange] => [MaxTiltRange] => [MinTiltStep] => [MaxTiltStep] => [HasTiltSpeed] => 0 [MinTiltSpeed] => [MaxTiltSpeed] => [HasTurboTilt] => 0 [TurboTiltSpeed] => [CanZoom] => 0 [CanZoomAbs] => 0 [CanZoomRel] => 0 [CanZoomCon] => 0 [MinZoomRange] => [MaxZoomRange] => [MinZoomStep] => [MaxZoomStep] => [HasZoomSpeed] => 0 [MinZoomSpeed] => [MaxZoomSpeed] => [CanFocus] => 0 [CanAutoFocus] => 0 [CanFocusAbs] => 0 [CanFocusRel] => 0 [CanFocusCon] => 0 [MinFocusRange] => [MaxFocusRange] => [MinFocusStep] => [MaxFocusStep] => [HasFocusSpeed] => 0 [MinFocusSpeed] => [MaxFocusSpeed] => [CanIris] => 0 [CanAutoIris] => 0 [CanIrisAbs] => 0 [CanIrisRel] => 0 [CanIrisCon] => 0 [MinIrisRange] => [MaxIrisRange] => [MinIrisStep] => [MaxIrisStep] => [HasIrisSpeed] => 0 [MinIrisSpeed] => [MaxIrisSpeed] => [CanGain] => 0 [CanAutoGain] => 0 [CanGainAbs] => 0 [CanGainRel] => 0 [CanGainCon] => 0 [MinGainRange] => [MaxGainRange] => [MinGainStep] => [MaxGainStep] => [HasGainSpeed] => 0 [MinGainSpeed] => [MaxGainSpeed] => [CanWhite] => 0 [CanAutoWhite] => 0 [CanWhiteAbs] => 0 [CanWhiteRel] => 0 [CanWhiteCon] => 0 [MinWhiteRange] => [MaxWhiteRange] => [MinWhiteStep] => [MaxWhiteStep] => [HasWhiteSpeed] => 0 [MinWhiteSpeed] => [MaxWhiteSpeed] => [HasPresets] => 0 [NumPresets] => 0 [HasHomePreset] => 0 [CanSetPresets] => 0 [Name] => New [Type] => Local [Protocol] => ) [Id] => 38 [Name] => Dahua [Type] => Ffmpeg [Protocol] => Dahua [CanWake] => 0 [CanSleep] => 0 [CanReset] => 1 [CanReboot] => 1 [CanZoom] => 1 [CanAutoZoom] => 0 [CanZoomAbs] => 0 [CanZoomRel] => 1 [CanZoomCon] => 0 [MinZoomRange] => 0 [MaxZoomRange] => 0 [MinZoomStep] => 0 [MaxZoomStep] => 0 [HasZoomSpeed] => 0 [MinZoomSpeed] => 0 [MaxZoomSpeed] => 0 [CanFocus] => 1 [CanAutoFocus] => 0 [CanFocusAbs] => 0 [CanFocusRel] => 1 [CanFocusCon] => 0 [MinFocusRange] => 0 [MaxFocusRange] => 0 [MinFocusStep] => 0 [MaxFocusStep] => 0 [HasFocusSpeed] => 0 [MinFocusSpeed] => 0 [MaxFocusSpeed] => 0 [CanIris] => 1 [CanAutoIris] => 0 [CanIrisAbs] => 0 [CanIrisRel] => 1 [CanIrisCon] => 0 [MinIrisRange] => 0 [MaxIrisRange] => 0 [MinIrisStep] => 0 [MaxIrisStep] => 0 [HasIrisSpeed] => 0 [MinIrisSpeed] => 0 [MaxIrisSpeed] => 0 [CanGain] => 0 [CanAutoGain] => 0 [CanGainAbs] => 0 [CanGainRel] => 0 [CanGainCon] => 0 [MinGainRange] => 0 [MaxGainRange] => 0 [MinGainStep] => 0 [MaxGainStep] => 0 [HasGainSpeed] => 0 [MinGainSpeed] => 0 [MaxGainSpeed] => 0 [CanWhite] => 0 [CanAutoWhite] => 0 [CanWhiteAbs] => 0 [CanWhiteRel] => 0 [CanWhiteCon] => 0 [MinWhiteRange] => 0 [MaxWhiteRange] => 0 [MinWhiteStep] => 0 [MaxWhiteStep] => 0 [HasWhiteSpeed] => 0 [MinWhiteSpeed] => 0 [MaxWhiteSpeed] => 0 [HasPresets] => 1 [NumPresets] => 20 [HasHomePreset] => 1 [CanSetPresets] => 1 [CanMove] => 1 [CanMoveDiag] => 1 [CanMoveMap] => 0 [CanMoveAbs] => 0 [CanMoveRel] => 1 [CanMoveCon] => 0 [CanPan] => 1 [MinPanRange] => 0 [MaxPanRange] => 0 [MinPanStep] => 0 [MaxPanStep] => 0 [HasPanSpeed] => 0 [MinPanSpeed] => 0 [MaxPanSpeed] => 0 [HasTurboPan] => 0 [TurboPanSpeed] => 0 [CanTilt] => 1 [MinTiltRange] => 0 [MaxTiltRange] => 0 [MinTiltStep] => 0 [MaxTiltStep] => 0 [HasTiltSpeed] => 0 [MinTiltSpeed] => 0 [MaxTiltSpeed] => 0 [HasTurboTilt] => 0 [TurboTiltSpeed] => 0 [CanAutoScan] => 0 [NumScanPaths] => 0) | /usr/share/zoneminder/www/skins/classic/includes/control_functions.php | 229
-- | -- | -- | -- | -- | -- | -- | --

```

connortechnology, 2019-10-08:
Please update to latest, I have fixed a few things..

derKaY, 2019-10-13:
Thanks ... but not yet via ZM Repro Master (apt update)?

connortechnology, 2019-10-14:
Latest zmrepo for buster is zoneminder_1.33.14~20191013.385.

What does dpkg -s zoneminder say?

Might need some screen shots of your control capabilities screen Monitor->Control -> Edit

derKaY, 2019-10-20:
sry for belated: Version: 1.33.14~20190920082028-buster

connortechnology, 2019-10-20:
kay.. so that is old.

derKaY, 2019-10-21:
now i´ve installed 1.33.14~20191021.454-buster - works fine!

closed 2019-10-21 by derKaY