ZoneMinder/zoneminder #2720: Generated mp4 have a still picture after a few hours of running ZM

rhaamo, 2019-10-06:
**Describe Your Environment**
- Version of ZoneMinder: zoneminder_1.33.14~20191002084954
- How you installed ZoneMinder: built with do_debian_package
- Full name and version of OS: ubuntu 19.04

**If the issue concerns a camera**
- Ubiquiti Unifi G3 flex
- 25fps
- 1080p
- ZoneMinder Source Type: ffmpeg, TCP, 1920x1080 / 1080p, colorspace 32bit
- Analysis FPS, Max and Alarm max are blank
- Storage: Frames + Analysis images if available; writer : H264 camera passthrough, no specific optional encoder params, store audio stream is checked
- Buffers: 57 (zoneminder kept saying to bump it), 25, 25, 25, 1000, 1

**Describe the bug**
MJPEG replay is ok, MP4 isn't, I'm getting the audio right, but the image stay still with the first frame I think.
Tried playing in browser, rsync the .mp4 and playing in VLC and same issue.

**To Reproduce**
Started getting that issue a few weeks ago, happens after a few hours of running zoneminder, restarting zoneminder fix the issue.

**Expected behavior**
Video properly working.

**Debug Logs**
I activated log debug, level 3, grepped the logs with my event number (243333), and one file with "zms_e243333" seemed the best match.
There is one about event matching / frames being written but nothing weird in it and it's only the jpeg pictures save.

<details>
<summary>very long debug log</summary>

```
10/06/19 07:51:04.504449 zms[2144].DB1-zm_logger.cpp/246 [LogOpts: level=DB3 effective=DB3, screen=OFF, database=INF, logfile=DB3->/var/log/zm/zm_debug.log.02144, syslog=INF]
10/06/19 07:51:04.569277 zms[2144].DB1-zms.cpp/95 [Query: mode=jpeg&scale=100&maxfps=30&replay=single&source=event&event=243333&rand=1570341059&auth=b93661a3ff8642c3a66ca2ec66619136]
10/06/19 07:51:04.569316 zms[2144].DB1-zms.cpp/172 [Unknown parameter passed to zms rand=1570341059]
10/06/19 07:51:04.569416 zms_e243333[2144].DB1-zm_logger.cpp/246 [LogOpts: level=DB3 effective=DB3, screen=OFF, database=INF, logfile=DB3->/var/log/zm/zm_debug.log.02144, syslog=INF]
10/06/19 07:51:04.569446 zms_e243333[2144].DB1-zm_user.cpp/239 [Attempting to authenticate user from auth string 'REDACTED']
10/06/19 07:51:04.569879 zms_e243333[2144].DB1-zm_user.cpp/277 [AUTH_HASH_TTL is 48]
10/06/19 07:51:04.569930 zms_e243333[2144].DB1-zm_user.cpp/304 [Checking auth_key 'REDACTED--REDACTED' -> auth_md5 'REDACTED' == 'REDACTED']
10/06/19 07:51:04.569947 zms_e243333[2144].DB1-zm_user.cpp/313 [No match for REDACTED]
.. snip snip they are the same ..
10/06/19 07:51:04.570656 zms_e243333[2144].DB1-zm_user.cpp/313 [No match for REDACTED]
10/06/19 07:51:04.570675 zms_e243333[2144].DB1-zm_user.cpp/304 [Checking auth_key 'REDACTED--REDACTED' -> auth_md5 'REDACTED' == 'REDACTED']
10/06/19 07:51:04.570688 zms_e243333[2144].DB1-zm_user.cpp/309 [Authenticated user 'admin']
10/06/19 07:51:04.570716 zms_e243333[2144].DB1-zm_utils.cpp/288 [Detected a x86\x86-64 processor with SSE4.2]
10/06/19 07:51:04.570758 zms_e243333[2144].DB2-zm_stream.h/167 [Setting replay_rate 100]
10/06/19 07:51:04.570771 zms_e243333[2144].DB3-zms.cpp/307 [Setting stream start to frame (1)]
10/06/19 07:51:04.579599 zms_e243333[2144].DB1-zm_storage.cpp/94 [No id passed to Storage constructor.  Using default path /var/cache/zoneminder/events instead]
10/06/19 07:51:04.579673 zms_e243333[2144].DB3-zm_stream.cpp/74 [FPS:14.47, MaxFPS:30.00, BaseFPS:14.47, EffectiveFPS:14.47, FrameMod:1, replay_rate(100)]
10/06/19 07:51:04.579696 zms_e243333[2144].DB3-zm_eventstream.cpp/210 [fps set by frame_count(207)/duration(14.310000)]
10/06/19 07:51:04.581532 zms_e243333[2144].DB1-zm_eventstream.cpp/282 [Loading video file from /var/cache/zoneminder/events/5/2019-10-06/243333/243333-video.mp4]
10/06/19 07:51:04.582245 zms_e243333[2144].INF-zm_ffmpeg.cpp/79 [Enabling ffmpeg logs, as LOG_DEBUG+LOG_FFMPEG are enabled in options]
10/06/19 07:51:04.650155 zms_e243333[2144].DB3-zm_ffmpeg.cpp/67 [Opening '/var/cache/zoneminder/events/5/2019-10-06/243333/243333-video.mp4' for reading
]
10/06/19 07:51:04.650207 zms_e243333[2144].DB3-zm_ffmpeg.cpp/67 [Setting default whitelist 'file,crypto'
]
10/06/19 07:51:04.778011 zms_e243333[2144].DB3-zm_ffmpeg.cpp/67 [Format mov,mp4,m4a,3gp,3g2,mj2 probed with size=2048 and score=100
]
10/06/19 07:51:04.778128 zms_e243333[2144].DB3-zm_ffmpeg.cpp/67 [ISO: File Type Major Brand: isom
]
10/06/19 07:51:04.778242 zms_e243333[2144].DB3-zm_ffmpeg.cpp/67 [Unknown dref type 0x206c7275 size 12
]
10/06/19 07:51:04.778691 zms_e243333[2144].DB3-zm_ffmpeg.cpp/67 [Processing st: 0, edit list 0 - media time: 0, duration: 3600
]
10/06/19 07:51:04.778787 zms_e243333[2144].DB3-zm_ffmpeg.cpp/67 [Unknown dref type 0x206c7275 size 12
]
10/06/19 07:51:04.804828 zms_e243333[2144].DB3-zm_ffmpeg.cpp/67 [Processing st: 1, edit list 0 - media time: 0, duration: 798768
]
10/06/19 07:51:04.805044 zms_e243333[2144].DB3-zm_ffmpeg.cpp/67 [Before avformat_find_stream_info() pos: 12874475 bytes read:44757 seeks:1 nb_streams:2
]
[... 420 lines cut ...]
10/06/19 07:51:05.068500 zms_e243333[2144].DB3-zm_zone.cpp/791 [Got coordinate 1084,290 from polygon string]
10/06/19 07:51:05.068516 zms_e243333[2144].DB3-zm_zone.cpp/791 [Got coordinate 1068,292 from polygon string]
10/06/19 07:51:05.068556 zms_e243333[2144].DB3-zm_zone.cpp/811 [Successfully parsed polygon string]
10/06/19 07:51:05.076669 zms_e243333[2144].DB3-zm_zone.cpp/759 [Parsing polygon string '980,380 1019,380 1018,423 982,424']
10/06/19 07:51:05.076736 zms_e243333[2144].DB3-zm_zone.cpp/791 [Got coordinate 980,380 from polygon string]
10/06/19 07:51:05.076749 zms_e243333[2144].DB3-zm_zone.cpp/791 [Got coordinate 1019,380 from polygon string]
10/06/19 07:51:05.076761 zms_e243333[2144].DB3-zm_zone.cpp/791 [Got coordinate 1018,423 from polygon string]
10/06/19 07:51:05.076772 zms_e243333[2144].DB3-zm_zone.cpp/791 [Got coordinate 982,424 from polygon string]
10/06/19 07:51:05.076786 zms_e243333[2144].DB3-zm_zone.cpp/811 [Successfully parsed polygon string]
10/06/19 07:51:05.084554 zms_e243333[2144].DB1-zm_monitor.cpp/2384 [Loaded monitor 5(Entree porte bois), 7 zones]
10/06/19 07:51:05.084612 zms_e243333[2144].DB3-zm_stream.cpp/352 [comms open at ]
10/06/19 07:51:05.084639 zms_e243333[2144].DB3-zm_eventstream.cpp/804 [frame rate is: (14.465409)]
10/06/19 07:51:05.084660 zms_e243333[2144].DB3-zm_stream.cpp/74 [FPS:14.47, MaxFPS:30.00, BaseFPS:14.47, EffectiveFPS:14.47, FrameMod:1, replay_rate(100)]
10/06/19 07:51:05.084679 zms_e243333[2144].DB2-zm_eventstream.cpp/829 [Not checking command queue]
10/06/19 07:51:05.084692 zms_e243333[2144].DB3-zm_eventstream.cpp/840 [Not paused at frame 1]
10/06/19 07:51:05.084704 zms_e243333[2144].DB1-zm_eventstream.cpp/855 [replay rate(100) in_event(1) time_to_event(0.000000)=curr_stream_time(1570340953.000000)-frame timestamp:1570340967.310000]
10/06/19 07:51:05.084720 zms_e243333[2144].DB3-zm_eventstream.cpp/890 [cur_frame_id (1-1) mod frame_mod(1)]
10/06/19 07:51:05.084732 zms_e243333[2144].DB3-zm_eventstream.cpp/896 [frame delta 0us ]
10/06/19 07:51:05.084744 zms_e243333[2144].DB3-zm_eventstream.cpp/899 [delta 0 = base_fps(14.465409)/effective fps(14.465409)]
10/06/19 07:51:05.084758 zms_e243333[2144].DB3-zm_eventstream.cpp/902 [delta 33333 = base_fps(14.465409)/effective fps(14.465409) from 30fps]
10/06/19 07:51:05.084772 zms_e243333[2144].DB2-zm_eventstream.cpp/620 [Sending frame 1]
10/06/19 07:51:05.086462 zms_e243333[2144].ERR-zm_eventstream.cpp/762 [Unable to send raw frame 1: Broken pipe]
```

</details>

connortechnology, 2019-10-08:
We need the zmc logs. zms just plays the file.

rhaamo, 2019-10-08:
Ok, there is too much logs now and grepping on them to find the right bit is a nightmare.
I will purge them, restart ZM and will come back with the logs in a day or two.

connortechnology, 2019-10-08:
Set LOG_LEVEL_FILE=>Debug, LOG_DEBUG_FILE to empty.  Then /var/log/zm/zmc_m5.log will contain the relevant logs

rhaamo, 2019-10-08:
Ah! That will help thanks!

rhaamo, 2019-10-09:
Ok the issue is reproduced.

I've noticed something "weird" for me:
```
.DB2-zm_packetqueue.cpp/45 [Inserting packet with dts -9223372036854775808 because queue is empty or invalid dts]
```

I've grepped the logs, they start to appears at xxH, a video generated before xxH is OK, a video after xxH is still.

For practical reasons I've created a gist with the whole minute when it happened, if you want more logs let me know I will manage to put the full file somewhere.

Just some other notes: my "legacy" 720p ffmpeg with h264 passthrough (but no audio !) does not have that issue and no negative dts in logs.

https://gist.github.com/rhaamo/f6986fc7d8ecb6c8760fc8839a405352

connortechnology, 2019-10-09:
Will take some time to look through this. One thing to note:

ImageBufferCount 57 is too small. Needs to be at least 76.

Is a warning for a reason.Your buffers aren't large to hold all frames since the last keyframe.

rhaamo, 2019-10-09:
Yeah I've seen the ImageBufferCount warning and bumped it a bunch since, I will bump it to 80 and restart.

for ram:
```
              total        used        free      shared  buff/cache   available
Mem:           7961        1182         247        1287        6530        5112
Swap:          4095         328        3767
```
I think that should be good ?

(also /dev/shm is 5.0G and used at 17% right now)

connortechnology, 2019-10-09:
So what I am seeing is that the decoder is giving us packets with no timestamps.  We try to queue these but maybe I should just drop them.  I don't know if the increased buffers will prevent the timestamp problem though.

connortechnology, 2019-10-09:
Also that log dum doesn't have any writing to disk in it.  That's the part we will need to see.

connortechnology, 2019-10-09:
You can always switch to mocord

rhaamo, 2019-10-09:
I don't see writings in zmc, zma does have some where it write jpg capture frames to disk.

Isn't mocord going to use way more disk space by recording continuously rather than just saving where an event occurs ?

connortechnology, 2019-10-09:
if saving just jpg's, then zma does the writing.  If mp4 passthrough then zmc does it.  You can also use the force alarm buttons on live view to trigger events.

rhaamo, 2019-10-11:
ok so bumping the buffer without restarting made it work a few hours more, then I'm now getting the still video again with negative dts in logs.

Is that logging config sufficient to have the writing stuff in zmc_m?.log ?
[image]

connortechnology, 2019-10-11:
yes

rhaamo, 2019-10-11:
Ok, I've done a `tail -f zmc_m5.log > zmc_m5_event_manual.log`, waited a few seconds, "force alarm" for ~12s, cancelled, wait ~10s, and here is the log:
https://gist.github.com/rhaamo/80c4f7bbaef7de55a066a8b84075ad9e

The event, as told by ZM is `10/11 18:58:31 until 10/11 18:58:49`

Hope it's a good one.

connortechnology, 2019-10-11:
So ffmpeg isn't able to figure out when to show each image.  Each video packet has a timestamp value of AV_NOPTS_VALUE.  So I've added some code to just start at 0 and increase each one appropriately... might work.
Please update and let's see how it goes.

rhaamo, 2019-10-11:
thanks !
update done, will report back in a few days.

rhaamo, 2019-10-14:
Looked this morning and I've started to get stills a day ago.
Just forced an event "10/14 08:25:17 until 10/14 08:25:30", here is the log: https://gist.github.com/rhaamo/80c4f7bbaef7de55a066a8b84075ad9e#file-oct_14-log (oct_14 file)

connortechnology, 2019-10-16:
I've pushed another small change that should result in increasing dts values. Let me know.

rhaamo, 2019-10-17:
I've updated yesterday a few moment after your message and didn't really checked my ZM.
Was wondering this morning why I was having no spurious alerts during the night and I discovered all monitors are broken (including the ones without the still issue).

I don't really know which log extract you needs.

zmdc:
```
no memkey in zmMemInvalidate
10/17/2019 06:50:26.418318 zmdc[12757].INF [ZMServer:408] [Starting pending process, zma -m 5]
no memkey in zmMemInvalidate
10/17/2019 06:50:26.491630 zmdc[13196].INF [ZMServer:408] ['zma -m 5' started at 19/10/17 06:50:26]
10/17/2019 06:50:26.491628 zmdc[12757].INF [ZMServer:408] ['zma -m 5' starting at 19/10/17 06:50:26, pid = 13196]
10/17/2019 06:50:26.672389 zmdc[12757].ERR [ZMServer:698] ['zma -m 5' exited abnormally, exit status 255]
no memkey in zmMemInvalidate
```

zma_m5:
```
10/17/19 06:50:51.878384 zma_m5[13245].DB1-zm_monitor.cpp/523 [Waiting for capture daemon last_write_index(80), last_write_time(0)]
```

zmc_m5:
```
0/17/19 06:52:23.735930 zmc_m5[13511].DB1-zm_logger.cpp/246 [LogOpts: level=DB3 effective=DB3, screen=OFF, database=INF, logfile=DB9->/var/log/zm/zmc_m5.log, syslog=INF]
10/17/19 06:52:23.736041 zmc_m5[13511].DB1-zm_utils.cpp/288 [Detected a x86\x86-64 processor with SSE4.2]
10/17/19 06:52:23.736637 zmc_m5[13511].DB1-zm_monitor.cpp/2089 [Capture Delay!? 0.000]
10/17/19 06:52:23.736681 zmc_m5[13511].DB1-zm_monitor.cpp/2111 [Got 1 for v4l_captures_per_frame]
10/17/19 06:52:23.736721 zmc_m5[13511].DB2-zm_camera.cpp/54 [New camera id: 5 width: 1920 height: 1080 colours: 4 subpixelorder: 8 capture: 1]
10/17/19 06:52:23.736760 zmc_m5[13511].INF-zm_ffmpeg.cpp/82 [Enabling ffmpeg logs, as LOG_DEBUG+LOG_FFMPEG are enabled in options]
10/17/19 06:52:23.792214 zmc_m5[13511].DB3-zm_image.cpp/443 [Setting up static colour tables]
10/17/19 06:52:23.792334 zmc_m5[13511].DB1-zm_monitor.cpp/416 [monitor purpose=1]
10/17/19 06:52:23.792372 zmc_m5[13511].DB1-zm_monitor.cpp/425 [mem.size(8) SharedData=600 TriggerData=560 VideoStoreData=4128 timestamps=1280 images=80x8294400 = 663552000 total=663558632]
10/17/19 06:52:23.792404 zmc_m5[13511].DB1-zm_storage.cpp/94 [No id passed to Storage constructor.  Using default path /var/cache/zoneminder/events instead]
10/17/19 06:52:23.792432 zmc_m5[13511].DB1-zm_monitor.cpp/434 [Storage path: /var/cache/zoneminder/events]
10/17/19 06:52:23.793179 zmc_m5[13511].DB3-zm_monitor.cpp/546 [Connecting to monitor.  Purpose is 1]
10/17/19 06:52:23.793239 zmc_m5[13511].DB3-zm_monitor.cpp/553 [Success opening mmap file at (/dev/shm/zm.mmap.5)]
10/17/19 06:52:23.793269 zmc_m5[13511].DB3-zm_monitor.cpp/579 [MMap file size is 0]
10/17/19 06:52:23.793293 zmc_m5[13511].DB1-zm_monitor.cpp/584 [Unable to map file /dev/shm/zm.mmap.5 (663558632 bytes) to locked memory, trying unlocked]
10/17/19 06:52:23.793339 zmc_m5[13511].DB1-zm_monitor.cpp/587 [Mapped file /dev/shm/zm.mmap.5 (663558632 bytes) to unlocked memory]
10/17/19 06:52:23.793363 zmc_m5[13511].DB3-zm_monitor.cpp/599 [mmapped to 0x7f1b9c0c1000]
10/17/19 06:52:23.793384 zmc_m5[13511].DB3-zm_monitor.cpp/619 [Aligning shared memory images to the next 64 byte boundary]
10/17/19 06:52:23.793406 zmc_m5[13511].DB3-zm_monitor.cpp/622 [Allocating 80 image buffers]
10/17/19 06:52:23.793513 zmc_m5[13511].DB3-zm_monitor.cpp/653 [Success connecting]
10/17/19 06:52:24.270027 zmc_m5[13511].DB1-zm_monitor.cpp/500 [Monitor Entree porte bois has function 3,
label format = '%N - %d/%m/%y %H:%M:%S', label X = 0, label Y = 0, label size = 1,
image buffer count = 80, warmup count = 25, pre-event count = 25, post-event count = 25, alarm frame count = 1,
fps report interval = 100, ref blend percentage = 12, alarm ref blend percentage = 6, track motion = 0]
10/17/19 06:52:24.270662 zmc_m5[13511].DB1-zm_zone.cpp/884 [Got 8 zones for monitor Entree porte bois]
10/17/19 06:52:24.270697 zmc_m5[13511].DB3-zm_zone.cpp/759 [Parsing polygon string '838,446 1012,493 1146,495 1140,790 744,835 535,850 522,447']
10/17/19 06:52:24.270716 zmc_m5[13511].DB3-zm_zone.cpp/791 [Got coordinate 838,446 from polygon string]
10/17/19 06:52:24.270729 zmc_m5[13511].DB3-zm_zone.cpp/791 [Got coordinate 1012,493 from polygon string]
10/17/19 06:52:24.270742 zmc_m5[13511].DB3-zm_zone.cpp/791 [Got coordinate 1146,495 from polygon string]
10/17/19 06:52:24.270755 zmc_m5[13511].DB3-zm_zone.cpp/791 [Got coordinate 1140,790 from polygon string]
10/17/19 06:52:24.270768 zmc_m5[13511].DB3-zm_zone.cpp/791 [Got coordinate 744,835 from polygon string]
10/17/19 06:52:24.270780 zmc_m5[13511].DB3-zm_zone.cpp/791 [Got coordinate 535,850 from polygon string]
10/17/19 06:52:24.270793 zmc_m5[13511].DB3-zm_zone.cpp/791 [Got coordinate 522,447 from polygon string]
10/17/19 06:52:24.270815 zmc_m5[13511].DB3-zm_zone.cpp/811 [Successfully parsed polygon string]
10/17/19 06:52:24.275389 zmc_m5[13511].DB3-zm_zone.cpp/759 [Parsing polygon string '745,834 1143,790 1914,1076 847,1076']
10/17/19 06:52:24.275427 zmc_m5[13511].DB3-zm_zone.cpp/791 [Got coordinate 745,834 from polygon string]
10/17/19 06:52:24.275441 zmc_m5[13511].DB3-zm_zone.cpp/791 [Got coordinate 1143,790 from polygon string]
10/17/19 06:52:24.275453 zmc_m5[13511].DB3-zm_zone.cpp/791 [Got coordinate 1914,1076 from polygon string]
10/17/19 06:52:24.275466 zmc_m5[13511].DB3-zm_zone.cpp/791 [Got coordinate 847,1076 from polygon string]
10/17/19 06:52:24.275480 zmc_m5[13511].DB3-zm_zone.cpp/811 [Successfully parsed polygon string]
[... cut ...]
10/17/19 06:52:44.351649 zmc_m5[13547].DB2-zm_ffmpeg.cpp/647 [/root/zm/zoneminder_1.33.14~20191017063406.orig/src/zm_ffmpeg.cpp:647:Captured Packet:  pts: 63630=0.707000, dts: 63630, size: 19902, stream_index: 1, flags: 0000, keyframe(0) pos: -1, duration: 3600
]
10/17/19 06:52:44.351701 zmc_m5[13547].DB3-zm_ffmpeg.cpp/70 [nal_unit_type: 1(Coded slice of a non-IDR picture), nal_ref_idc: 3]
10/17/19 06:52:44.368142 zmc_m5[13547].WAR-zm_ffmpeg_camera.cpp/956 [Unable to receive frame 0: Success. error count is 0]
```

connortechnology, 2019-10-18:
Fixed some segfaults... please update

rhaamo, 2019-10-18:
updated, still having non-working monitors.

zmdc:
```
no memkey in zmMemInvalidate
10/18/2019 17:24:27.103581 zmdc[23283].INF [ZMServer:408] [Starting pending process, zma -m 7]
10/18/2019 17:24:27.168939 zmdc[23283].INF [ZMServer:408] ['zma -m 7' starting at 19/10/18 17:24:27, pid = 23590]
10/18/2019 17:24:27.168944 zmdc[23590].INF [ZMServer:408] ['zma -m 7' started at 19/10/18 17:24:27]
10/18/2019 17:24:27.271192 zmdc[23283].INF [ZMServer:696] ['zma -m 5' exited normally]
10/18/2019 17:24:27.358037 zmdc[23283].INF [ZMServer:408] [Starting pending process, zma -m 5]
no memkey in zmMemInvalidate
10/18/2019 17:24:27.443705 zmdc[23283].INF [ZMServer:408] ['zma -m 5' starting at 19/10/18 17:24:27, pid = 23591]
10/18/2019 17:24:27.443709 zmdc[23591].INF [ZMServer:408] ['zma -m 5' started at 19/10/18 17:24:27]
no memkey in zmMemInvalidate
10/18/2019 17:24:27.521376 zmdc[23283].ERR [ZMServer:698] ['zma -m 7' exited abnormally, exit status 255]
no memkey in zmMemInvalidate
10/18/2019 17:24:27.701323 zmdc[23283].ERR [ZMServer:698] ['zma -m 5' exited abnormally, exit status 255]
no memkey in zmMemInvalidate
no memkey in zmMemInvalidate
```

zma_m5:
```
10/18/19 17:24:58.218677 zma_m5[23650].DB1-zm_monitor.cpp/523 [Waiting for capture daemon last_write_index(80), last_write_time(0)]
10/18/19 17:24:59.218934 zma_m5[23650].DB1-zm_monitor.cpp/523 [Waiting for capture daemon last_write_index(80), last_write_time(0)]
10/18/19 17:25:00.219219 zma_m5[23650].DB1-zm_monitor.cpp/523 [Waiting for capture daemon last_write_index(80), last_write_time(0)]
10/18/19 17:25:01.686573 zma_m5[23705].DB1-zm_logger.cpp/246 [LogOpts: level=DB3 effective=DB3, screen=OFF, database=INF, logfile=DB9->/var/log/zm/zma_m5.log, syslog=INF]
10/18/19 17:25:01.686648 zma_m5[23705].DB1-zm_utils.cpp/288 [Detected a x86\x86-64 processor with SSE4.2]
10/18/19 17:25:01.687387 zma_m5[23705].DB1-zm_monitor.cpp/2089 [Capture Delay!? 0.000]
10/18/19 17:25:01.687435 zma_m5[23705].DB1-zm_monitor.cpp/2111 [Got 1 for v4l_captures_per_frame]
10/18/19 17:25:01.687472 zma_m5[23705].DB2-zm_camera.cpp/54 [New camera id: 5 width: 1920 height: 1080 colours: 4 subpixelorder: 8 capture: 0]
10/18/19 17:25:01.687514 zma_m5[23705].DB3-zm_image.cpp/443 [Setting up static colour tables]
10/18/19 17:25:01.687592 zma_m5[23705].DB1-zm_monitor.cpp/416 [monitor purpose=2]
10/18/19 17:25:01.687614 zma_m5[23705].DB1-zm_monitor.cpp/425 [mem.size(8) SharedData=600 TriggerData=560 VideoStoreData=4128 timestamps=1280 images=80x8294400 = 663552000 total=663558632]
10/18/19 17:25:01.687630 zma_m5[23705].DB1-zm_storage.cpp/94 [No id passed to Storage constructor.  Using default path /var/cache/zoneminder/events instead]
10/18/19 17:25:01.687645 zma_m5[23705].DB1-zm_monitor.cpp/434 [Storage path: /var/cache/zoneminder/events]
10/18/19 17:25:01.687684 zma_m5[23705].DB3-zm_monitor.cpp/546 [Connecting to monitor.  Purpose is 2]
10/18/19 17:25:01.687711 zma_m5[23705].DB3-zm_monitor.cpp/553 [Success opening mmap file at (/dev/shm/zm.mmap.5)]
10/18/19 17:25:01.687740 zma_m5[23705].DB3-zm_monitor.cpp/579 [MMap file size is 663558632]
10/18/19 17:25:01.687759 zma_m5[23705].DB1-zm_monitor.cpp/584 [Unable to map file /dev/shm/zm.mmap.5 (663558632 bytes) to locked memory, trying unlocked]
10/18/19 17:25:01.687784 zma_m5[23705].DB1-zm_monitor.cpp/587 [Mapped file /dev/shm/zm.mmap.5 (663558632 bytes) to unlocked memory]
10/18/19 17:25:01.687800 zma_m5[23705].DB3-zm_monitor.cpp/599 [mmapped to 0x7f95548c3000]
10/18/19 17:25:01.687817 zma_m5[23705].DB3-zm_monitor.cpp/619 [Aligning shared memory images to the next 64 byte boundary]
10/18/19 17:25:01.687835 zma_m5[23705].DB3-zm_monitor.cpp/622 [Allocating 80 image buffers]
10/18/19 17:25:01.687927 zma_m5[23705].DB3-zm_monitor.cpp/653 [Success connecting]
10/18/19 17:25:01.687947 zma_m5[23705].ERR-zm_monitor.cpp/491 [Shared data not initialised by capture daemon for monitor Entree porte bois]
10/18/19 17:25:02.118464 zma_m5[23709].DB1-zm_logger.cpp/246 [LogOpts: level=DB3 effective=DB3, screen=OFF, database=INF, logfile=DB9->/var/log/zm/zma_m5.log, syslog=INF]
10/18/19 17:25:02.118631 zma_m5[23709].DB1-zm_utils.cpp/288 [Detected a x86\x86-64 processor with SSE4.2]
10/18/19 17:25:02.119298 zma_m5[23709].DB1-zm_monitor.cpp/2089 [Capture Delay!? 0.000]
10/18/19 17:25:02.119343 zma_m5[23709].DB1-zm_monitor.cpp/2111 [Got 1 for v4l_captures_per_frame]
10/18/19 17:25:02.119402 zma_m5[23709].DB2-zm_camera.cpp/54 [New camera id: 5 width: 1920 height: 1080 colours: 4 subpixelorder: 8 capture: 0]
10/18/19 17:25:02.119470 zma_m5[23709].DB3-zm_image.cpp/443 [Setting up static colour tables]
10/18/19 17:25:02.119609 zma_m5[23709].DB1-zm_monitor.cpp/416 [monitor purpose=2]
10/18/19 17:25:02.119648 zma_m5[23709].DB1-zm_monitor.cpp/425 [mem.size(8) SharedData=600 TriggerData=560 VideoStoreData=4128 timestamps=1280 images=80x8294400 = 663552000 total=663558632]
10/18/19 17:25:02.119682 zma_m5[23709].DB1-zm_storage.cpp/94 [No id passed to Storage constructor.  Using default path /var/cache/zoneminder/events instead]
10/18/19 17:25:02.119712 zma_m5[23709].DB1-zm_monitor.cpp/434 [Storage path: /var/cache/zoneminder/events]
10/18/19 17:25:02.119749 zma_m5[23709].DB3-zm_monitor.cpp/546 [Connecting to monitor.  Purpose is 2]
10/18/19 17:25:02.119795 zma_m5[23709].DB3-zm_monitor.cpp/553 [Success opening mmap file at (/dev/shm/zm.mmap.5)]
10/18/19 17:25:02.119840 zma_m5[23709].DB3-zm_monitor.cpp/579 [MMap file size is 663558632]
[... cut ...]
10/18/19 17:26:39.574645 zmc_m5[24136].DB1-zm_ffmpeg.cpp/70 [concealing 6439 DC, 6439 AC, 6439 MV errors in P frame]
10/18/19 17:26:39.638571 zmc_m5[24136].WAR-zm_ffmpeg_camera.cpp/956 [Unable to receive frame 0: Success. error count is 0]
10/18/19 17:26:39.726781 zmc_m5[24136].INF-zm_ffmpeg_camera.cpp/755 [Unable to read packet from stream 0: error -541478725 "End of file".]
10/18/19 17:26:39.816100 zmc_m5[24136].INF-zm_monitor.cpp/2453 [Return from Capture (-1), signal loss]
0/18/19 17:26:39.902179 zmc_m5[24136].INF-zmc.cpp/316 [Failed to capture image from monitor 5 Entree porte bois (1/1)]
```

connortechnology, 2019-10-18:
Another hour, another build to try. Looks like I didn't push a relevant change. Please try again.

rhaamo, 2019-10-18:
Ok monitors are back alive, thanks !

I will report back on the initial issue as soon as I get an error or everything runs ok for a week.

rhaamo, 2019-10-19:
still movie bug is back.

event: 10/19 19:11:15 until 10/19 19:11:25 

log zmc_m5: https://gist.github.com/rhaamo/c7cffd959e4020d95401c5ac24021062

connortechnology, 2019-10-20:
I'm a little confused by what's going on with the timestamps here.  I've added some more debugging.

rhaamo, 2019-10-22:
here is with your debug:
event: 10/22 10:03:13 until 10/22 10:03:24
log: https://gist.github.com/rhaamo/7f8606c5c591095cf3e871da1d4dbcbd

connortechnology, 2019-10-22:
Please try latest

rhaamo, 2019-10-23:
event: 10/23 07:57:54 until 10/23 07:58:12 
log: https://gist.github.com/rhaamo/0c9d7cbf11cf97c1e1ced79ee43f7429

connortechnology, 2019-10-23:
Another day another change to try

rhaamo, 2019-10-23:
thanks, updated, will report back.

rhaamo, 2019-10-25:
after 2days of run more or less, everything seems good.

let's wait a few more days (had to restart ZM a few minutes ago) to see if we can call it fixed.

rhaamo, 2019-11-03:
Ok so the zm service has been running since a week and I've not seen any issue on the few times I've checked, latest events are ok, let's says it's fixed.

Thanks !

closed 2019-11-03 by rhaamo