ZoneMinder/zoneminder #2723: Event playback: cannot change scale or codec

rhaamo, 2019-10-11:
**Describe Your Environment**
- Version of ZoneMinder: latest of a few minutes
- How you installed ZoneMinder: do-release-script
- Full name and version of OS: ubuntu
- Browser name and version: Firefox 69.0.3

**Describe the bug**
Going to an event list, click on an event, change scale or codec, nothing changes.

Changing scale of the live monitor does work.

**Debug Logs**
Here are the console logs:
change scale:
```
ReferenceError: streamMode is not defined skins_classic_views_js_event-base-1570817359.js:157:18
```

change codec:
```
ReferenceError: eventData is not defined skins_classic_views_js_event-base-1570817359.js:147:3
```

connortechnology, 2019-10-11:
I suspect that the error happens earlier.  Check the js logs after page load before clicking on buttons.  Maybe check view source for php errors corrupting the javascript.

rhaamo, 2019-10-11:
the web_php.log doesn't present any error.

View source shows something:
```
var monitorUrl = '<br />
<b>Notice</b>:  Undefined property: ZM\Storage::$ServerId in <b>/usr/share/zoneminder/www/includes/Storage.php</b> on line <b>119</b><br />
https://xxx/zm/index.php';
```

The part with `// PHP variables to JS` does have connKey, eventData and streamMode defined.

connortechnology, 2019-10-11:
Aha.  Try latest update

rhaamo, 2019-10-12:
Just updated and scale/codec works.
Thank you !

closed 2019-10-12 by rhaamo