ZoneMinder/zoneminder #2724: 1.34 Name suggestions

pliablepixels, 2019-10-12:
Have you decided a name yet? Here are a few:

If we plan to continue Metallica:
- Wherever I may roam
- One (Not a great name, but it's gotta be there before we move on from Metallica)

If Zeppelin is not taboo:
- Going to California

Or we go Canada with:
- Razor love, Neil Young

connortechnology, 2019-10-14:
I think we can stick with Metallica for now.  I propose that we only use a song name for major releases, not point releases.  Should last us for 20 years at the rate we are going.

I think we half agreed on My Friend Of Misery.

connortechnology, 2019-10-14:
Going to California would have been perfect for 1.32, as most of it was me in California working for Tesla.

pliablepixels, 2019-10-15:
I just checked. Metallica has a total of 151 songs. One per major release will last way beyond 20 years ;-)

closed 2019-10-20 by pliablepixels