ZoneMinder/zoneminder #2728: Audio is not written correctly when capturing events using H264 Camera Passthrough

neillbell, 2019-10-17:
**Describe Your Environment**
- Version of ZoneMinder: 1.33.14~20191011140513-bionic
- How you installed ZoneMinder: ppa:iconnor/zoneminder-master
- Full name and version of OS: Ubuntu 18.04.3 LTS

**Camera 1, MP2 Audio**
- HikVision DS-2CD2442FWD-IW
- 10 fps
- 1920x1080
- ZoneMinder Source Type: ffmpeg
- Video Writer: H264 Camera Passthrough
- Audio Type: MP2, 16kHz sampling, 64kbps bitrate

**Camera 2, G.711 Audio**
- SV3C SD5W-1080PS-HX 
- 20 fps
- 1920x1080
- ZoneMinder Source Type: ffmpeg
- Video Writer: H264 Camera Passthrough
- Audio Type: MP2, G.711
 
**Describe the bug**
Audio is not written correctly when capturing events using H264 Camera Passthrough mode. With G.711, audio is distorted and the audio stream continues playing after the video stream ends, both with Chrome and with mpv. With MP2, the video writer crashes. 

**To Reproduce**
Steps to reproduce the behavior:
1. Visit the camera feed page on the console
2. Click on "Force Alarm" to start an event capture.
3. Click on "Cancel Forced Alarm" to end the event capture.
4. Play back the generated event.

**Expected behavior**
Video of the event with accompanying undistorted, synced audio.

**Debug Logs**
Camera 1
[zmc_m9.log](https://github.com/ZoneMinder/zoneminder/files/3741012/zmc_m9.log)
Camera 2
[zmc_m5.log](https://github.com/ZoneMinder/zoneminder/files/3741016/zmc_m5.log)

connortechnology, 2019-10-17:
WHAT?!  I broke it again?! Dang..

neillbell, 2019-10-17:
Sorry about that. I really didn't want to bug you about it.

connortechnology, 2019-10-17:
So I may need access to your cameras again in order to fix this quickly.  I did just push a change that should fix the segfault.

neillbell, 2019-10-18:
No problem. I went ahead and re-enabled the port forwarding like before. It's set up for MP2 audio. Let me know if you would me to set up the metronome again.

I'm not seeing your change on the PPA. Does it take a while to propagate?

connortechnology, 2019-10-18:
Okay fixed another crash.  audo/video duration looks good, need some audio (metronome) in order to determine quality of the audio.

neillbell, 2019-10-18:
I've set up the metronome and updated from the PPA. Unfortunately, everything is broken now. I can't get any camera to work.

neillbell, 2019-10-18:
Here's a log.
[zmc_m4.log](https://github.com/ZoneMinder/zoneminder/files/3745038/zmc_m4.log)

connortechnology, 2019-10-18:
It is working perfectly for me.  Audio sync is perfect.

Please make sure you do zmupdate.pl -f.  There is a new config option to set the timezone from Options instead of in php.ini.

neillbell, 2019-10-18:
I'm still having a problem. On a spare server I even purged ZM, deleted the database and reinstalled. Still, nothing works except V4L cameras. For what it's worth, things work fine if I downgrade to 1.33.14~20191011140513-bionic.

I also can't find that new timezone option. Where is it?

Would you like me to switch the camera to G.711 audio so you can test that.

connortechnology, 2019-10-18:
packages are still building. Current version is zoneminder - 1.33.14~20191018144429-bionic 

Yes let's try G.711

connortechnology, 2019-10-18:
timezone option is in Options -> System
Search for Timezone

neillbell, 2019-10-18:
Sorry, I missed a PPA update. It's all good now.

I've switch the camera to G.711.

commit d9d91c4a (2019-10-20): Fix #2728 by resetting the frame size before reading from the fifo
closed 2019-10-20 by connortechnology