ZoneMinder/zoneminder #2731: zms error when deleting events mid playback

dsl, 2019-10-21:
Development version, built from source.

When deleting an event mid playback, I usually see two errors in the logs:

`Oct 21 17:49:55 temjin zms_e693[24354]: ERR [zms_e693] [Can't open /var/cache/zoneminder/events/1/2019-10-21/693/00052-capture.jpg: No such file or directory]
Oct 21 17:49:55 temjin zms_e693[24354]: ERR [zms_e693] [Can't open /var/cache/zoneminder/events/1/2019-10-21/693/00052-capture.jpg: No such file or directory]`

In and of itself, this isn't a major issue, but I've also seen occasions where something gets upset and spams the logs repeatedly with this message. In an hour it logged the same message over 40000 times. Restarting apache stopped the spam.

Discussion on slack suggested that there needs to be more locking between the webui and the backend, so that if an event is to be deleted, the playback can be stopped first.

connortechnology, 2019-10-23:
Interesting that this happens, because when you click delete, we send a pause command to zms before sending the ajax command to delete the event.  As discussed, there isn't any locking, but I am still surprised that the pause doesn't take effect.  Perhaps we need to send STOP instead of pause.

dsl, 2019-10-25:
So in fact there are two cases here. First one is watch a clip to completion, give it at least five seconds for the dust to settle then delete it:

`10/25/19 01:09:23.945565 zms_e1009[310].DB1-zm_eventstream.cpp/856 [replay rate(100) in_event(1) time_to_event(-35.000000)=curr_stream_time(1571930076.000000)-frame timestamp:1571930076.600000]
10/25/19 01:09:24.006048 zms_e1009[310].DB1-zm_eventstream.cpp/856 [replay rate(100) in_event(1) time_to_event(-35.060000)=curr_stream_time(1571930076.060000)-frame timestamp:1571930076.600000]
10/25/19 01:09:24.096405 zms_e1009[310].DB1-zm_eventstream.cpp/856 [replay rate(100) in_event(1) time_to_event(-35.150000)=curr_stream_time(1571930076.150000)-frame timestamp:1571930076.600000]
10/25/19 01:09:24.156938 zms_e1009[310].DB1-zm_eventstream.cpp/856 [replay rate(100) in_event(1) time_to_event(-35.210000)=curr_stream_time(1571930076.210000)-frame timestamp:1571930076.600000]
10/25/19 01:09:24.267425 zms_e1009[310].DB1-zm_eventstream.cpp/856 [replay rate(100) in_event(1) time_to_event(-35.320000)=curr_stream_time(1571930076.320000)-frame timestamp:1571930076.600000]
10/25/19 01:09:24.317927 zms_e1009[310].DB1-zm_eventstream.cpp/856 [replay rate(100) in_event(1) time_to_event(-35.370000)=curr_stream_time(1571930076.370000)-frame timestamp:1571930076.600000]
10/25/19 01:09:24.438420 zms_e1009[310].DB1-zm_eventstream.cpp/856 [replay rate(100) in_event(1) time_to_event(-35.490000)=curr_stream_time(1571930076.490000)-frame timestamp:1571930076.600000]
10/25/19 01:09:25.005908 zms_e1009[310].DB1-zm_eventstream.cpp/492 [Got QUERY command, sending STATUS]
10/25/19 01:09:25.006037 zms_e1009[310].DB1-zm_eventstream.cpp/525 [Size of msg 24]
10/25/19 01:09:25.006113 zms_e1009[310].DB1-zm_eventstream.cpp/819 [Have command queue]
10/25/19 01:09:26.057063 zms_e1009[310].DB1-zm_eventstream.cpp/492 [Got QUERY command, sending STATUS]
10/25/19 01:09:26.057192 zms_e1009[310].DB1-zm_eventstream.cpp/525 [Size of msg 24]
10/25/19 01:09:26.057268 zms_e1009[310].DB1-zm_eventstream.cpp/819 [Have command queue]
10/25/19 01:09:27.108568 zms_e1009[310].DB1-zm_eventstream.cpp/492 [Got QUERY command, sending STATUS]
10/25/19 01:09:27.108699 zms_e1009[310].DB1-zm_eventstream.cpp/525 [Size of msg 24]
10/25/19 01:09:27.108769 zms_e1009[310].DB1-zm_eventstream.cpp/819 [Have command queue]
10/25/19 01:09:28.159791 zms_e1009[310].DB1-zm_eventstream.cpp/492 [Got QUERY command, sending STATUS]
10/25/19 01:09:28.159889 zms_e1009[310].DB1-zm_eventstream.cpp/525 [Size of msg 24]
10/25/19 01:09:28.159945 zms_e1009[310].DB1-zm_eventstream.cpp/819 [Have command queue]
10/25/19 01:09:29.210693 zms_e1009[310].DB1-zm_eventstream.cpp/492 [Got QUERY command, sending STATUS]
10/25/19 01:09:29.210920 zms_e1009[310].DB1-zm_eventstream.cpp/525 [Size of msg 24]
10/25/19 01:09:29.211005 zms_e1009[310].DB1-zm_eventstream.cpp/819 [Have command queue]
10/25/19 01:09:30.261987 zms_e1009[310].DB1-zm_eventstream.cpp/492 [Got QUERY command, sending STATUS]
10/25/19 01:09:30.262115 zms_e1009[310].DB1-zm_eventstream.cpp/525 [Size of msg 24]
10/25/19 01:09:30.262193 zms_e1009[310].DB1-zm_eventstream.cpp/819 [Have command queue]
10/25/19 01:09:31.313198 zms_e1009[310].DB1-zm_eventstream.cpp/492 [Got QUERY command, sending STATUS]
10/25/19 01:09:31.313318 zms_e1009[310].DB1-zm_eventstream.cpp/525 [Size of msg 24]
10/25/19 01:09:31.313380 zms_e1009[310].DB1-zm_eventstream.cpp/819 [Have command queue]
10/25/19 01:09:32.364485 zms_e1009[310].DB1-zm_eventstream.cpp/492 [Got QUERY command, sending STATUS]
10/25/19 01:09:32.364656 zms_e1009[310].DB1-zm_eventstream.cpp/525 [Size of msg 24]
10/25/19 01:09:32.364748 zms_e1009[310].DB1-zm_eventstream.cpp/819 [Have command queue]
10/25/19 01:09:33.415776 zms_e1009[310].DB1-zm_eventstream.cpp/492 [Got QUERY command, sending STATUS]
10/25/19 01:09:33.415897 zms_e1009[310].DB1-zm_eventstream.cpp/525 [Size of msg 24]
10/25/19 01:09:33.415961 zms_e1009[310].DB1-zm_eventstream.cpp/819 [Have command queue]
10/25/19 01:09:34.466539 zms_e1009[310].DB1-zm_eventstream.cpp/492 [Got QUERY command, sending STATUS]
10/25/19 01:09:34.466655 zms_e1009[310].DB1-zm_eventstream.cpp/525 [Size of msg 24]
10/25/19 01:09:34.466711 zms_e1009[310].DB1-zm_eventstream.cpp/819 [Have command queue]
10/25/19 01:09:35.517652 zms_e1009[310].DB1-zm_eventstream.cpp/492 [Got QUERY command, sending STATUS]
10/25/19 01:09:35.517769 zms_e1009[310].DB1-zm_eventstream.cpp/525 [Size of msg 24]
10/25/19 01:09:35.517828 zms_e1009[310].DB1-zm_eventstream.cpp/819 [Have command queue]
10/25/19 01:09:36.568718 zms_e1009[310].DB1-zm_eventstream.cpp/492 [Got QUERY command, sending STATUS]
10/25/19 01:09:36.568863 zms_e1009[310].DB1-zm_eventstream.cpp/525 [Size of msg 24]
10/25/19 01:09:36.568940 zms_e1009[310].DB1-zm_eventstream.cpp/819 [Have command queue]
10/25/19 01:09:37.619842 zms_e1009[310].DB1-zm_eventstream.cpp/492 [Got QUERY command, sending STATUS]
10/25/19 01:09:37.619990 zms_e1009[310].DB1-zm_eventstream.cpp/525 [Size of msg 24]
10/25/19 01:09:37.620068 zms_e1009[310].DB1-zm_eventstream.cpp/819 [Have command queue]
10/25/19 01:09:38.716777 zms_e1009[310].DB1-zm_eventstream.cpp/492 [Got QUERY command, sending STATUS]
10/25/19 01:09:38.716923 zms_e1009[310].DB1-zm_eventstream.cpp/525 [Size of msg 24]
10/25/19 01:09:38.716997 zms_e1009[310].DB1-zm_eventstream.cpp/819 [Have command queue]
10/25/19 01:09:39.767984 zms_e1009[310].DB1-zm_eventstream.cpp/492 [Got QUERY command, sending STATUS]
[... cut ...]
10/25/19 01:09:41.596499 zms_e1009[310].DB1-zm_eventstream.cpp/525 [Size of msg 24]
10/25/19 01:09:41.596550 zms_e1009[310].DB1-zm_eventstream.cpp/819 [Have command queue]
10/25/19 01:09:46.621573 zms_e1009[310].ERR-zm_eventstream.cpp/676 [Can't open /var/cache/zoneminder/events/1/2019-10-24/1009/00391-capture.jpg: No such file or directory]
10/25/19 01:09:46.764617 zms_e1009[310].ERR-zm_eventstream.cpp/676 [Can't open /var/cache/zoneminder/events/1/2019-10-24/1009/00391-capture.jpg: No such file or directory]`

dsl, 2019-10-25:
And a runaway errors event, where I deleted the clip midway through viewing. Note, this doesn't happen every time:

    10/25/19 00:58:03.926225 zms_e1004[28691].DB1-zm_eventstream.cpp/856 [replay rate(100) in_event(1) time_to_event(-10.380000)=curr_stream_time(1571929982.380000)-frame timestamp:1571929984.000000]
    10/25/19 00:58:04.016518 zms_e1004[28691].DB1-zm_eventstream.cpp/856 [replay rate(100) in_event(1) time_to_event(-10.460000)=curr_stream_time(1571929982.460000)-frame timestamp:1571929984.000000]
    10/25/19 00:58:04.140116 zms_e1004[28691].DB1-zm_eventstream.cpp/856 [replay rate(100) in_event(1) time_to_event(-10.580000)=curr_stream_time(1571929982.580000)-frame timestamp:1571929984.000000]
    10/25/19 00:58:04.266791 zms_e1004[28691].DB1-zm_eventstream.cpp/492 [Got QUERY command, sending STATUS]
    10/25/19 00:58:04.266927 zms_e1004[28691].DB1-zm_eventstream.cpp/525 [Size of msg 24]
    10/25/19 00:58:04.267167 zms_e1004[28691].DB1-zm_eventstream.cpp/819 [Have command queue]
    10/25/19 00:58:04.267205 zms_e1004[28691].DB1-zm_eventstream.cpp/856 [replay rate(100) in_event(1) time_to_event(-10.690000)=curr_stream_time(1571929982.690000)-frame timestamp:1571929984.000000]
    10/25/19 00:58:04.267382 zms_e1004[28691].ERR-zm_eventstream.cpp/676 [Can't open /var/cache/zoneminder/events/1/2019-10-24/1004/00117-capture.jpg: No such file or directory]
    10/25/19 00:58:04.447901 zms_e1004[28691].DB1-zm_eventstream.cpp/308 [Got PAUSE command]
    10/25/19 00:58:04.448011 zms_e1004[28691].DB1-zm_eventstream.cpp/525 [Size of msg 24]
    10/25/19 00:58:04.448071 zms_e1004[28691].DB1-zm_eventstream.cpp/819 [Have command queue]
    10/25/19 00:58:09.458446 zms_e1004[28691].ERR-zm_eventstream.cpp/676 [Can't open /var/cache/zoneminder/events/1/2019-10-24/1004/00118-capture.jpg: No such file or directory]
    10/25/19 00:58:09.594544 zms_e1004[28691].ERR-zm_eventstream.cpp/676 [Can't open /var/cache/zoneminder/events/1/2019-10-24/1004/00118-capture.jpg: No such file or directory]

Those errors then repeat forever until the process is killed.

closed 2019-12-31 by stale[bot]

csindle, 2020-07-22:
I have exactly the same issue (on latest Firefox [78]). Shout if I can provide any more information.