ZoneMinder/zoneminder #2734: Javascript support libraries are out of date

dsl, 2019-10-21:
Development version, built from source.

A number of the javascript support libraries (jquery, jquery ui, chosen, mootools for starters) are out of date. Ideally they should be updated from a security perspective.

connortechnology, 2019-10-21:
Hold on.... 
mootools is up to date  at 1.6.0
jquery yeah we are at 2.2 whereas 3.4.1 is the latest.  We don't actually use much jquery so we can probably update without any breaks.
chosen is up to date at 1.8.7

connortechnology, 2019-10-21:
After consultation, the choice of jquery 2.2 is about browser support, mainly ancient IE.  We will update to the latest and greatest in 1.35.

dsl, 2019-10-22:
Right, that makes sense about 1.35.

Re chosen, my mistake, the trivial "issue" I meant to flag up with that one is we're including both the compressed and uncompressed versions, and probably don't need both.

Re mootools, I've no idea why I flagged that, sorry. You're absolutely right.

connortechnology, 2019-11-30:
Debian are really particular about including compressed vs uncompressed.  They view a minified version as being the same as compiled and thus demand that you include the source.  So we include both.

closed 2019-11-30 by connortechnology